JMRI/JMRI #4271: LayoutEditor: Add “Eliminate Anchor” option to Anchor popup menu.

geowar1, 2017-10-17:
Unlike “Delete” it will also replace the two adjacent track segments with a single track segment.
(Kinda of a counter to the “Split” track Segment idea I’m also proposing (as a separate issue).)

dsand47, 2017-10-17:
Or maybe **_join_** or **_merge_** track segments.

geowar1, 2017-10-17:
I'm currently using "Join"… ;-)

BTW: Have you notices that some popup menus say "Delete" and others "Remove"? Anyone know the difference? Is there one?

pnbgit, 2017-10-17:
I think "Merge segments" would be the best description for this one, join might be used when bringing two things together, for example.   If the two segments are in different blocks, does this menu option still appear and which block does the merged segment end up in?

geowar1, 2017-10-17:
On Oct 17, 2017, at 12:41 PM, Pete Brownlow <notifications@github.com> wrote:
> I think "Merge segments" would be the best description for this one, join might be used when bringing two things together, for example. 

Good call… I’m going with “Merge Adjacent Tracks” (and "Split Track Segment”).

> If the two segments are in different blocks, does this menu option still appear and which block does the merged segment end up in?

I thought about checking to see if the anchor was a block boundary and just not allowing the merge but then decided I didn’t feel like writing that code…

So right now the connect2 track segment just disappears (along with its assigned block).

;-)

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-10-17:
LayoutEditor Add “Split” action to Track Segment popup menu #4277

dsand47, 2017-10-17:
I would disable the merge option for anchor points that are block boundaries.  This way you don't have to relocate signal heads, masts, and sensors which can be attached to the anchor point.  Otherwise the code has to relocate these items to the new block boundary.  

To make things worse, if the connect2 track segment is the only piece of track in its block, the whole block disappears and now you potentially have two sets of heads, masts, and sensors to deal with.

pnbgit, 2017-10-18:
Dave beat me to it - I was going to ask what you will do with any signal heads, masts or NX sensors assigned to the block boundary between block1 and block2 if block2 now disappears?   

Presumably this also has a knock on effect to signal mast logic, of any preceding signals that previously had block2 being unoccupied in their list of conditions before the signal will clear.

Seems to me it might be easier just to disable the option for block dividers rather than deal with all the other consequences - presumably there is already code to choose which options are available at block dividers because set signal heads, set signal masts and set sensors only appear when it is a block divider.

geowar1, 2017-10-18:
> I would disable the merge option for anchor points that are block boundaries.
Done! (Replaced "Merge Adjacent Tracks" with "Cannot Merge Block Boundaries" (disabled).)
Now I'm just waiting for someone to assign it to me and merge it... (hint, hint... ;-)
(PR #4275 also)

pnbgit, 2017-10-28:
Just testing this on 4.9.6ish package build #1976.  Merge and splitting track segments is working nicely.  I did note, however, that when merging two track segments, the following error appears in the system console: 

2017-10-28 13:56:13,799 layoutEditor.PositionablePoint        ERROR - Attempt to assign more than allowed number of connections [AWT-EventQueue-0]

Maybe this is just transitory as the merge is done, so I don't know if it is something that needs fixing or not.

geowar1, 2017-10-28:
See PR https://github.com/JMRI/JMRI/pull/4327

geowar1, 2017-10-29:
PR #4327 has merged; please regress and close if fixed. TIA.

closed 2017-10-29 by geowar1