JMRI/JMRI #4286: Slot Monitor, Digitrax,  unable to parse Loconet Message

marcfour, 2017-10-19:
Slots 121 thru 126.   As seen via Loconet Monitor.  This is encountered with both JMRI 4.9.4 and JMRI 4.8. There is also a message againts a non-existant  Uhlenbrock device in both JMRI Versions.  This is encountered on a Digitrax DCS100, PR3 interface, JAVA 1.8.0_144

Resetting the DCS100 to clear all memory values does not affect results.

Report of slot 120 information:
	Loco 0 is Not Consisted, Free, operating in 28 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: Off/Paused; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
Unable to parse LocoNet message.
contents: BB 79 00 3D
Unable to parse LocoNet message.
contents: E7 0E 79 00 00 00 00 04 00 00 00 00 00 6B
Unable to parse LocoNet message.
contents: BB 7A 00 3E
Unable to parse LocoNet message.
contents: E7 0E 7A 00 00 00 00 04 00 00 00 00 00 68
Request Fast Clock information.
Response Fast Clock is Synchronized, Frozen, rate is 0:1. Day 0, 00:53. Last set by ID 0x00 0x00 (0).
	Master supports LocoNet 1.1; Track Status: Off/Paused; Programming Track Status: Available.
Request Programming Track information.
Programming Response: Uhlenbrock IB-COM / Intellibox II Programming Read Was Successful: CV1 value 0 (0x00, 00000000b).
Unable to parse LocoNet message.
contents: BB 7D 00 39
Unable to parse LocoNet message.
contents: E7 0E 7D 00 00 00 00 04 00 00 00 00 00 6F
Unable to parse LocoNet message.
contents: BB 7E 00 3A
Unable to parse LocoNet message.
contents: E7 0E 7E 00 00 00 00 04 00 00 00 00 00 6C

devel-bobm, 2017-10-19:
If you can define these "reserved" slots, we can provide a slot monitor which decodes them.

So far as I can tell, there are only three slots in the range 0x78 (120) thru 0x7F (127) which have defined functionality - the programming slot (0x7C or 124), the fast-clock slot (0x7b or 123), and the "Command station opsws" slot (0x7F or 127).  So far as I can tell, these are decoded in a not-too-unreasonable way.

Without actual hex data, I cannot investigate the Uhlenbrock-related text.

I think that reporting "unable to parse" for those reserved slots which are not otherwise defined is an entirely-acceptable answer.  They most definitely are not general-purpose slots, so interpreting them as holding loco data is not acceptable.  And without a definition from Digitrax, we cannot interpret them as anything else.

marcfour, 2017-10-23:
If I back track JMRI to 4.6, these " unable to parse " messages do not show up at all.  The Uhlenbrock  IB-COM message still appears when the Slot Monitor queries slot 124 (0x7C)  for what ever reason that is.  So something was changed after 4.6 and it affected Slot Monitor's  capability to parse the Loconet messages.

closed 2017-11-11 by bobjacobsen