JMRI/JMRI #4300: Exceptions when using Web Server

DanielBoudreau, 2017-10-25:
After turning on web services and accessing the main jmri page, I see the following exceptions being reported to the system console:

2017-10-25 12:59:52,313 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-10-25 12:59:52,325 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Daniel Boudreau\JMRI\log\messages.log [main]
2017-10-25 12:59:52,326 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Daniel Boudreau\JMRI\log\session.log [main]
2017-10-25 12:59:52,339 apps.Apps                             INFO  - PanelPro version 4.9.4+R4e943c1 starts under Java 1.8.0_144 on Windows 7 amd64 v6.1 at Wed Oct 25 12:59:52 EDT 2017 [main]
2017-10-25 12:59:52,586 apps.Apps                             INFO  - Starting with profile demo.3ec7db2f [main]
2017-10-25 12:59:52,871 node.NodeIdentity                     INFO  - Using jmri-0C84DCBAA785-3ec7db2f as the JMRI Node identity [main]
2017-10-25 12:59:54,850 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI_4.9.4\ [main]
2017-10-25 12:59:54,851 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Daniel Boudreau\JMRI\demo\ [main]
2017-10-25 12:59:54,852 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Daniel Boudreau\JMRI\demo\ [main]
2017-10-25 12:59:54,853 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Daniel Boudreau\JMRI\ [main]
2017-10-25 12:59:54,853 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Daniel Boudreau\ [main]
2017-10-25 12:59:54,854 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI_4.9.4\jython\ [main]
2017-10-25 13:00:00,743 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2017-10-25 13:00:01,194 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=4.0} [WebServer]
2017-10-25 13:00:24,197 json.JsonWebSocket                    WARN  - Unhandled Error (closing connection) [WebServer-59]
java.lang.NullPointerException: Required nonnull default for jmri.PowerManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:185)
	at jmri.server.json.power.JsonPowerHttpService.doPost(JsonPowerHttpService.java:83)
	at jmri.server.json.power.JsonPowerSocketService.onMessage(JsonPowerSocketService.java:40)
	at jmri.server.json.JsonClientHandler.onMessage(JsonClientHandler.java:154)
	at jmri.server.json.JsonClientHandler.onMessage(JsonClientHandler.java:94)
	at jmri.server.json.JsonWebSocket.onMessage(JsonWebSocket.java:76)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:234)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:66)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:226)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:162)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:367)
	at org.eclipse.jetty.websocket.common.extensions.AbstractExtension.nextIncomingFrame(AbstractExtension.java:176)
	at org.eclipse.jetty.websocket.common.extensions.compress.PerMessageDeflateExtension.nextIncomingFrame(PerMessageDeflateExtension.java:105)
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension.forwardIncoming(CompressExtension.java:142)
	at org.eclipse.jetty.websocket.common.extensions.compress.PerMessageDeflateExtension.incomingFrame(PerMessageDeflateExtension.java:85)
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:220)
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:220)
	at org.eclipse.jetty.websocket.common.Parser.parse(Parser.java:256)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:675)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:505)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:273)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:95)
	at org.eclipse.jetty.io.SelectChannelEndPoint$2.run(SelectChannelEndPoint.java:93)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume(ExecuteProduceConsume.java:303)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume(ExecuteProduceConsume.java:148)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589)
	at java.lang.Thread.run(Unknown Source)
2017-10-25 13:00:24,204 json.JsonWebSocket                    ERROR - Required nonnull default for jmri.PowerManager does not exist. [WebServer-59]
java.lang.NullPointerException: Required nonnull default for jmri.PowerManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:185)
	at jmri.server.json.power.JsonPowerHttpService.doPost(JsonPowerHttpService.java:83)
	at jmri.server.json.power.JsonPowerSocketService.onMessage(JsonPowerSocketService.java:40)
	at jmri.server.json.JsonClientHandler.onMessage(JsonClientHandler.java:154)
	at jmri.server.json.JsonClientHandler.onMessage(JsonClientHandler.java:94)
	at jmri.server.json.JsonWebSocket.onMessage(JsonWebSocket.java:76)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:234)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
[... 302 lines cut ...]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:234)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:66)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:226)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:162)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:367)
	at org.eclipse.jetty.websocket.common.extensions.AbstractExtension.nextIncomingFrame(AbstractExtension.java:176)
	at org.eclipse.jetty.websocket.common.extensions.compress.PerMessageDeflateExtension.nextIncomingFrame(PerMessageDeflateExtension.java:105)
	at org.eclipse.jetty.websocket.common.extensions.compress.CompressExtension.forwardIncoming(CompressExtension.java:142)
	at org.eclipse.jetty.websocket.common.extensions.compress.PerMessageDeflateExtension.incomingFrame(PerMessageDeflateExtension.java:85)
[... cut ...]
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume(ExecuteProduceConsume.java:303)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume(ExecuteProduceConsume.java:148)
	at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589)
	at java.lang.Thread.run(Unknown Source)

rhwood, 2017-10-28:
JMRI always creates a PowerManager instance, so why isn't it in your use case? How is your system configured to prevent that?

DanielBoudreau, 2017-12-05:
Seems to be fixed in 4.9.8.

closed 2017-12-05 by DanielBoudreau