JMRI/JMRI #4307: Regression: File names problem when saving from Layout Editor

pnbgit, 2017-10-26:
I am seeing this on the current not yet released 4.9.5  (build #6) on Windows 10, Java 8.

When you save a panel from the Layout Editor, and the file name has spaces, those spaces are replaced by the characters %20 which are not translated to spaces on save.

This problem is not present in the current release 4.8

[image]

geowar1, 2017-10-27:
Found it: <JmriConfigurationManager.java:227-229>:
       // make this url the default "Save Panels..." file
        JFileChooser ufc = jmri.configurexml.StoreXmlUserAction.getUserFileChooser();
        ufc.setSelectedFile(new File(url.getFile()));

It's being passed as a file... generated from a url… but how would I prevent the UTI/URI string conversion here?

rhwood, 2017-10-27:
You create the file from the URI not from the String representing the URI.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

geowar1, 2017-10-27:
See PR #4318

rhwood, 2017-10-28:
Fixed in #4318.

closed 2017-10-28 by rhwood