JMRI/JMRI #4329: Regression: LayoutEditor.LayoutTurnout popup menu “Set Sensors” dialog not getting turnout id from clicked on turnout

pnbgit, 2017-10-29:
This bug has just recently appeared, I think it is ok in the 4.9.5 files build #6, but it is present in 4.9.6ish package build #1976.

When you right click on a turnout and select "Set sensors", it should give you a dialog which shows the set sensor dialog for each leg of the turnout that is a block boundary.

eg:

[image]

But now, if I right on another turnout and choose set sensors, I only ever get the window for the first turnout I selected. 

This means that it is impossible to set up NX routing for any block boundary which includes a turnout, unless the user is prepared to edit the XML directly.

See also #4268 where it is also now impossible to set NX sensors at block boundaries that are an anchor point.  

This means that setting up NX routing is now broken - the only place you can set an NX sensor is at an edge connector.

dsand47, 2017-10-29:
I checked both 1976 and 1980 and the set sensors for a turnout are working as expected.

For this particular case, make sure that there are no track segments with the same block name attached to the missing leg of the turnout.  Especially watch for zero length track segments.  These are created when you drag a turnout onto another turnout to connect them.

Dave Sand

pnbgit, 2017-10-29:
Hi Dave, it is not a leg missing, it is that when you go to another turnout, and then right click on that one and choose set sensors, you get the window for the first turnout you tried.   A track segment with the same block name or a zero length track segment with the same or no block name would not cause these symptoms.
 After you confirmed it worked as you expected, did you then try another turnout and check you got the correct window for that second one, and not a repeat of the window for the first turnout you tried?

dsand47, 2017-10-29:
You are correct.  The scope of the problem is actually quite large.  It appears to affect any block boundary assignments for sensors, signal masts and probably signal heads (I say probably since I don't use signal heads).  The affected boundaries include end bumper, anchor point, edge connector and turnouts.

The actual problems tend to vary depending on the point type and object being connected.

The title of this issue should be changed to reflect the larger scope.

Dave Sand

geowar1, 2017-10-29:
See PR #4340

pnbgit, 2017-10-30:
Testing on package build # 1987 - Agree with Dave that this affects signal masts also and affects also anchor points, so presumably end bumpers and edge connectors also.

geowar1, 2017-11-01:
See PR #4362.

pnbgit, 2017-11-05:
I'm not sure where we are with this one now - PR #4362 does not seem to be showing as merged, Just tried it on package build 1997 and, whilst the problem of only ever getting the first turnout or anchor point appears to be fixed, I now get error messages when trying to set signal masts at turnouts. After right click on the turnout and select "set signal masts", the following messages box comes up:

[image]

This message box shows twice then the correct windows comes up, I think it does correctly show any existing signals.  It may only happen when not all legs of the turnout are block boundaries.

A similar problem, if I try to set sensors at a block divider anchor point, I get:

[image]

The set sensors window then appears, but any sensors previously set here are NOT shown in this window.  

(This is all on a panel where all this works fine in 4.8)

geowar1, 2017-11-05:
On Nov 5, 2017, at 10:39 AM, Pete Brownlow <notifications@github.com> wrote:
> I'm not sure where we are with this one now - PR #4362 <https://github.com/JMRI/JMRI/pull/4362> does not seem to be showing as merged,

Dave Sand and I are still testing… :(

> Just tried it on package build 1997 and, whilst the problem of only ever getting the first turnout or anchor point appears to be fixed, I now get error messages when trying to set signal masts at [...]
I think that’s been fixed… (That code is definitely “under the knife”.)

> This message box shows twice then the correct windows comes up, I think it does correctly show any existing signals. It may only happen when not all legs of the turnout are block boundaries.
> 
Yeah, I think that was fixed. (DaveS and I got “out of sync” with our fixes… and the merge didn’t go well…)

> A similar problem, if I try to set sensors at a block divider anchor point, I get:
>  <https://user-images.githubusercontent.com/16050680/32417911-3fbab402-c258-11e7-9b8a-5c4f9b00fed8.png>
> The set [...]
If DaveS and I get everything tested today we should be release the WIP on that PR… hopefully by tomorrow.

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

geowar1, 2017-11-10:
@pnbgit : can you confirm fixed (via PR #4282 now merged to master) and close?

pnbgit, 2017-11-10:
I checked it on test release 4.9.5 and all seems good now.  Not sure about PR #4282 , that's entitled "Update Eclipse instructions".

geowar1, 2017-11-10:
Oops… typo. That should be PR #4362.

pnbgit, 2017-11-10:
Hmm, slightly confusing.  in PR #4362 it says that 4 days ago geowar1 commented "Ok, I'm bailing to this PR… gonna reopen a new one…", then deleted the branch, but then it says that "bobjacobsen added this to the 4.9.5 milestone 3 days ago"  - so I'm not quite sure what that means.  No mention of merging anything anywhere in that PR.

geowar1, 2017-11-10:
Oh yeah, that was the one that git lost it's mind on… decided I'd changed 3290 files… (huh?)
Re-opened as #4382

pnbgit, 2017-11-10:
Yep, that'll be the one.  I just wanted to check the PR we were talking about was merged into the 4.9.5 test release, as that is what I have been testing.  So yes, looks like this one is now sorted.