JMRI/JMRI #4330: LayoutEditor.EdgeConnector add choices to show with no/in/out/both arrows

pnbgit, 2017-10-29:
The current change from the edge connector being not explictly displayed (it just showed as a positionable point like an anchor point or edge connector) to displaying as an "X" can make the panel display look cluttered.

[image]

George has proposed to add options to allow the user to select the edge connectors to display as "no/in/out/both arrows", and asked me to write this up as an issue so he doesn't forget.

IMHO, for backwards compatibility, it would be good to revert to the previous behaviour until the user has these options.

geowar1, 2018-02-27:
This should have been fixed by #4798. If you can verify then please close this issue. THX!

geowar1, 2019-12-22:
@pnbgit can you confirm this fixed… and close? Or if not let me know so I can investigate further.

closed 2020-01-06 by geowar1