JMRI/JMRI #4331: LayoutEditor.PositionablePoint: Hide if connected to hidden track segment”.

pnbgit, 2017-10-29:
The new buffer stop end bars are nice - but if the track segment it is connected to is hidden, the end bar still shows up.

Here is an example of a yard that is hidden to the users, but still present in the panel so that the block routing still works:

[image]

As you can see, the buffer stop end bars still appear out of edit mode:

[image]

This might also affect edge connectors which now have a visible display.

geowar1, 2017-10-31:
I think I'm actually going to solve this issue by making PositionablePoints (anchors, end bumpers and edge connectors) have user set "decorations" (like bridge  and tunnel entrances and exits, in, out or double (in, out or both) arrows, etc.). The defaults will be nothing for anchors, a single crosstie for end bumpers and an in/out "X" for edge connectors. (You could rename this issue: "LayoutEditor.PositionablePoint: Make 'decorations' user choosable.")

geowar1, 2019-12-22:
@pnbgit can you confirm this fixed… and close? Or if not let me know so I can investigate further.

closed 2020-01-06 by geowar1