JMRI/JMRI #4333: LayoutEditor.LayoutTurnout popup menu; Flatten “View block routing”

pnbgit, 2017-10-29:
"View block routing" on a turnout, it used to go straight in to the block routing window (as it still does on a block divider), but now you have to do another click into a sub-menu with only one option:

[image]

It would be good if this could revert to the previous behaviour for turnouts, as they can only be in one block anyway. The view block routing window tells you which block you are working with anyway.  The only time you need a submenu is for a crossover or flat crossing where there can be more than one block.

See also #4332

geowar1, 2017-10-29:
See PR #4346

geowar1, 2017-10-30:
@pnbgit: if you can, please confirm fix in PR #4246 and close. /poke me otherwise… ;-)

pnbgit, 2017-10-30:
Tested on package build # 1987 - rotate menu now flattened and current degrees shows in the top menu.

closed 2017-10-30 by pnbgit