JMRI/JMRI #4334: LayoutEditor:Icon popup menu “properties” changes background color"

pnbgit, 2017-10-29:
If I right click on an icon, and select "properties", then change something such as turn on or off bold font, the icon background changes from transparent to white.

[image]

[image]

geowar1, 2017-10-30:
See PR #4347

geowar1, 2017-10-30:
Assign OP (pnbgit) to verify as fixed and close.

pnbgit, 2017-10-30:
Tested on package build # 1987 and now fixed.

closed 2017-10-30 by pnbgit

pnbgit, 2017-12-09:
This one seems to have come back on latest build #4 of 4.9.7

bobjacobsen, 2017-12-09:
Are you sure you're really running what you think you are?  You're seeing several regressions that I'm not seeing.