JMRI/JMRI #4335: Roster All Entries page not displaying properly

SKT54, 2017-10-29:
Have just installed Decoder Pro for the first time and started to add my locos to the roster. However as you will see from the attached screenshot the page is not displaying properly as you can hardly see the entries.
[image]

bobjacobsen, 2017-10-30:
A jmriusers post had this suggestion: 'In the JMRI Preferences Window, select “Display” from the list on the left, and in the GUI tab, set the font size.'

Also, what Java version do you have installed?

SKT54, 2017-10-30:
It is now okay without me doing anything?  I opened Decoder Pro without my laptop being connected to the command station and it is fine. I also checked after connecting to the command station and it is also fine. I did follow your advice however as I preferred to have a larger font.

Many thanks for taking the time to help.

closed 2021-04-14 by stale[bot]