JMRI/JMRI #4342: Changes to DrawFrame ColorChooser in PR4287 make Shapes unable to edit colors

petecressman, 2017-10-29:
Shapes cannot create or edit colors.  Lines 150, 151 in .../shapes/DrawFrame.java improperly display JColorChooser.  What is this change trying to fix?

rhwood, 2017-10-29:
What change are you referring to? Please include the PR or commit number. As it is, I can't tell what you are referring to.

If you are referring to commit ae78c74, which changes the default panel displayed in the color chooser that was added in commit 1c22413.

petecressman, 2017-10-29:
PR4287 added lines 150 and 151 to jmri.jmrit.display.controlPanelEditor.shape.DrawFrame.java.  The previous behavior was quite satisfactory to me.  What was the change trying to improve on? Whatever the purpose, it made choosing colors impossible.

rhwood, 2017-10-29:
If the result of saving a Shape allows a full gamut of colors to be saved, and does not restrict the usable colors, then feel free to submit a PR to revert that change to DrawFrame.

Some JMRI things that have color values only allow 16 named colors, while others allow for more. If the shape can only save the 16 named colors, then stick to this change.

It's entirely possible that the author of #4287 thought that the objects handled by DrawFrame can only be saved with the 16 named colors.

petecressman, 2017-10-29:
The PR cannot make any color change, let alone 16.

pabender, 2017-10-30:
Sent from my iPad

> On Oct 29, 2017, at 4:05 PM, petecressman <notifications@github.com> wrote:
> 
> The PR cannot make any color change, let alone 16.
> 

Tell me how you come to that conclusion?

The change in question was made for consistency.  It doesn't do anything except override the default color chooser panel with the custom one added for other parts of JMRI, and I know the color chooser panel itself works, as I tested it every single place I added it.

To choose a new color, you hit the set button, which then brings up a color picker dialog.  You then select your color in the dialog and when the dialog closes the new color populates the swatch shown on the color chooser panel.

This particular window was one of the few places that previously had s color picker in place.

Paul

pabender, 2017-10-30:
@petecressman 

I went and looked at that screen again, and I am able to change the colors using the new color chooser:

[image]

So that color picker is definitely working.

Now, I did notice while getting that setup two things that aren't working:
1) In order to set those colors, I had to go back and edit the shapes.  It didn't work when I created them (which was odd)
2) If I moved the window before clicking done, the displayed color in the window reverted to the previous selection.

So there is at least one bug in there that we need to at further, but the basic functionality actually does work as expected.  

I suspect the bug is that the color model isn't being updated at the right time.

Paul

pabender, 2017-10-30:
@petecressman 

Upon further review, the button swatch color chooser only fails to set the color correctly if you set the color and THEN draw the shape.  If you draw the shape first, even for new shapes, it does the right thing.

pabender, 2017-10-30:
@petecressman 

One more observation.  I see the exact same behavior by commenting out lines 150 and 151, i.e. the old behavior.

The culpret behind this is in the colorChange method:

https://github.com/JMRI/JMRI/blob/06190dc261e3b3d93272cb8aa081461282589582/java/src/jmri/jmrit/display/controlPanelEditor/shape/DrawFrame.java#L355-L357

That if statement at the begining says if there isn't a shape selected, don't update the color.

So, is this a different process for choosing the color, absolutely.  Is there a new bug, absolutely not.

petecressman, 2017-11-03:
My error. Thank you for explaining how to use this new ColorChooser.  It is a fine improvement to Panel Editor. However, for Control Panel Editor it disables features that have been a part of CPE for several years and is inconsistent with its other ColorChoosers.

I find it a bit troublesome to have to click to open the ColorChooser window, then click to chose a color swatch, then click to close the window and set the color into the shape. The former code with the tabbed ColorChooser integrated into the edit dialog only requires one click to chose and view the color change on the shape.  When browsing for a color it is much more convenient to only use one click rather than three, as well as the alternating between two windows. 
Finally, I really miss the ability to tune the color choice with the other tabs.  For example, it's impossible to get a good orange color from the swatch tab.  I think it likely that other CPE users might agree with me.

CPE and PE have had different means of choosing colors for a long time.  I don't see any reason for enforcing them to be the same.

petecressman, 2017-11-03:
PR 4310 reverts the ColorChooser for shapes to the former chooser among its other changes.  I hope you find this acceptable.

silverailscolo, 2017-11-03:
@petecressman As just another CPE user, I don't agree this should be reverted to 'as it used to be'. Setting the color of objects is not a daily routine, where every click counts.
Indeed adding this improved and compact color picker also to CPE might attract new users recognize how to work with colors. It contains 5 types of color pickers, did you already try the other four?

pabender, 2017-11-03:
@silverailscolo the question really boils down to should those color picker tabs be displayed in the edit shape window ( the behavior prior to #4287) or should the be displayed ( as they are today in master ) in a dialog that appears after clicking the set button.

In both cases, the same tabs are displayed to actually change the color.

Personally, I am in favor of the consistency of behavior we've put in place in #4287.  Everywhere we select color that was changed in that PR causes the dialog window to appear.  ( There is still some work that needs to be done, primarily in the operations code, and in parts of the jmri.jmrit.display package that interact with operations ).

silverailscolo, 2017-11-04:
@pabender to me has clearly demonstrated in the course of his color chooser PR that in order to provide the richness in color model options within the interface, it is more elegant to present this as a 2 step solution. If we don't, I find the other editable subjects - like font, size, etc - get overwhelmed, drawing too much user attention to color.

petecressman, 2017-11-04:
OK. I've stated my preference for previewing color change combinations for Line and Fill directly on the shape without the extra clicks and extra window.  If I'm singular in this, fine, do as you wish.

pabender, 2017-11-04:
@rhwood, 

since you know these better than I do, do any of the user interface guidelines give us guidance here?  Do you have a preference?

@bobjacobsen, @KenC57 if you have a minute, could you provide some feedback as well.

geowar1, 2017-11-04:
IMHO: I like the original color menus but with the full [16 html](https://en.wikipedia.org/wiki/Web_colors#HTML_color_names) colors… plus a "Custom..." entry that would open the color picker.
I do like having the little box that shows the actual color thou.
(FYI: as a child (last week) I only ever had the smallest box of 8 crayons… 16 is a splurge! (And I haven't a clue what color is Fuchsia!) ;-)

pabender, 2017-11-04:
Sent from my iPad
> On Nov 4, 2017, at 2:23 PM, George Warner <notifications@github.com> wrote:
> 
> IMHO: I like the original color menus but with the full 16 html colors… plus a "Custom..." entry that would open the color picker.
> 
The colors were not initially selected because the names were HTML color names.  They were selected because they were Java color constants.  HTML does happen to use the same names, and defines a whole bunch more, but I am not sure the RGB values are the same.

While I did define other color selectors when I was building PR #4287, including a drop down list and a set of radio buttons, I don't really have much interest in going back to having color names appear in the user interface.

Paul

geowar1, 2017-11-04:
And I'm not particularly interested in dealing with the internationalization issues… So it's most likely going to stay the way you wrote it. ;-)
Note: Another alternate implementation would be to remove the "Set" button and just have clicking the color well bring up the color picker.

petecressman, 2017-11-04:
The color chooser for shapes in control panel editor has no bearing on any of the color choosers in Panel Editor.  The warrant internationalize branch includes 2 significant bug fixes.

pabender, 2017-11-04:
Sent from my iPad
> On Nov 4, 2017, at 5:16 PM, petecressman <notifications@github.com> wrote:
> The color chooser for shapes in control panel editor has no bearing on any of the color choosers in Panel Editor.
> 
But switching to a common color chooser means it impacts every part of the system.  It's a common widget, intended to be used everywhere we select  colors that appear on screen.  

I see no reason to use a color picker for things that don't appear on a screen ( or on paper ) and may not have easily understood translations to the screen ( or paper ).  The biggest example of this is the operations code for setting car colors.  If we were to start using those on screen, then we would have to rethink how we do that part of the operations code.
> The warrant internationalize branch includes 2 significant bug fixes.
> 
If you can split off the reversion of the color selection change, I'll merge it in.

Paul

rhwood, 2017-11-08:
The Java Look and Feel Design Guidelines offer no guidance on embedding a color picker in a dialog or elsewhere, but all examples (four) in that book do have the color picker in its own dialog.

The Windows, macOS/OS X, and GNOME (Linux) guidelines all state that a single consistent color picker dialog will be used within applications (and that this dialog will be the OS-provided dialog). Using the OS-provided dialog is not possible within the Java application context, but these guidelines do emphasis a need for consistency in UI both within a single application, and to the extent possible, with the OS. 

Sorry to be late responding.

pabender, 2017-11-08:
Thanks @rhwood

closed 2017-11-09 by pabender