JMRI/JMRI #4344: Change made to pom.xml disables Eclipse automatic partial builds

petecressman, 2017-10-29:
In order to compile any code change in Eclipse, a clean build must be done.  pom.xml is marked as having errors and opening the file shows the message "Cannot read lifecycle mapping for artifact"

rhwood, 2017-10-29:
If you right click on that error, what is the "Quick Fix"?

rhwood, 2017-10-29:
Right click on the JMRI project, go to "Run As" and select "Maven generate-sources". That will allow Maven to build what the Eclipse compiler can't that JMRI relies on.

petecressman, 2017-10-29:
Did Click on "Click for details", which shows a blank popup. Did "Run As" and select "Maven generate-sources" which ran a script. Closed and relaunched Eclipse. pom.xml still marked as having errors.  Same behavior - Eclipse cannot do partial builds.

rhwood, 2017-10-29:
What version of Eclipse are you using?

I am using Oxygen (4.7.1a), and after running "Maven generate-sources" and *not* quitting Eclipse, am having the error disappear and another complete, and longer build occurs that results in ~4192 warnings and 10 info messages.

petecressman, 2017-10-29:
! am using "Eclipse Java EE IDE for Web Developers"
Version: Neon.3 Release (4.6.3)
Build id: 20170314-1500
"Run As" now has no options so i cannot repeat the Maven thing. Seems to be totally screwed up.

rhwood, 2017-10-29:
I just downloaded and installed Eclipse Neon.3 (version cut and paste is:
```
Eclipse Java EE IDE for Web Developers.

Version: Neon.3 Release (4.6.3)
Build id: 20170314-1500
```

Run As menu is/is not populated depending on if "jmri" project is/is not selected. It might be better to right click the project and select Run-As from the pop up menu, since that ensures the project is selected.

petecressman, 2017-10-29:
Couldn't find any project name let alone one called "jmri".  However, when the Project menu had a "Close Project" item, "Run As" popup had the "Maven generate-sources" item.  Running the item again reported Success.  See output below:

[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building JMRI 4.9.6-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- build-helper-maven-plugin:3.0.0:timestamp-property (timestamp-property) @ jmri ---
[INFO] 
[INFO] --- jacoco-maven-plugin:0.7.9:prepare-agent (prepare-agent) @ jmri ---
[INFO] argLine set to -javaagent:C:\\Users\\SVD\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.7.9\\org.jacoco.agent-0.7.9-runtime.jar=destfile=C:\\Git\\JMRI\\target\\jacoco.exec -Xmx1536m
[INFO] 
[INFO] --- javacc-maven-plugin:2.6:jjtree-javacc (default) @ jmri ---
Java Compiler Compiler Version 5.0 (Tree Builder)
(type "jjtree" with no arguments for help)
Reading from file C:\Git\JMRI\java\src\jmri\jmris\simpleserver\JmriServerParser.jjt . . .
File "Node.java" does not exist.  Will create one.
File "SimpleNode.java" does not exist.  Will create one.
File "ASTcommand.java" does not exist.  Will create one.
File "ASTaddress.java" does not exist.  Will create one.
File "ASTpowercmd.java" does not exist.  Will create one.
File "ASTpower.java" does not exist.  Will create one.
File "ASTonoff.java" does not exist.  Will create one.
File "ASTturnoutcmd.java" does not exist.  Will create one.
File "ASTturnout.java" does not exist.  Will create one.
File "ASTturnoutdevice.java" does not exist.  Will create one.
File "ASTthrownclosed.java" does not exist.  Will create one.
File "ASTlightcmd.java" does not exist.  Will create one.
File "ASTlight.java" does not exist.  Will create one.
File "ASTlightdevice.java" does not exist.  Will create one.
File "ASTreportercmd.java" does not exist.  Will create one.
File "ASTreporter.java" does not exist.  Will create one.
File "ASTreporterdevice.java" does not exist.  Will create one.
File "ASTsensorcmd.java" does not exist.  Will create one.
File "ASTsensor.java" does not exist.  Will create one.
File "ASTsensordevice.java" does not exist.  Will create one.
File "ASToperationscmd.java" does not exist.  Will create one.
File "ASToperations.java" does not exist.  Will create one.
File "ASTtrains.java" does not exist.  Will create one.
File "ASTlocations.java" does not exist.  Will create one.
File "ASTterminate.java" does not exist.  Will create one.
File "ASTtraincmd.java" does not exist.  Will create one.
File "ASTattributelist.java" does not exist.  Will create one.
File "ASTattributename.java" does not exist.  Will create one.
File "ASTvalue.java" does not exist.  Will create one.
File "JmriServerParserTreeConstants.java" does not exist.  Will create one.
File "JmriServerParserVisitor.java" does not exist.  Will create one.
File "JJTJmriServerParserState.java" does not exist.  Will create one.
Annotated grammar generated successfully in C:\Git\JMRI\target\javacc-1509310433387\node\JmriServerParser.jj
Java Compiler Compiler Version 5.0 (Parser Generator)
(type "javacc" with no arguments for help)
Reading from file C:\Git\JMRI\target\javacc-1509310433387\node\JmriServerParser.jj . . .
File "TokenMgrError.java" does not exist.  Will create one.
File "ParseException.java" does not exist.  Will create one.
File "Token.java" does not exist.  Will create one.
File "SimpleCharStream.java" does not exist.  Will create one.
Parser generated successfully.
Java Compiler Compiler Version 5.0 (Tree Builder)
(type "jjtree" with no arguments for help)
Reading from file C:\Git\JMRI\java\src\jmri\jmris\srcp\SRCPParser.jjt . . .
File "Node.java" does not exist.  Will create one.
File "SimpleNode.java" does not exist.  Will create one.
File "ASThandshakecommand.java" does not exist.  Will create one.
File "ASTcommand.java" does not exist.  Will create one.
File "ASTserviceversion.java" does not exist.  Will create one.
File "ASTconnectionmode.java" does not exist.  Will create one.
File "ASTbus.java" does not exist.  Will create one.
File "ASTaddress.java" does not exist.  Will create one.
File "ASTvalue.java" does not exist.  Will create one.
[... 92 lines cut ...]
Annotated grammar generated successfully in C:\Git\JMRI\target\javacc-1509310436371\node\SRCPClientParser.jj
Java Compiler Compiler Version 5.0 (Parser Generator)
(type "javacc" with no arguments for help)
Reading from file C:\Git\JMRI\target\javacc-1509310436371\node\SRCPClientParser.jj . . .
File "TokenMgrError.java" does not exist.  Will create one.
File "ParseException.java" does not exist.  Will create one.
File "Token.java" does not exist.  Will create one.
File "SimpleCharStream.java" does not exist.  Will create one.
Parser generated successfully.
[INFO] Processed 3 grammars
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 28.494 s
[INFO] Finished at: 2017-10-29T13:53:57-07:00
[INFO] Final Memory: 12M/64M
[INFO] ------------------------------------------------------------------------

Eclipse again says: 

Errors exist in required project(s)
JMRI
Proceed with launch?

and pom.xml is marked as the offending file.

rhwood, 2017-11-02:
@petecressman Is this now working for you, or is this still an issue?

Can you include a screenshot of the Package Explorer if it is still an issue?

petecressman, 2017-11-03:
Attached are Package Explorer screenshots from Neon and Oxygen after pulling from master at 5PM pdst 11/2.
[image]
[image]

pom.xml on JMRI/JMRI master had additional lines 711 to 717
        <dependency>
            <groupId>info.cukes</groupId>
            <artifactId>cucumber-picocontainer</artifactId>
            <version>${cucumber.version}</version>
            <scope>test</scope>
</dependency>
which I inserted to the Oxygen repository and committed and pushed  The error was still marked on pom.xml.

I then cloned a completely new repository from JMRI/JMRI for Oxygen.  The error is still marked on pom.xml.  
Then did a Maven Build for PanelPro.  Compiled and ran cleanly.  The error is still marked on pom.xml.

rhwood, 2017-11-03:
@petecressman If you are finding that JMRI/master has code not in your working copy, pull the changes from JMRI/master using the "git pull" command (right click on project, Team -> Pull) instead of attempting to manually edit the files to match. Manually editing files to match will only cause conflicts and other later problems.

The only scenario I was able to cause this in was when the ```git``` command (```git.exe``` in Windows) is not on the command line. This will be fixed by #4367.

Do note that your project in some other ways does not match what is cloned when importing JMRI sources into Eclipse for the first time. A newly imported project has the name "jmri" (note lower case), not "JMRI", so you may have other, conflicting differences.

closed 2017-11-03 by rhwood

petecressman, 2017-11-03:
The images obscured most of the text that I wrote.  I did the compare with head AFTER I did the git pull! Seeing the error still marked I am grasping at straws for finding a solution.

petecressman, 2017-11-03:
Pulling #4367 into the Neon repository gave the following errors:

    ERROR: org.eclipse.equinox.p2.director code=0 To: org.eclipse.platform.feature.group [4.6.0,4.7.0)
  ERROR: org.eclipse.equinox.p2.director code=1 Cannot satisfy dependency:
    ERROR: org.eclipse.equinox.p2.director code=0 From: Eclipse IDE for Java Developers 4.7.1.20171009-0410 (epp.package.java 4.7.1.20171009-0410)
    ERROR: org.eclipse.equinox.p2.director code=0 To: org.eclipse.platform.feature.group [4.7.1.v20171009-0410]
  ERROR: org.eclipse.equinox.p2.director code=1 Cannot satisfy dependency:
    ERROR: org.eclipse.equinox.p2.director code=0 From: Eclipse Platform 4.6.3.v20170301-0400 (org.eclipse.platform.feature.group 4.6.3.v20170301-0400)
    ERROR: org.eclipse.equinox.p2.director code=0 To: org.eclipse.ant.core [3.4.100.v20160505-0642]
  ERROR: org.eclipse.equinox.p2.director code=1 Cannot satisfy dependency:
    ERROR: org.eclipse.equinox.p2.director code=0 From: Eclipse Platform 4.7.1.v20171009-0410 (org.eclipse.platform.feature.group 4.7.1.v20171009-0410)
    ERROR: org.eclipse.equinox.p2.director code=0 To: org.eclipse.ant.core [3.5.0.v20170509-2149]

Took 15 seconds.
There are failed tasks.
Press Back to choose different settings or Cancel to abort.

petecressman, 2017-11-03:
Pulling into the Oxygen repository flashed an error popup that i couldn't capture.  It mentioned an bug flagged as "wouldn't be fixed".  Couldn't get the reference number in time.  A second pull says "everything up to date"

Throughout all of this, pom.xml is marked as having errors and opening it has the message: "Cannot read lifecycle mapping for artifact org.apache.maven.plugin:mav (click for details)"

Clicking "(click for details)" shows a tiny blank popup.

Run "Run as" - "Maven generate-sources" executes with BUILD SUCCESS.

Run Run Configuration "Mavin Build" (for PanelPro)- shows popup with message "Errors exist in required projects: JMRI"

Click "Proceed with launch" compiles and runs PanelPro.
Console does show before the normal INFO entries:
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/GitO/JMRI/lib/slf4j-log4j12-1.7.25.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Users/SVD/.m2/repository/org/slf4j/slf4j-log4j12/1.7.25/slf4j-log4j12-1.7.25.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.Log4jLoggerFactory]

rhwood, 2017-11-04:
I have tested on three different computers, including on a newly installed operating system, and cannot reproduce these errors you are describing.

For reference, on the newly installed system (CentOS 7.4 GNOME Workstation), the only tools I installed were Eclipse Oxygen for Java SE and the Java 1.8 OpenJDK JDK before following the instructions at http://jmri.org/help/en/html/doc/Technical/Eclipse.shtml to download and build JMRI.

rhwood, 2017-11-04:
Now that I think about it, if you open the JMRI repos in a new workspace, do you get different behavior?

petecressman, 2017-11-04:
This must be something peculiar to my setup.  (thanks for all your testing).  I did the new repository download on Oxygen exactly as the instructions say in the link you referenced - including the Maven Build under run->run configurations-> JMRI PanelPro.  Could there be something wrong with my master at https://github.com/petecressman/JMRI ?

rhwood, 2017-11-04:
Did it work in the new download?

I won’t be able to look at your fork until Monday morning, but will look at it then.

Randall Wood

petecressman, 2017-11-04:
Oxygen reports the following problems:

Archive for required library: 'C:/Users/SVD/.m2/repository/info/cukes/cucumber-jvm-deps/1.0.5/cucumber-jvm-deps-1.0.5.jar' in project 'JMRI' cannot be read or is not a valid ZIP file	JMRI		Build path	Build Path Problem

Cannot read lifecycle mapping metadata for artifact org.apache.maven.plugins:maven-install-plugin:maven-plugin:2.4:runtime Cause: invalid END header (bad central directory offset)	pom.xml	/JMRI	line 1	Maven Project Build Lifecycle Mapping Problem

Project 'JMRI' is missing required source folder: 'target/generated-sources/javacc'	JMRI		Build path	Build Path Problem

Project 'JMRI' is missing required source folder: 'target/generated-sources/jjtree'	JMRI		Build path	Build Path Problem

I have the file
 'C:/GitO/JMRI/lib/cucumber-jvm-deps/1.0.5/cucumber-jvm-deps-1.0.5.jar'
There is a "quick fix" to configure build path, but I hesitate to use the dialog because i don't understand it. 
If I click "Apply and Close" am I generating the right build path?
[image]

rhwood, 2017-11-04:
> Oxygen reports the following problems:
> Archive for required library: 'C:/Users/SVD/.m2/repository/info/cukes/cucumber-jvm-deps/1.0.5/cucumber-jvm-deps-1.0.5.jar' in project 'JMRI' cannot be read or is not a valid ZIP file	JMRI	Build path	Build Path Problem
> 
Delete the directory "C:/Users/SVD/.m2/repository/info/cukes/cucumber-jvm-deps" This will force maven (embedded into Eclipse Neon and Oxygen) to re-download that package (and its metadata). This may fix that issue for you.
> Cannot read lifecycle mapping metadata for artifact org.apache.maven.plugins:maven-install-plugin:maven-plugin:2.4:runtime Cause: invalid END header (bad central directory offset)	pom.xml	/JMRI	line [...]
These two problems are because maven won’t continue until after that first problem is resolved. These directories contain classes that are generated by maven from other JMRI sources.
> I have the file
> 'C:/GitO/JMRI/lib/cucumber-jvm-deps/1.0.5/cucumber-jvm-deps-1.0.5.jar'
> There is a "quick fix" to configure build path, but I hesitate to use the dialog because i don't understand it.
> If I click "Apply and Close" am I generating the right build path?
> 
No. Maven needs to be able to complete its first build.

rhwood, 2017-11-06:
@petecressman I cloned the master branch of [your repo](https://github.com/petecressman/JMRI.git) and cannot reproduce these errors (I get the expected build error before running the "maven generate-sources" command from the Run As menu and no errors after that).

Can you try switching to a new workspace in Eclipse (File->Switch Workspace). In that dialog, you can create a new workspace, and after that you should be able to import your existing JMRI working copy (or clone a new working copy from GitHub). I suspect that your current workspace has something in it that is causing this error.

PR #4367 (2017-11-03): Fix Eclipse failure to build if git not on path
If the git command is not on the user's execution path, builds in Eclipse fail. This fixes that by allowing the metadata generator to substitute ```UNKNOWN``` for the git revision if the git command is not available or fails for other reasons.
Also changes the revision to be 7 characters in length to match both Github's short revision length and the length of ```UNKNOWN```.

Fixes #4344, at least on new clones of the JMRI code base from Github.
commit 59ed74f5 (2017-11-03): Merge pull request #4367 from rhwood/eclipse-maven

Fix Eclipse failure to build if git not on path.

Fixes #4344