JMRI/JMRI #4348: Frequent "MetadataTest.tearDown:24 » NullPointer" failures in Travis CI

bobjacobsen, 2017-10-29:
In the last 12 Travis CI runs I've looked at (6 pairs), there have been 5 failures with "MetadataTest.tearDown:24 » NullPointer" or similar.  Four had one occurrence, one have two occurrences.

pabender, 2017-10-29:
Sent from my iPhone

> On Oct 29, 2017, at 5:03 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> In the last 12 Travis CI runs I've looked at (6 pairs), there have been 5 failures with "MetadataTest.tearDown:24 » NullPointer" or similar. Four had one occurrence, one have two occurrences.
> 

I am working on addressing this as part on pr #4325.  

There is an exception generated by the web server that is causing a message with a null logging event message to be compared, and that is corrupting the message log.

pabender, 2017-10-30:
Closed via #4325

closed 2017-10-30 by pabender