JMRI/JMRI #4352: Enhancement: Add External System Connection for USB ==> DMX512 (Lighting controllers)

geowar1, 2017-10-30:
I've been using LED RGBW light strips under my layout to illuminate there when I'm working… and plan on using them over the layout to illuminate it. I'd like to be able to change intensity and color for dawn, daylight, dusk and night effects. While investigating I ran across DMX512 controllers; they're relatively inexpensive (less than $50 for usb to DMX512 dongle, DMX512 LED Light strip controller and several strands of lights). 

I was able to write some (libusb) code to find the USB==>DMX512 dongle and set the red, green and blue intensities independently on four strands of LED (12 channels). Pretty happy with it.

Unfortunately what they don't have is JMRI support… good thing I know a few JMRI developers… ;-)

So I started looking into implementing an external system connection for DMX512 (using the USB to DMX512 dongle).

So that's what this project is. ;-)

Looks like most of what I need is at: 
<http://jmri.sourceforge.net/help/en/html/doc/Technical/NewSystem.shtml>,
<http://jmri.sourceforge.net/help/en/html/doc/Technical/SystemStructure.shtml>
and <http://jmri.sourceforge.net/help/en/html/doc/Technical/MultiConnectionUpdate.shtml>

One of the things that is missing is an AbstractUSBPortController… and no, don't tell me that it's a serial device… Hardware wise it may be transferring things serially but it doesn't have a baud rate per say. (And shouldn't. [sorry, pet peeve])

I do see support for libusb and javax.usb. I know libusb well… (And USB/HID, I wrote the original HID Manager documentation (tech note) for Apple.)

Anyway, I'll push a WIP:PR soon… ('cause I suspect I'll need some hand holding.)

bobjacobsen, 2017-10-30:
Happy to help with this. 

I agree that some of the lower level stuff implementations (i.e. AbstractSerialPortController) don't really work well for non-serial connections.  Even PortController and AbstractPortController assumes you want to communicate via streams of characters.  

One approach might be to work around that by having a new implementation start with AbstractPortController, ignore the getOutputStream() and getInputStream() methods, and add methods appropriate for USB HID.  Once that's working, we can figure out how to refactor to new base classes.

Again, happy to help.

pabender, 2017-10-31:
Sent from my iPad

> On Oct 30, 2017, at 7:47 PM, Bob Jacobsen <notifications@github.com> wrote:
> One approach might be to work around that by having a new implementation start with AbstractPortController, ignore the getOutputStream() and getInputStream() methods, and add methods appropriate for USB HID. Once that's working, we can figure out how to refactor to new base classes.

The Raspberry Pi implementation might be a good example to start with, since it is directly accessing GPIO pins.

The RaspberryPiAdapter class extends AbstractPortConroller, it returns null for both input and output streams.  In this case, the connect and recover methods are empty too, but the connect method would be an appropriate place to do device initialization for  a USB device.

Of course there might be commands that are appropriate to send/receive to a USB device over a stream, even if it isn't a serial port.

Paul

geowar1, 2017-11-20:
Merged! I'd like to thank @bobjacobsen , @rhwood, @KenC57  and @pabender for their assistance, advice, code review, etc. … the end result was much better for it. Thank you sirs.

closed 2017-11-20 by geowar1