JMRI/JMRI #4355: Enhancement: LayoutEditor- Collect Options Menu Options

geowar1, 2017-10-31:
Currently the Options menu arrangement is somewhat… scattered?
I'd like to collect the options into groups with ether their own submenu or dialog box:

Layout Editor Options:
- Edit Mode
- ToolBar (+submenus)
- Allow Repositioning
- Allow Layout Control
- Use Direct Turnout Control
- Show Help Bar in Edit Mode
- Panel Scrollbars... (+submenus)
- Show Icon tooltips (+submenus)
- Enable Antialiasing (Smoother Lines)
- Highlight Selected Block
- Edit Title...
- Set Background Color
- Set Default Text Color
- Save Location and Size
- Hide Track Construction Lines

Layout Editor Add Options
- Add Background Image...
- Add Fast Clock
- Add Turntable
- Add Reporter Label...

Grid Options:
- Show Grid in Edit Mode
- Snap to Grid when Adding
- Snap to Grid when Moving
- Set Grid Sizes...
- (new) Set Grid Colors... (Primary & Secondary)

Track Options:
- Set Track Line Widths...
- Default Track Colors... (+submenus)
- Automatically Assign Blocks to Track

Turnout Options:
- Use Direct Turnout Control
- Allow Turnout Animation
- Turnout Options (+submenus)

Suggestions/feedback wanted and welcome.

pnbgit, 2017-10-31:
Which does raise the question of why add fast clock, turntable and reporter label are items on the options menu rather than radio buttons in the toolbar, so you would then place it where you want it with shift-left click as you do to add anything else?

geowar1, 2017-10-31:
Agreeded… I've been considering changing all the radio buttons into icon buttons on a palette that you can drag-n-drop or click (icon then editor view) to add... But that's a different enhancement… ;-)

pnbgit, 2017-11-01:
The suggestion has been made that some things should go into user preferences.  I think that anything that you frequently switch back and forth during development should remain on the options menu, with shortcuts keys if possible. You don't have to want to go down into Edit->Preferences just to turn grid snapping on/off, for example. 

Being able to turn scrollbars on and off quickly can also be useful, especially when working on a panel that is designed for a bigger screen than the one you are currently using.

Highlight selected block might be good as a tickbox on the toolbar, next to the block combo box, rather than in the menu.

Add turntable and reporter label should probably also be on the toolbar.

rhwood, 2017-11-02:
Suggest you rethink menus entirely along the lines of the Acorn, GIMP, or Inkscape applications.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

geowar1, 2017-11-03:
> Suggest you rethink menus entirely along the lines of the Acorn, GIMP, or Inkscape applications.

Can you be a little more specific? I've used all three of those apps (Acorn the most) and can't immediately think of anything about their menus that is significantly different than any of the other applications I use.

geowar1, 2017-11-03:
> Suggest you rethink menus entirely along the lines of the Acorn, GIMP, or Inkscape applications.

Can you be a little more specific? I've used all three of those apps (Acorn the most) and can't immediately think of anything their menu arrangement that is siffinigantly  different than all other macOS share

geowar1, 2017-11-03:
See PR #4368

rhwood, 2017-11-04:
What I mean is that those tools don’t have an “options” menu per drawing, but have what we have crammed into “options” spread throughout the edit/View/or other Tools menus. For example, the grid controls might be better in a “view” menu, since they affect the view.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

geowar1, 2017-11-04:
PR #4368 merged… and closed!

closed 2017-11-04 by geowar1