JMRI/JMRI #4356: Enhancement: Add option to create missing connection (as internal) on layout load error.

geowar1, 2017-10-31:
I find myself often loading 3rd party panels and usually have to wade thru a ton of "Missing Connection" dialogs… before I can go into preferences and create new internal connection(s) with the appropriate system prefixes; save and re-launch/re-load the layout.
I'd like to add an option (opt-in) to that dialog to create a missing connection (as internal) with the appropriate system prefix so that the panel can continue to load without errors.
(I can then open preferences and change any auto-added connections to "real" hardware if I choose.)

rhwood, 2017-10-31:
If you get the full profile instead of just the panel (since it is dependent on the profile), you wouldn’t have this problem. Profiles are, by design, portable across computers, so this would “just work” if the sent Profile was opened and the connection changed locally to a simulator.

geowar1, 2017-10-31:
I agree… but I find that even if I request the whole profile I don't always get it… Especially on the larger layouts that might have dozens of panels… they'll usually just send me the panel that's having a problem.

pnbgit, 2017-10-31:
I think this would be a useful enhancement too.   Question is, has JMRI already got past the point of no return for being able to create a connection and continue loading the panel by the time the error dialog is displayed?

rhwood, 2017-11-05:
It is too late. At the point where a panel is loaded any changes in connections would require that the application be restarted.

Furthermore, we deliberately want the various packages in JMRI to be loosely coupled; this loose coupling allows multiple applications, some not maintained by us, to be built on the JMRI code base.

We have provided mechanisms, including Profile exporting, to make sharing panels, and the other components they rely on (connections, scripts, images, sounds, other panels, and other resources) to be shared together. I highly recommend you encourage users to share those when they have problems instead of just the individual panel.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-11-18:
To echo @rhwood's comment:  Once you've gotten an error that some manager isn't present, it's too late to create that manager (and the structures that it relies on) in a reliable way.  We'd have to have a mechanism to back up in the file being read, but even that's complicated:  How far to back up?

Another way of achieving what you want might be for JMRI to write a _comment_ at the front of panel files with info about the context: A list of active system prefixes or something like that. That way, you could take a quick look at the file and see what's needed.  But you can do that now with something like:

```
grep -rA3 jmri.jmrix
```

to get an idea of what hardware types are referenced in the file. You'll have to infer the prefixes sometimes.

closed 2020-05-04 by geowar1