JMRI/JMRI #4366: NCE decoder update

mmosher5501, 2017-11-02:
[NCE_D13SR_TC.zip](https://github.com/JMRI/JMRI/files/1439497/NCE_D13SR_TC.zip)
update for version 3.8 and add new models

dheap, 2017-11-03:
@mmosher5501, the attached file appears to have been based on the code base prior to the merging of #4357 two days ago.

At least some of the work in this issue overlaps the changes made by @marcfour in #4357, so there are conflicts that need to be resolved.

For example, from current Head of master:
```
  <version author="Marc N Fournier" version="12" lastUpdated="20171031"/>
  <version author="Alain Le Marchand" version="11" lastUpdated="20140628"/>
```
But from the attached file:
```
  <version author="Michael Mosher" version="12" lastUpdated="20171102"/>
  <version author="Alain Le Marchand" version="11" lastUpdated="20140628"/>
```

Dave

mmosher5501, 2017-11-03:
I downloader Marc file, the ver comment states""  while the high ver has been changed to 38, but I can't see the NH decoders anywhere in the family tree.

My file adds: D13J,D13NH,D13NHP,D13W,D13WP,D16MTC,N12-NEM,N14K1,N14K2,NH15-44,NMP15 models listed on NCE web.
Looks like my D13NH should be D13NHJ

mmosher5501, 2017-11-03:
OK changed mine to V13 
added Marc's version & comment
changed  D13NH should be D13NHJ
[NCE_D13SR_TC.zip](https://github.com/JMRI/JMRI/files/1442247/NCE_D13SR_TC.zip)

PR #4409 (2017-11-12): NCE decoder update #4366
@mmosher5501 update for version 3.8 and add new models

Closes #4366
closed 2017-11-12 by bobjacobsen