JMRI/JMRI #4387: German Translation of Closed/Trown in

pischky, 2017-11-07:
I find the German Translations of Closed and Thrown (C/T) to "O" and "G" very irretating
in the "Configure Command Station"  dialog. The correct transalation for a position of a switch 
of Closed is "Gerade" and that of Thrown is "Abzweig". That should result in (G/A) I do not know what "O" should mean. But I would suggest that "C" and "T" are used in the German Translation because the German Translation of the Manuals used this terms. 

[AnleitungDigitrax_DCS50](http://zephyr.thwoditsch.de/AnleitungDigitrax_DCS50_V2.5.pdf)

silverailscolo, 2017-11-07:
I agree on the desired German terms. I can pick this up in a next translation PR.
@pischky if you find other terms or would like to discuss choices, feel free to enter them here or mail me via dcc2 at ebroerse dot nl The most important items are listed in our [Glossary](http://jmri.sourceforge.net/help/en/glossary/index.shtml) as an aid towards consistency.

silverailscolo, 2017-11-08:
@pischky can you confirm you are configuring an IntelliBox command station on LocoNet?
I have located the O/G characters but to clean up duplication between the LocoNet command station bundles, I suggest I also group them as a single pair of shared keys (for all translations).

pischky, 2017-11-08:
Thanks. I found the problem while doing a configuration of a DCS100 (Chief). I can also test on an IB-I.
I never used JMRI for that purpose. Will have to figure out about it...

silverailscolo, 2017-11-08:
Thanks. No big deal, as I can fix it for all LocoNet command station dialogs in one go.

bobjacobsen, 2017-11-08:
> 
> I have located the O/G characters but to clean up duplication between the LocoNet command station bundles, I suggest I also group them as a single pair of shared keys (for all translations).
> 

Before you group those, could you please check that their values are all the same?

I seem to recall that they were separated because the LocoNet-using and Lenz-using communities wanted to use different text.  But perhaps that was just for the full label, not the one-character abbreviation.

Bob

pischky, 2017-11-08:
Hello,

sorry I can not help here. I do not know about grouping.

I would not expect to the dialog that is working for the DCS100
will also work for an Intellibox. This are two different command
stations in my understanding.

Please let me know if I should test. I have DCS100 and IB-I here
on my desk.

Martin

Am 08.11.2017 um 13:26 schrieb Bob Jacobsen:
> Before you group those, could you please check that their values are all
> the same?
> 
> I seem to recall that they were separated because the LocoNet-using and
> Lenz-using communities wanted to use different text. But perhaps that
> was just for the full label, not the one-character abbreviation.
> 
> Bob

-- 
----------------------------------------------------------------
Martin Pischky, Oisseler Strasse 1C, 30559 Hannover, Germany
http://www.pischky.de mailto:martin@pischky.de
tel: +49 511 7000665 mobile: +49 151 11592222
----------------------------------------------------------------

silverailscolo, 2017-11-08:
Within the LocoNet command station properties are some identical items, perhaps copied from a first implementation. I don't intend to change their wording as presented, but put these items in one place.

pabender, 2017-11-08:
Sent from my iPad

> On Nov 8, 2017, at 7:26 AM, Bob Jacobsen <notifications@github.com> wrote:
> I seem to recall that they were separated because the LocoNet-using and Lenz-using communities wanted to use different text. But perhaps that was just for the full label, not the one-character abbreviation.

For single characters, Lenz uses "+" and "-" for turnouts, though I am not sure we currently have that as a single character translation in the code.

The initial split between Lenz and everything else, was just to be able to set the button label to show the + and -.  That was before we started really internationalizing our code. ( I want to say BCE uses this to show Normal and Reverse, but I could be wrong about that )

Paul

pischky, 2017-11-08:
Hello,

Note: my initial intention was only to improve the command station
configuration dialog (here Chief/DCS100). For turnouts (Weichen) there
may be different terms required.

Martin

Am 08.11.2017 um 15:43 schrieb Paul Bender:
> For single characters, Lenz uses "+" and "-" for turnouts, though I am
> not sure we currently have that as a single character translation in the
> code.

-- 
----------------------------------------------------------------
Martin Pischky, Oisseler Strasse 1C, 30559 Hannover, Germany
http://www.pischky.de mailto:martin@pischky.de
tel: +49 511 7000665 mobile: +49 151 11592222
----------------------------------------------------------------

silverailscolo, 2017-11-08:
Please assign to me, here's my work thus far: [eb-german-jmrix-cs](https://github.com/silverailscolo/JMRI/tree/eb-german-jmrix-cs/java/src/jmri/jmrix/loconet/cmdstnconfig) that changes the German letters to G/A.
Will also fix the French character encoding and add a Dutch language set.

silverailscolo, 2017-11-15:
Fix is included in PR #4396, German OpSw letters changed to G/A (for C/T)

bobjacobsen, 2017-11-19:
Will be in next test release.

closed 2017-11-19 by bobjacobsen