JMRI/JMRI #4388: Com in use after exiting Decoder Pro or Panel Pro after upgrade to 4.8

CritterRhode, 2017-11-08:
Upgraded from 4.6 to 4.8 and now every time I close the program and then try and reopen it, I get an error of the Com port is in use. Even after doing a normal Windows 10 shutdown and boot up, the error still happens. That is with no extra power to the Digitrax PR3 and also trying having it unplugged on boot. I have to use a shortcut command of "C:\Windows\System32\shutdown.exe /s /f /t 0" to get this to reset. 

Digitrax Pr3 connected to Command station DCS200
 Windows 10 Home 64bit with Java  Version 8 Update 151 Build 1.8.0_151-b12
Ryzen 7 1700 
16 Gigabytes of ram 
Norton Security

CritterRhode, 2017-11-13:
Was able to get a LocNet LocoBuffer-USB to test with and that works on the same USB port and cable. It came up on the same com port of COM 3. Have not had a problem with it. But putting the Digitrax PR3 back on gives the same com error.

devel-bobm, 2017-11-13:
If JMRI thinks it is actually talking to a PR3 device, but that PR3's USB cable is not connected, then it often means that JMRI is instead talking to a COM port which is assigned to something other than the PR3.  Often this means that JMRI is talking to some internal device that acts as a serial port, like some  "security" devices such as fingreprint detectors or RFID readers or ...

With "normal" USB devices, the device provides a unique serial number as part of "enumeration".  The operating system can reserve a particular COM port number for use by the device which has that specific serial number.  And the operating system will use that COM port number whenever that device is plugged in, regardless of where it is plugged in on that machine, or even down a chain of USB hubs.

But the PR3 is a "non-quite-conforming" USB device.  It does not provide a unique serial number to the host computer; it always reports serial number "0".  The host O/S (i.e. Windows) cannot allocate a "fixed" COM port to a USB device which reports serial number "0".  That means that the COM port used with the PR3 is likely to change around a lot, especially if you change the USB port to which it is connected!

Net result: it is very likely that Windows has assigned something other than COM3 for use with your PR3.  I have personally seen a PR3 assigned to a COM port as high as COM50!

I recommend that you use Windows Device Manager to determine exactly which COM port it is using for the PR3 when it is plugged in to a live USB jack.  If Windows Device Manager reports COM3 then you are experiencing a failure-mode that I don't recognize, and wiser heads than mine will have to follow up.

If Windows Device Manager reports that the PR3 is using a different COM port than COM3, then you will need to reconfigure JMRI to match.  And then try to avoid moving the PR3 to another USB port.

Please update this "issue" to reflect your debugging results.

closed 2021-04-14 by stale[bot]