JMRI/JMRI #4391: Over site causing 306 error when writing in Opa mode

Walter71342JMRI, 2017-11-08:
In the preferences>roster is a check box to Read CV after a write. "If possible", If that is checked JMRI tries to read on the Main.

Walter71342JMRI, 2017-11-09:
The statement for box is "When possible, confirm CV writes by following then with a read."  That implies that JMRI will check that a read will work. On the Main , in Ops mode, unless special hardware is installed it will not and give the 306 error. That makes a user think that the writes failed. Most command stations send the ack for the write when the decoder acks the command from the command station on a write. The only DCS I know of that does not wait for the decoder ack is NCE.  I can understand why this may be nice on NCE systems.

Because the state is "When possible" JMRI should check that reads should work. 1,Could do a test read and post a message if it fails.  2. Some how verify that the required hardware is installed.

Or remove this option and let the user read the CV if they want to.

Walt

bobjacobsen, 2017-11-09:
Your real issue here is that JMRI can't figure out in advance whether an ops-mode read on a Digitrax system will work or not. If you can figure out how to do that, then we can implement it. 

Until then, only people with transponding installed & operating on that specific decoder should do ops-mode writes with this option selected.

Perhaps more documentation on that would be helpful?

closed 2017-11-09 by bobjacobsen

Walter71342JMRI, 2017-11-09:
JMRI could try a read in Ops mode, if it fails don't do it.

Walter71342JMRI, 2017-11-09:
I do not have an NCE system but it looks like the same thing would happen there on an OPS mode write.
If JMRI can not figure out "When Possible" can that verbiage be deleted?

Walter71342JMRI, 2017-11-09:
Bob I added some comments to the issue thread.
Walt