JMRI/JMRI #4392: Color Selection Widget

pabender, 2017-11-09:
We've been working over the last few weeks on color selection methods, resulting in PR #4287 and a discussion in #4342 about how the new color chooser works.

There are two situations where we trigger color choosing right now:

1. We have a color choice through a menu item
2. We have a color choice through a widget on the screen. 

Prior to #4287, item the menu items were almost exclusively done using a set of radio buttons in a menu.  After #4287, the changed menu items now trigger a modal color chooser dialog using the standard Java JColorChooser object.  I don't think there's any question as to whether or not that is the right decision.

For the on screen widget, there's been some question ( see #4342 ) about how the change impacted the one or two places where there was a JColorChooser embedded into a window already.

@rhwood's comments on #4342 indicate the various style guidelines state that we should be consistent in how we choose colors, and the examples all trigger the default system dialog box.

@petecressman isn't happy with the way that makes the ControlPanelEditor shapes dialog (which was a place that previously had a JColorChooser) but embeding the full JColorChooser (which is what @petecressman prefers) doesn't work for everywhere we choose colors.

See:
[image]

The color chooser currently in place looks basically like this (@geowar1 fixed a couple of cosmetic things since this screenshot was created):

[image]

I've also created a radio button color chooser and a drop down color choosers, but we decide while working on #4287 those were too limiting.  (these two color choosers mimic the two previously used common methods of color selection).

@gewar1 has proposed  a dropdown box with an option for "other".  and @rhwood has proposed offline the use of a splitbutton, which is a basically a cross between what we have now and a dropdown menu.

I'm not married to the current color selection method, but I do want us to have a consistent method of selecting colors across the program.  Once we come to some kind of a consensus, I'm happy to make the necessary changes to make it happen.

rhwood, 2017-11-09:
One possibility is:

[image]

petecressman, 2017-11-09:
I don't understand why the way CPE uses a color chooser should be imposed on PE.   If folks want choosing color on a PE object to be nothing but color tabs and a stock preview panel and other folks want an integrated color chooser that previews all editable attributes on the object itself as done on CPE - then why can't that be? I do think it would be an improvement to CPE to tab its attribute editor as is done in PE's property menu.  (This is something that has been on my TODO list for some time)

I do find it a bit absurd when color changes to a font are previewed as background on the stock preview panel - likewise for border colors.  Not exactly WYSIWYG.

pabender, 2017-11-09:
On 11/09/2017 02:17 PM, petecressman wrote:
> I don't understand why the way CPE uses a color chooser should be
> imposed on PE.

Let's start out with one statement just to make sure everyone is on the 
same page:

This isn't a Layout Editor vs Control Panel Editor vs Panel Editor 
thing.  This is about providing a single common widget for choosing 
displayed colors anywhere it displayed within a GUI window.  In that 
sense, it is no different than the widgets we have for entering the 
names for named bean objects, and like those, it is a common widget to 
be used throughout the code.

(We are also applying this to places like operations, for example).

> If folks want choosing color on a PE object to be nothing
> but color tabs and a stock preview panel and other folks want an
> integrated color chooser that previews all editable attributes on the
> object itself as done on CPE - then why can't that be?

OK, so the current color choice method can't allow that preview you 
want.  It wasn't designed to work that way.

That said, we can improve on this.

If we implement the color chooser that @rhwood suggested, the preview 
will function, so long as you choose one of the colors in the menu on 
the splitbutton.

If you want to pick some other color, then you click the button and the 
full color chooser would appear.  Once you pick a color, it would go 
into the recently used colors list, and you wouldn't have to open the 
full color chooser again to get the same color.

There would be a minimum of 13 colors (the named constants in 
java.awt.Color) and a maximum of 20-25 colors (the defaults + 8 or so 
user selected colors) available without opening the color chooser dialog.

> I do find it a bit absurd when color changes to a font are previewed as
> background on the stock preview panel - likewise for border colors. Not
> exactly WYSIWYG.

In the dialog, we don't have a lot of control over the associated 
preview, because you can't replace the preview panel.

In order to have a more widespread implementation of live preview in 
places where it is displayed, we need to add color change listeners in 
many places.  The fact that we can even talk about doing this is an 
improvement over the old code, which just created menus using a set of 
text strings in most cases.

pabender, 2017-12-07:
#4519 brings up that we need to be able to handle background colors as transparent.  Whatever new color selection widget we use needs to conditionally accommodate that.

Fixing that in general may require saving the alpha values for the selected colors.

geowar1, 2017-12-11:
May I upvote @rhwood UI suggestion? I was considering writing a custom color picker panel for the "standard" colors… but his suggestion would work equally well.

pabender, 2017-12-11:
@geowar1 that's the direction I am headed.  I have to get grades submitted and then I plan on working on that.

dsand47, 2018-06-10:
I don't know that status of any work in this area, but I decided to try creating a variation on the @rhwood  suggestion.

[image]

I decided to modify ButtonSwatchColorChooserPanel.  The recent list is a static ArrayList.  A static method is available for processes to add colors outside of the custom chooser panel, such as direct JColorChooser.showDialog calls and xml loading processes.

If this approach is acceptable, I will continue development.

Dave Sand

geowar1, 2018-06-10:
Huge fan! I considered doing this back when the “standard” colors were killed off (with good reason). Hopefully this swatch can be added to the standard(?) color picker (there are cases where I do prefer the system ones).

pabender, 2018-06-10:
My biggest objection is the names of colors being present.  When we started all this, the problem was with color names not being properly internationalized, and reintroducing the names is a step in the wrong direction.

As a secondary concern, I believe thecwidget used needs to allow addition of new colors directly.

pabender, 2018-06-10:
Thinking about this again... my objection to the color names being present is not an issue so long as the color names are not stored in data files (it has to be numeric values).

Not being able to select the color directly through the widget is actually a regression.  We already have a drop down selection list widget, which we decided was too restrictive since it didn't directly allow the user to select other colors.

dsand47, 2018-06-10:
If the color button is clicked, the standard Java color selector is opened.  The arrow opens the recent/standard pop up menu.

The original design used a split button.  Maybe that helps with recognizing that there are actually two buttons.  

The names for the standard colors are translated if available and are only used on the panel.  A color object is returned to the calling application.

[image]

I have not finished the menu bundle messages.

pabender, 2018-06-10:
The splitbutton lets you see that there is a single widget, which I really do think is what we want.  Make sure you're adding this as a new class (call it SplitButtonColorChooserPanel) and we can adjust the UI later if we decide that's required. 

You are using jmri.util.ColorUtil.colorToName for the color to name translation?

dsand47, 2018-06-10:
No.  **colorToName** is private.  **colorToColorName** returns names that start with a lower case character and cannot be used as Bundle.getMessage keys since the current keys start with an upper case character.  It is possible to upper case the first character.  Or duplicate the entries in NamedBeanBundle.properties.

Or add a method to ColorUtil which returns a translated color name.  **_colorToLocalizedName_**?

pabender, 2018-06-10:
Yes, adding another method would be appropriate.  

The existing methods are intended to handle stored color values in the data files.

dsand47, 2018-06-11:
As I was working on the split button color chooser, I realized that it is a special case.  There are a lot of other color selections that use the Java color chooser directly.  They don't use an intermediate panel like the block color editor.

To me, the real solution is to add a JMRI custom panel the standard Java chooser.  This way there is a common process for color selection.  

[image]

The formatting needs a lot of work, but the concept works.  The custom panel is first in the list and contains the standard colors and the recently used colors.  For most people, this is all they will need.  The other Java supplied choosers are available by clicking on the appropriate tab.

The current class name for this approach is **_JmriColorChooser_**.

pabender, 2018-06-11:
Sent from my iPad

> On Jun 11, 2018, at 4:28 PM, Dave Sand <notifications@github.com> wrote:
> As I was working on the split button color chooser, I realized that it is a special case. There are a lot of other color selections that use the Java color chooser directly. They don't use an intermediate panel like the block color editor.
> 
Right, the code changes I made last fall just set the chooser panel in the standard jcolorchooser.  The button swatch color chooser panel is designed to be a compact color chooser panel.
> To me, the real solution is to add a JMRI custom panel the standard Java chooser. This way there is a common process for color selection.
>
>
> The formatting needs a lot of work, but the concept works. [...]
While I generally agree that having this in one location is a good idea, remember that this doesn’t work universally.

As you can see in the first post to this issue, the original idea was to just show a standard color chooser ( with the tabs ) everywhere we needed to select colors.  That was fine for some locations, but for other locations, such as the one shown, that threw the whole UI off to the point of making it unusable, that was why I went towards the more compact color chooser panels in the first place.

petecressman, 2018-06-12:
Adding a new tab to JColorChooser with pre-selected colors is a great idea. I look forward to using it.

dsand47, 2018-06-13:
The current plan provides 3 procedures for selecting colors.  The standard Java Color Chooser has been extended to include a custom JMRI panel that contains the 13 standard Java colors along with the recently used colors in the current panel configuration.  This makes it easy for people to re-select colors that they have already used.

### Direct Call

This is designed to be used by menu based color setting, such as LE **Set Background Color...**.  An extended dialog color chooser is displayed.

[image]

### Compact Chooser

This displays a small chooser consisting of a color swatch with the current color and a Set button.  When the Set button is pressed, the full chooser dialog (above) is displayed.  This example is the LE track Edit Track color options.

[image]

### Embedded Chooser

This option provides the extended Chooser panel that can be embedded but the application is responsible for updating the recent color list.

[image]

If anyone is aware of any other scenarios, please let me know.

pabender, 2018-06-13:
Sent from my iPhone

> On Jun 13, 2018, at 5:47 PM, Dave Sand <notifications@github.com> wrote:
> Compact Chooser
> 
> This displays a small chooser consisting of a color swatch with the current color and a Set button. When the Set button is pressed, the full chooser dialog (above) is displayed. This example is the LE track Edit Track color options.
> 
> 
> 

Changing this is actually the widget this issue was started to change.  There have been objections raised to how this looks, so not replacing it with something else doesn't solve this issue. ( I wrote this, and I don't mind it, but others do).

pabender, 2018-06-20:
Here's my current effort towards replacing the ButtonSwatchColorChooser with SplitButtonColorChooser

This is what the Edit Block window's layout editor tab looks like, without the popup:

[image]

And this is what it looks like with the popup (sorry, for the whole screen shot, I was doing this quickly..)

[image]

This is available now in my splitbuttoncolorchooser branch ( https://github.com/pabender/JMRI/tree/splitbuttoncolorchooser )

geowar1, 2018-06-20:
Is there a way to get the full picker if we don’t want one of the presets (or recents)?

pabender, 2018-06-20:
@geowar1 you only get the popup menu if you click on the arrow.  If you click anywhere else on the button, it pulls up the full color picker.

And notice that unlike the ButtonSwatchColorPicker, the swatch is actually an icon on on the button, not a separate element.

geowar1, 2018-06-20:
Excellent; that’s what I needed to know. ;-)

dsand47, 2018-06-20:
I did a quick integration of my project.

[image]

Since there aren't names for all possible colors, the recent list shows the RGB values.  This helps people identify closely matched colors.

ColorUtil has a colorToLocalizedName method for standard names.  I changed ColorListPopupMenu to use the localization method..

pabender, 2018-06-20:
@dsand47 thanks.

If you can push that up to your repository and then do a pull request against mine, I will merge that in and do a pull request against master for after the next production release.

dsand47, 2018-06-23:
@pabender I believe I have done the pull request properly.

pabender, 2018-10-01:
closed via #5502

closed 2018-10-01 by pabender