JMRI/JMRI #4395: Using a branch for parallel development

bobjacobsen, 2017-11-09:
Over the past couple months, we’ve had rapid, overlapping development around the LayoutEditor.  This is _great_, because it’s moving that part of the code forward rapidly, and sometimes that kind of enthusiastic work gets complicated.  

At the same time, it’s been hard to find a temporary pause in the the progress of the master branch were we could build a test release.  We need to be able to do that.

So I’d like to propose something that should be the best of both worlds.  We move development on `LayoutEditor` from the `master` branch to a `LayoutEditor-development` branch.  People will still be able to make PRs to that, we’ll have Jenkins build installers so that people can easily test it, and it can progress in whatever way people want.  Periodically, when it’s at a stable point, a PR can merge `LayoutEditor-development` onto `master`.  

Meanwhile, our usual run of small & independent changes can still accrue in the usual way on `master`. (We can also merge `master` onto `LayoutEditor-development` if some development on `master` is needed for LayoutEditor work)

If we decide to do this, there’s a bit of setup necessary which I'm happy to do, but then the rest is pretty straight-forward:

*) When creating a PR for LayoutEditor work, select that `LayoutEditor-development` branch instead of `master`

*) If you’re creating a new branch for a new bit of development, create it from `LayoutEditor-development` instead of `master`

That’s it!

Thoughts, please?

bobjacobsen, 2017-11-09:
> 
> Just to check:
> for development on other (smaller, non-LE) stuff, your final bullet point
> *) If you’re creating a new branch for a new bit of development, create it from LayoutEditor-development instead of master
> tells me to branch not from master? Would then no new GitHub development branch be created off master, or is there a third sort of development I'm missing?

Sorry, I want clear. 

If starting a new LayoutEditor project, start from that LayoutEditor-development branch. 

For other things, that one might want to merge to master separately, start from master. 

It’s not a disaster if one starts from master and ends up on the LE-dev branch: that’s effectively just pulling in master to the LE-dev branch.

bobjacobsen, 2017-11-10:
The LayoutEditor-development branch now exists.  If you're creating a new PR for work on LayoutEditor, please create it against that branch:

[image]

geowar1, 2017-11-10:
I don't see any value added for LE having it's own development branch. It will just add complexity to an already overly complicated development process. How will the new LE-developement branch keep up to date on changes to the master? How will changes to the LE-development branch get merged into the master branch? (How and who?) Do I branch from LE-dev then PR to merge to LE-dev and then PR again to merge to master? (Ouch!) If there's a bug fix to master do I submit a PR to get it into LE-dev?
I think that at some pre-determned point in time the master top-of-tree is marked as the test-release candidate. Any PR submitted after that would have to justify it being part of the release. Otherwise it goes into the next release. Features? nope. Enhancement? nope. Bugfixes for bug found while testing the test client? perhaps.

bobjacobsen, 2017-11-10:
(replying is a different order)

> How will the new LE-developement branch keep up to date on changes to the master?

```
git checkout LayoutEditor-development
git merge master
```

> How will changes to the LE-development branch get merged into the master branch? (How and who?)

That's a PR from the LayoutEditor-development` branch to `master`, same as any other branch is merged.

The difference is that people can make multiple changes to the `LayoutEditor-development` branch, test it, and in general continue to develop _without_ having to put it on the `master` branch.

>  If there's a bug fix to master do I submit a PR to get it into LE-dev?

Not sure I understand the question.

- If it's a bug that's been released, yes, please make a PR to `master`.
- If it's a bug in the LayoutEditor development (of which we've had quite a few recently), and this process has enabled that to be discovered before it caused a problem by being merged to master, then make it on the `LayoutEditor-development` branch.

More generally, fix bugs where they are, but try to _keep_ _them_ _off_ `master`.

> I think that at some pre-determned point in time the master top-of-tree is marked as the test-release candidate. Any PR submitted after that would have to justify it being part of the release. [...]

Since we've been doing JMRI releases for more than a dozen years, this has actually been thought about a little bit.  

Until recently, the head of `master` (and it's predecessors in SVN and CVS) was pretty stable:  Only complete chunks were added there.  PRs were stable and it was comparatively rare that commits were added to PRs after they'd been created; about the only reason one saw that was because of a mistake in committing or not running a check that was then found by the CI tests.

At any point, the head of master was 90% or more likely to be releasable.  A quick look at the last two years finds only a couple of examples of post-initial-tag changes to the test releases.

> I don't see any value added for LE having it's own development branch. It will just add complexity to an already overly complicated development process. 

The recent change is that development is going fast and in parallel.  I have no complaint about that. But the fact is that we're seeing bugs introduced onto master.  We need a way to keep the development fast, but we also need a way to keep `master` stable.  

There alternatives to having a branch in the JMRI/JMRI repository.

For example, we could start running Jenkins against a branch in individual's GitHub repositories to create installers, and request that they have people test those installers for bugs and completeness of changes before making PRs to the main repository.  I work on a couple projects that do that.

Having LayoutEditor work concentrated on a branch, so the people working on it can work as they like, merging when they reach a good point but not interfering with releasability of `master` until then, seems like a reasonable approach to me.  I'd certainly be happy to hear of a better one that meets the goal.

But until we do have a way to meet the goal, I think the @JMRI/maintainers should be very hesitant to merge any large changes to `master`.  We're only about two weeks away from starting the process of stabilizing for the pre-holiday production release, and we can't start another 6-week flail to get to stability.

bobjacobsen, 2017-11-10:
Another reason:

PR #4386 has been under development for so long, it's now got conflicts with master.  That's a pain, and it doesn't have to happen.  It's been delayed because it's not clear that it's ready to merge to `master`, but that's not a consideration if it's being merged to some other branch, i.e. `LayoutEditor-development`.  It could have been merged so that the conflicts would naturally have been resolved as work proceeds.

geowar1, 2017-11-11:
On Nov 10, 2017, at 2:43 PM, Bob Jacobsen <notifications@github.com> wrote:
> Another reason:
> PR #4386 <https://github.com/JMRI/JMRI/pull/4386> has been under development for so long, it's now got conflicts with master. That's a pain, and it doesn't have to happen. It's been [...]

It was “under development” for under 4 hours. It was ready to merge when it was submitted as my 1st comment said:

(If passes checks then Merge After Next Test Release (4.9.5?))
The only reason I didn’t ask for an immediate merge is because I didn’t want it to hold up the test release.

At no point in time did I indicate that it was WIP. All I expected was for it to get the "After Next Production Release” tag.

All jmri developers already fork master for our development efforts… why doesn’t the test release just fork off instead?

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

pabender, 2017-11-12:
@geowar1 

To address a couple of your questions/comments:

> All jmri developers already fork master for our development efforts… why doesn’t the test release just fork off instead?

We do branch off master to build test releases, but when master has an issue, that gets copied to the branch as well.

>  don't see any value added for LE having it's own development branch. It will just add complexity to an already overly complicated development process.

Here's the value for the project as a whole... 

We have known for years that the entire jmri.jmrit.display.layoutEditor package is a fragile, under tested piece of code.  This remains true today ( According to the latest jenkins/jacoco build, http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/1129/jacoco/ , that package by itself has 24000 untested lines of code, which is more than 10% of the total lines of untested code (approaximately 220000 lines).

Because of the lack of tests, we don't have a high level of confidence that any work done on the layoutEditor code won't break something (intentionally or not) that the tests don't cover.  That lack of confidence manifested itself in the bugs that delayed the release of 4.9.5 by 6 weeks.

So there are really two things we should be doing:

1. We make the code more maintainable, which includes both cleanup/refactoring and adding additional tests.
2. We develop the code on a separate branch, where we build packages that users can download and test, and we encourage users to download that code. 

The tests are key in any case, because that will (eventually) give us the confidence to merge things directly into master without first having users try it from another branch.

bobjacobsen, 2017-11-26:
See "LayoutEditor development" project and `LayoutEditor-development` branch.

closed 2017-11-26 by bobjacobsen