JMRI/JMRI #4400: White Icons do not display in Preview pane in Add Items.

CCatania01, 2017-11-10:
In CPE, Add Items, the Preview pane does not display white icons.  The selected icon object is in the window (invisible) and can be dragged to the panel.  Tested with USS/track/turnout gifs on standard LED display and MacBookPro.

CCatania01, 2017-11-12:
I did some investigation and testing on the issue.
The background color of the Preview pane is the set panel background color.  

The feature is fine for panels with user specified background colors.  However, when placing pre-defined white icons or images onto a blank CPE panel, the Preview pane is white and white images do not show.  

Background Icons, like the CTC machine column slices, are usually placed on a blank white panel, and completely cover the viewable window area, so the background color does matter.

Manually changing the background color to something other than white solves the problem.

Documenting the behavior might be a reasonable fix., or checking for a white background and setting the Preview pane to light gray.

bobjacobsen, 2017-11-12:
Some apps use a gray-white checkerboard pattern for the background in cases like this.  GraphicConverter uses one that's keyed to the size of the icon, with the checkerboard squares about 1/3 the size of the icon.

silverailscolo, 2017-11-15:
The following screen shot (from 4.9.5 on Windows 7) shows the issue on the left:
[image]
As this GUI basically has a light gray base, white line still show. New suggestion is simulated on the right hand side.
The user option to change the background color could be copied from the Icon pane (click Edit Icon to open it in CPE) using 3 radio buttons:
[image]
In my legacy PS11, transparency behind opaque pixels is displayed as described (the squares don't resize, more squares are added on Zoom In):
[image]

rhwood, 2017-11-15:
Suggest background on icon selector be in combobox, not radio buttons and include both black a checkerboard pattern as additional options.

pabender, 2017-11-15:
I think the checkerboard is easiest to understand from a users perspective.

This is commonly used in other applications.

CCatania01, 2017-11-15:
I agree with Randall's suggestion.  The selections could be: current pane background (the default), Black, Gray, and Checkerboard.

silverailscolo, 2017-11-15:
Just had a look in the code, but it might take me some time to think of the right place to add this.
For the short term, PR #4424 adds a note to the help as suggested (copied from Palette Javadoc):
"The icons are displayed on the background of the last editor to call the Item Palette."

rhwood, 2017-11-15:
There was a recent PR that changed the palettes from being per-CPE panel to having a single palette. You might to find and revert that to ensure the correct panel background is used in the palette.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

silverailscolo, 2017-11-26:
In CPE the > Add Items pane will look like:
[image]
with the "View on:" combo box allowing a Preview background choice of:
[image]
The first option is not applicable to Catalog, which can be made to look like:
[image]
Expect to finish a PR for this next week (see [branch](https://github.com/silverailscolo/JMRI/tree/eb-catalog-i18n)).

CCatania01, 2017-11-27:
Egbert,
Looks great.  Thank you for working on this.
Do I close this after the merge, or do you?
Chuck Catania

closed 2017-12-05 by bobjacobsen