JMRI/JMRI #4401: New decoder definition for ESU Lokpilot (older ESU Lokpilot decoder made in 2002)

alain355, 2017-11-10:
Hello dear Team

I created the included profile for the first ESU Lokpilot, made in 2002. Quite basic decoder with BEMF, Slow speed, 4 outputs, short and long address, no MU CV, no speed table, no Function re-mapping: F0f to output A, F0r to B, F1 to Aux1 and F2 to Aux2.

I added it to the existing "ESU LokPilot Silent BEMF decoders" Family, as it is detected by my Lokprogrammer V2.7.9 as a Lokpilot , but it also supports Marklin Delta.

Btw, I've attached the ESU English Manual.

Validated and tested OK using JMRI 4.9.5 on an ECoS Command station ;)

Thanks to include it in next release.

Cheers from Paris

Alain

[ESU_LokPilot_V1.zip](https://github.com/JMRI/JMRI/files/1462117/ESU_LokPilot_V1.zip)

dheap, 2017-11-11:
@alain355 Thanks very much for this.

I'll check the model name against the ESU naming conventions before merging this one. I've been bringing all ESU Model names exactly into alignment with ESU's model designations.

Does LokProgrammer V4.6.1 recognise the decoder? If so, exactly what model name does it report?

Exactly what model name does Lokprogrammer V2.7.9 report?

Dave

dheap, 2017-11-11:
@alain355 Could you please try Extended Decoder Information against this decoder in both LokProgrammer V2.7.9 and LokProgrammer V4.6.1 and report back the results (copy and paste)? 

V2.7.9 LokPilot Models
[image]

V4.6.1 Legacy Models
[image]

alain355, 2017-11-11:
Hi DaveSorry I made a mistake in my previous answera few mn ago: it was the Lokprogrammer HW/rev info). See the relevant infos in the picture bellow.

Decoder  is detected a a Lokpilot with 2.7.9 and as an error with 4.6.1, see attached screenshots, If you look at the CVs found by LP2.7.9 it's what I've defined in the xml file I've sent (the first 2 CVs are out of screen).With 2.7.9

 With LP4.6.1: not detected

Alain & Francine Carasso
acarasso_fr@yahoo.fr

 Envoyé le : Samedi 11 novembre 2017 13h51

alain355, 2017-11-11:
Strange, my 1st answer sent 30 mn before from my phone is not showing on the trail: here is what I wrote:
>> Sent at 1400 Central European Timezone
Hi Dave
As it's a very old one (see the ESU user's guide: january 2002), Lokpro 4.6.1 does not know it. With 2.7.9 it shows as a lokpilot, with very fews CVs (no consist CV, no function remapping, no speed curve).
With JMRI, I tried lokpilot DCC first as shown by the Lokpro, (read all CVs), and nearly 3/4 of the CVs were at 0 and unwritable except the few found by Lokpro v 2 7.9. Thus I searched my hard drive and found the old ESU lokpilot manual I've attached in the zip file. Then I created the XML using some previous ESU definitions.

alain355, 2017-11-11:
Screenshots not showing in my 1st answer, thus adding them again
[image]
[image]

dheap, 2017-11-11:
@alain355 Thanks for that. It looks like V2.7.9 knew it as the original "LokPilot". The ESU Downloads area has the same manual as you have in the [Former Products](http://www.esu.eu/en/downloads/instruction-manuals/former-products/) section, where it is under the heading "LokPilot V1.0".

alain355, 2017-11-11:
@dheap 
Yes Dave, that's why I named it ESU_Lokpilot_V1 like the Manual name, Maybe I should have named it ESU_Lokpilot1.0.xml?
And as you may guess, I downloaded the manual from the ESU "Former Products" section years ago.

Alain

dheap, 2017-11-11:
@alain355 As a compromise between consistency with existing definitions, the desire to move towards ESU model naming practice as closely as possible and clarity for users, I think I'll name the file "ESU_LokPilot1.0.xml", the model as "LokPilot V1.0" and put it in its own family of "ESU LokPilot V1.0".

In this case, the existing "ESU_LokPilotDCC" model should be migrated into the "ESU LokPilot V1.0" family as well, but I'll deal with that later, we have a [proper procedure](http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#update) for family/model migration.

Thanks again for your work on this.

[image]

dheap, 2017-11-12:
Closing via #4413 and #4415. Thanks @alain355.

closed 2017-11-12 by dheap