JMRI/JMRI #4423: CI no longer checking JavaDoc?

bobjacobsen, 2017-11-15:
The Jenkins build task is failing on a Javadoc warning: "warning: no description for @param"

I'll fix that specific issue, happy to do that, but it raises the question of why it wasn't flagged in CI.  Are we no longer running the Javadoc steps in (at least one of) the CI jobs?

Does anybody remember which one was supposed to be running it?

pabender, 2017-11-15:
I thought it was supposed to be run in the headless travis job, but I could
be wrong.

rhwood, 2017-11-15:
I had a headless travis job fail for me this morning on that error, so travis is checking. It is entirely possible however, that Travis is checking only public and protected methods while Jenkins is checking public, protected, package protected, and private methods, which would cause Travis to pass and Jenkins to fail.

Which PR pushed the invalid annotation. Knowing that may help.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-11-15:
Don't know which PR (hard to find on phone), but `git blame` says commit a3650b94139ffe5de3476df686857bcbf7f5bc85 parent 000d195

bobjacobsen, 2017-11-15:
Travis seems to run (from scripts/travis.sh)
```
    mvn javadoc:javadoc verify -U -P travis-coverage --batch-mode -Dsurefire.printSummary=${PRINT_SUMMARY} -Dsurefire.runOrder=${RUN_ORDER} -Dant.jvm.args="-Djava.awt.headless=${HEADLESS}" -Djava.awt.headless=${HEADLESS} -Dcucumber.options="--tags ~@Ignore" 
``

while Jenkins is doing  `ant realclean warnings tests-warnings headlesstest checkPropertiesFiles javadoc`

The `@param` that was flagged in Jenkins was on a `public static void` method of the `public` LayoutEditor class.

bobjacobsen, 2017-11-15:
This commit was probably merged in #4382, which was merged after all CI tests passed.

rhwood, 2017-11-15:
Travis CI was warning on that tag in the CI for #4382, but not failing, so I'll take a look later this week (if not this evening).

silverailscolo, 2017-11-15:
I came across that tag by accident (LayoutEditor.java line 2107) as I expected it to fail some time.
Fix - added description - is included in PR #4420.

rhwood, 2017-11-15:
Maven (and, therefor, Travis CI) will only fail if there is an error processing Javadocs, not a warning.

We can use Checkstyle to enforce the requirement that Javadoc annotations be complete. Since I am working on that in #4425, I will simply mark this resolved once that is implemented.

rhwood, 2018-03-16:
Assignee is no longer involved.

closed 2021-04-14 by stale[bot]