JMRI/JMRI #4427: DCC Aspect Signal Head, can't select all values

KenC57, 2017-11-16:
The tool tip says values of 0-31 are valid but the current scroll box only goes to 13. It should be a combo box as that is much simpler to enter values. And the range in the tool tip is correct.

dheap, 2017-11-16:
I found these (there are two) last night, changed them and was planning a PR. The inorrect range (1-13) was accidentally introduced by @silverailscolo when he changed from JTextFields. [Add tooltips to Merg and DCC add/edit head · JMRI/JMRI@9da257a]( 
https://github.com/JMRI/JMRI/commit/9da257a2a6e60567f0d1d9317e710e78859a8c40#diff-558a29f4c84e79e96192a3f1a9170c2f).

I'm happy to stay with the JSpinners once the range is fixed as you can then type in any valid value. Opinions?

silverailscolo, 2017-11-16:
Sorry. 13, 31, better check the numbers...
The handy thing about spinners in this application is you can type and tab through all on the pane.

dheap, 2017-11-16:
@silverailscolo Easy to do. I'll have a PR in to fix a few minutes.

dheap, 2017-11-16:
Closed via #4428.

KenC57, 2017-11-24:
I think there is something about 13 and the 'Lit' checkmark in the table. The result it anything using the 'Lit' picks aspect code 13. This is reported by a user. I'm re-opening this because I think it was something not thought of during the original fix.

KenC57, 2017-11-25:
Did more digging and thinking. Will open a new Issue that better covers what I think it happening.

closed 2017-11-25 by KenC57