JMRI/JMRI #4436: Build failed - error: reference to EntryExitPairs is ambiguous

AlanUS, 2017-11-19:
Build failed in local repository after I have synchronized a couple of hours ago. Was still OK when I synchronized on 18 Nov. 2017 before all merges performed by @bobjacobsen 

Build log:
```
Execute failed: java.io.IOException: Cannot run program "git" (in directory "E:\Development\GitHub\ALM_JMRI"): CreateProcess error=2, Le fichier spécifié est introuvable
ant -f E:\\Development\\GitHub\\ALM_JMRI decoderpro-old
init:
jjtree:
javacc:
copyfiles:
update-template-code:
Copying 1 file to E:\Development\GitHub\ALM_JMRI\java\tmp
compile-generated-source:
Copying 1 file to E:\Development\GitHub\ALM_JMRI\target\classes\jmri
debug:
Compiling 47 source files to E:\Development\GitHub\ALM_JMRI\target\classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
E:\Development\GitHub\ALM_JMRI\java\src\jmri\jmrit\picker\PickListModel.java:833: error: reference to EntryExitPairs is ambiguous
        EntryExitPairs manager = jmri.InstanceManager.getDefault(jmri.jmrit.entryexit.EntryExitPairs.class);
  both class jmri.jmrit.signalling.EntryExitPairs in jmri.jmrit.signalling and class jmri.jmrit.entryexit.EntryExitPairs in jmri.jmrit.entryexit match
1 error
2 warnings
E:\Development\GitHub\ALM_JMRI\build.xml:632: The following error occurred while executing this line:
E:\Development\GitHub\ALM_JMRI\build.xml:579: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 3 seconds)

```

rhwood, 2017-11-19:
Try running ```ant clean``` and then building. The recent changes move some classes and the error you have is not unanticipated when that happens.

dheap, 2017-11-19:
@AlanUS I got exactly the same error last night but had noticed the changes to `EntryExitPairs` so did an `ant realclean` and the problem went away.

Maybe an `ant clean` (as @rhwood has suggested) is sufficient.

AlanUS, 2017-11-24:
@rhwood @dheap 
Thanks for the tip. `ant clean` did the job.

closed 2017-11-24 by AlanUS