JMRI/JMRI #4446: Create Test Release 4.9.7

bobjacobsen, 2017-11-21:
Seventh and last in the usual sequence leading to 4.10 in December 2017, and traditionally the start of focusing down to the final production release. Last from HEAD of master in this release sequence; the convergence to JMRI 4.10 starts after this. Expected around November 26.  Note that this is shortly after the 4.9.6 test release, not the usual full month.

Paelgin, 2017-11-22:
I tried this version, but the WebServer crashes JMRI on my Linux box.  4.9.4 was the last Linux version that did not.  I found out when I downloaded and installed 4.9.5 on my working 4.9.4 box (Debian 4.9.1, Java 1.8.0_151, PR3 to a Zephyr DCS50).  I reloaded 4.9.4 and it worked, so I re-installed a new copy of 4.9.5 which crashed as soon as I tried to start the WebServer.  I also tried a Jenkins build (4.9.7ish from this morning) which also crashed.  A java window opens with an "OK" box and a message of "Unrecoverable error encountered.  This application will quit"  Then "java.lang.NoSuchMethodError: org.eclipse.jetty.server.handler.ResourceHandler.(Lorg/eclipse/jetty/server/ResourceService;)V" and 10 more lines.  A picture of the screen was posted on the Yahoo Group files section under Problems Being Worked On/Phil Abernathy.
Thanks for all you do!!!  btw, the windows versions have not shown this issue.
Phil Abernathy

rhwood, 2017-11-22:
If you install a newer version of JMRI in a different directory than the existing version, does this issue remain, or is it only when you install a newer version of JMRI in the same directory as the older version?

Paelgin, 2017-11-22:
Amazing.  A different directory works as expected.  I now apparently have two copies installed, one in /home/tsmri/Downloads and one under /usr/local;as a Windows guy, I am dabbling in Linux on an old ex-XP box.  The new install location works, and the old one fails.  Must be something left over I need to get rid of.

As an explanation, I copied the tgz file to /usr/local and, as root, ran the command "tar -xf JMRI.4.9.6.R23c3893.tgz" which created the directory /usr/local/JMRI (which I chmod to 777).  I have a batch file (oops, is that a bash script?) (with permissions changed to 777 so as a user, I can execute it) that adds the java bin to the $PATH, then chdir to where PanelPro is located, then runs "./PanelPro".
 
One simple question: how can I add my user name to ttyACM0 permissions without having to chmod 777 /dev/ttyACM0 (as root) every time I reboot?
Phil Abernathy

Paelgin, 2017-11-22:
At least it worked one time. I shut down and restarted and it JMRI never got to the splash screen.  
I have found that Open-jdk or Open-jre is hanging around, and may be causing at least part of the problem.  I have java1.8.0_151 installed, but I found that before I run my bash file to add java bin directory in the path, the command java -version gets answered, which shouldn't happen.  I might wipe the disk and see if I can avoid loading any Open-j*.  But that's another day.
Phil

 
  On Tue, Nov 21, 2017 at 8:48 PM, Phil Abernathy<philabernathy@yahoo.com> wrote:   Amazing.  A different directory works as expected.  I now apparently have two copies installed, one in /home/tsmri/Downloads and one under /usr/local;as a Windows guy, I am dabbling in Linux on an old ex-XP box.  The new install location works, and the old one fails.  Must be something left over I need to get rid of.

As an explanation, I copied the tgz file to /usr/local and, as root, ran the command "tar -xf JMRI.4.9.6.R23c3893.tgz" which created the directory /usr/local/JMRI (which I chmod to 777).  I have a batch file (oops, is that a bash script?) (with permissions changed to 777 so as a user, I can execute it) that adds the java bin to the $PATH, then chdir to where PanelPro is located, then runs "./PanelPro".
 
One simple question: how can I add my user name to ttyACM0 permissions without having to chmod 777 /dev/ttyACM0 (as root) every time I reboot?
Phil Abernathy

dheap, 2017-11-22:
> @Paelgin One simple question: how can I add my user name to ttyACM0 permissions without having to chmod 777 /dev/ttyACM0 (as root) every time I reboot?

Enter the following command **exactly** (except for the double quotes):
"sudo adduser ${USER} dialout"
   (This assumes "dialout" is the group shown beside your device in the list above.)

Do not:

- Try to bypass group membership by running as root. That will create further problems.
- Try to change permissions of the serial port you see. They will be lost every time you plug/unplug/logout/reboot.

Then logout and back in again so the group membership takes effect.

Paelgin, 2017-11-23:
Working now!  I formatted the hard drive and re-installed Debian 9.2.1. I d/l and installed 4.9.6 and everything works. Apparently merely loading 4.9.5 over 4.9.4 had left something that didn't work together.  Thanks to all who contributed and commented!  Sudo was not installed, so I apt installed it, but had to visudo to get user added. I added my user to dialout and after restarting, ttyACM0 worked as expected.  As a windows guy, I had to google and play, but it all works now (including my panels) on this machine like my Win10 PC.  I am hoping to deploy this at the club and get my CHIP $9 computer back for other uses.

I found and deleted the symbolic link /usr/local/bin/java (which is in the $PATH) which linked to the OpenJava.  My bash script (PP), which runs on startup, first exports the path to java 1.8.0_151 into the $PATH variable, and then changes to the JMRI folder and runs ./PanelPro.  

Phil

bobjacobsen, 2017-12-03:
It's time to build 4.9.7. This is the last test release in this series from HEAD of master before the next production release. The next test release will be 4.11.1, right after the 4.10 production release (unless we change those names).

I propose to create the branch from master tomorrow morning (about 20 hours from now).  If you're working on a PR which you think (a) would be good to get into the December production release instead of following in a test release and (b) you're sure isn't going to cause new problems for users, please merge it before then.

Even after the test release branch is created, maintainers should please _not_ merge After-next-production-release PRs just yet, to make sure we have something stable in this test release series.  

Thanks.

bobjacobsen, 2017-12-05:
The release-4.9.7 branch has been created. 

From now on, please document your changes in the [jmri4.11.1.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.11.1.shtml) release note file.

Maintainers, please set the 4.11.1 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.9.7/)

If you're developing any additional (post-4.9.7) changes that you want in the JMRI 4.10 production release, please start from this branch, i.e. do
```
git checkout -b release-4.9.7
```
to start your work.

bobjacobsen, 2017-12-08:
This release has been created.

closed 2017-12-08 by bobjacobsen