JMRI/JMRI #4460: DCC Signal Decoder, unlit is stuck with code 8 (or maybe 13)

KenC57, 2017-11-25:
Regardless of the value used for the 'Dark' entry in the head, clicking the Lit will toggle with code 8 as dark. It should use the code matching the 'Dark' value.

Also there should be a way to disable an aspect. A signal mast decoder can disable aspects. Well, on a head, I guess appearances would be the better word. But not all heads support all appearances. This would be important when building a mast from single head decoders.

A user was finding it as value 13, but I'm finding it stuck at 8.

After editing a DCC signal decoder head a few times, it stopped giving proper values to the net. I was testing with LocoNet Sim and looking at the data to see what was being sent. A number of the values I had tried setting was beyond the default 0->8.

dheap, 2017-12-02:
Fixed the Unlit -> Dark mapping via #4500.

KenC57, 2017-12-03:
That fixed it. Thanks @dheap

closed 2017-12-03 by KenC57