JMRI/JMRI #4464: NCE Powerhouse Pro fails long sequence of reads

bobjacobsen, 2017-11-25:
Had two reports of Powerhouse Pro "locking up" when doing a read all panes.  Got a chance to check it just now, and it's pretty much (but not always) repeatable:

- NCE Powerhouse Pro
- JMRI 4.9.2  (haven't updated yet, will try this later this week)
- Tsunami2 "TSU-2200 Alco Diesels" decoder selected

"Read all panes" proceeds until reading CV254, and then there's no further response from the NCE command station.  (Usually, not always, power returns to the layout; a couple times the locomotive on the programming track was powered up which was a bit unexpected)  The command station doesn't respond to anything else on the serial link.

- Just reading CV254 (or others in the area) is OK, doesn't provoke the failure
- Reading individual panes is OK, doesn't provoke the failure

- Power cycling the command station restores normal operation

I suspect, but haven't found a way to test, that it's somehow a long series of programming operations that's overflowing something in the command station.  But doing a bunch of short ones doesn't provoke it, so I'd like to figure out how DecoderPro can break that long sequence up so problem's avoided.

dheap, 2017-11-25:
I think I can pretty pretty safely assert that I've read more CVs with JMRI and Power Pro than anyone else in the world. I develop the ESU code and could not count the number of times I have done a Read All Sheets on an ESU decoder with ~1,000 CVs.

I can say I've seen no pattern for this particular failure; decoder brand (ESU, QSI, SoundTraxx), number of CVs read, CV data, CV number, how far into the process it breaks, how many times it breaks...

Some days are golden; read decoders all day with never a miss, others are stone - need to power cycle the command station one or many times to get though a single decoder.

Dave

dheap, 2017-11-25:
However:
I have noticed in the past that a Read Full Sheet of the CVs pane is far more likely to succeed than a Read All Sheets.

Yesterday, when testing the code I was working on at the time I noted a related problem (Read All Sheets versus Read All CVs). In this case it was writing rather than reading and it didn't necessarily lock up the command station. Turning on lots of debugging revealed that the Write Sheet procedure consistently sent overlapping writeCV commands for certain CVs without waiting for a programmingOpReply. Further investigation narrowed the problem to CVs that appeared more than once in different variables and/or were in a SplitVal on the sheet. It appears that Sheet operations occur by variable rather than CV number.

I can't put any more time into this for a day or so (I need to get reacquainted with my wife), but I put it on the to-do list.

But thinking about this I can see a possible connection. If we send overlapping read requests to the Power Pro (before receiving a 21 ACK from the Power Pro), I could understand it losing its mind.

I may be on the wrong track with this connection, but at least its worth a check of Read All CVs versus Read All Sheets. Maybe they were my Diamond versus Stone days!

Dave

dheap, 2017-11-26:
Today was a Diamonds Day:

I did 10 consecutive reads of a LokSound V4 decoder, 5 with Read All CVs and 5 with Read All Sheets.

Since the ESU decoders have ~1,000 CVs (of which no more than 256 are non-indexed - no CVs beyond 512) that means~2,400 uninterrupted consecutive read/write instructions per batch (I had Indexed CV caching disabled).

That means a total of 24,000 read/write instructions in 10 batches of 2,400.

The result: not once did the Power Pro (with SoundTraxx PTB-100 attached) command station "lose the plot".

Tried again with index caching and rad after write enabled (cuts writes from 1,300 to just tens), No difference,

I dont have a TSU2 decoder but will look into the definitions and do some tests.

dheap, 2017-11-26:
Write All Sheets will definitely fail (hang) with any TSU2 or Econami decoder because the Power Pro cannot directly write any CV>256 and SoundTraxx does not provide a workaround.

My Read All Sheets with a "TSU-2200 Alco Diesels" and an ESU decoder attached (in these decoder no read fails, just returns a value of 0) would not hang.

dheap, 2017-11-26:
*(in these decoder no read fails, just returns a value of 0 **for any unsupported CV**)

dheap, 2017-11-26:
I've dropped an email to Marcus Ammann. He does the offical NCE servicing for Australia and has most likely to have a TSU2 on hand. I'd like to see if he can reproduce the failures @bobjacobsen has experienced.

dheap, 2017-11-26:
@bobjacobsen Can you check NCE firmware version (PROG 5 ENTER) on the Pro Cab throttle? Marcus has access to various firmware versions.

dheap, 2017-11-26:
Tried to download V4.9.2 but the download links were pulled due to serious issues with this version.

dheap, 2017-11-26:
CVs253-256 are readonly Product Information CVs from these decoders.

dheap, 2017-11-26:
254 and 256 are readonly enumVal variables...

dheap, 2017-11-26:
According to the SoundTraxx ProductID manual, CVs 253-256 do not exist unless CV 7 reads back 70 or 71. There's a lot that is unusual about these variables.
[productID.pdf](https://github.com/JMRI/JMRI/files/1503852/productID.pdf)

bobjacobsen, 2017-11-26:
I asked the layout owner (I'm away from there for a bit) about the version and he replied:
```
NCE Radio V1.5
Command Station: Version 18:04
    Aug 31 2011
```

That sounds pretty old, but the [JMRI NCE page](http://jmri.org/help/en/html/hardware/nce/NCE.shtml) only discusses EPROMs of 2007 and earlier(!)  Is there an update we should get?

dheap, 2017-11-26:
I was already starting to have my supsicions about whose layout it was, but this almost certainly confirms my suspicions. We can probably count on one hand the number of people in the world who have that firmware version. It's an unreleased beta version (newer than any production versions). I fitted to mine it several years ago in an attempt to diagnose another problem that was alternately blamed on NCE and JMRI, but that stalled due to lack of cooperation in getting requested tests done. We need to take this offline.

devel-bobm, 2017-11-27:
@dheap wrote:

> Yesterday, when testing the code I was working on at the time I noted a related problem (Read All Sheets versus Read All CVs). In this case it was writing rather than reading and it didn't [...]

I wonder if there is a similar phenomenon which affects Digitrax programming.

I have seen some "out-of-sequence" LocoNet traffic when JMRI writes index registers for a "read full sheet" on the "CVs" tab for a LokSound V4.0 decoer.  And I have seen a LocoNet trace where the DCS240 programming interface seemed to "lock-up" immediately following one of these "out-of-sequence" index register writes.

dheap, 2017-11-27:
I've just emailed @bobjacobsen some logs documenting this problem,

dheap, 2017-11-28:
Reporting back on two days of exhaustive (and exhausting) testing:

I have conducted around 50,000 read/write operations in batches of between 400 and 2,400 reads/writes long sequences, using both LokSound and Tsunami2 definitions. At no time did my NCE command station "lose the plot" and become unresponsive. Initially I did not have access to any Tsunami2 decoders so I used an ESU decoder as the read source (ESU decoders never return a read fail for unsupported CVs, just a return value of 0).

Marcus Ammann ran tests for me with V4.9.4 under Windows (using an Econami decoder as had no access to a Tsunami2 - the include files for the offending CVs/variables are the same). He tried both [NCE Firmware](https://sites.google.com/site/markgurries/home/nce-info/nce-cmd-station-software/powerpro-softare) versions 2007C ( the current released version) and the unreleased 2011 beta version fitted to the as-reported command station. He was unable to recreate the issue.

Gary Ible advised me that he had a loco fitted with a Tsunami2 decoder and offered to lend it to me. He performed tests on his NCE system (with 2007 version firmware and the latest main PCB version) using V4.9.6 under Windows and was unable to recreate  the issue.

This morning I performed fairly exhaustive testing with the loco. It was fitted with a "TSU-1100 Baldwin Diesels" decoder so I performed testing with both that definition and the "TSU-2200 Alco Diesels" definition. I tested with both Head of master and V4.9.2 under Mac. My command station is fitted with the 2011 firmware, the earlier version of the main PCB and a PTB-100. I was unable to recreate the issue.

One thing I forgot to test before returning the loco was any possible effects of reenabling the "Allow caching when writing index CVs" and "confirm CV writes" options*. (I had tested both ways with my earlier checks).

So (apart from suggesting that the originator of the report update to the latest test version and try disabling the "Allow caching when writing index CVs" and "confirm CV writes" options*), I would have to suspect that there is something unique about that particular setup.

@bobjacobsen you mentioned "two reports". Were they from the same source?

* These two options disable some JMRI functionality when used with an NCE Power Cab system, but that will be the subject of another issue.
Likewise the out-of-sequence reads/writes both @devel-bobm and I have observed will be a separate issue. It may or may not be related to this issue.

bobjacobsen, 2018-01-03:
I spend a few more hours locking at this on one of the layouts that's reporting it.  Thanks for all the work & info above. I wasn't able to repeat the fault routinely, but it did recur enough times that I'm convinced the problem is out-of-order operations by JMRI.  The four I caught all involved split-var operations where the PowerPro took a little longer than usual to reply, and things just went sideways after that.  Will work on that aspect next.

closed 2021-04-14 by stale[bot]