JMRI/JMRI #4477: Occasional stack overflow while testing

pabender, 2017-11-27:
We are getting an occasional, but definitely re-occuring, stack overflow error whiel running tests.  It appears to be while building a ComboBox during one of the operations GUI tests.  Here's the start of the stack trace (which repeats to the end of the log of course...)
(this was copied from https://ci.appveyor.com/project/pabender/jmri-vis06/build/1791 )

    [junit] Testcase: testTrainByCarTypeFrame(jmri.jmrit.operations.trains.OperationsTrainsGuiTest):	Caused an ERROR
    [junit] null
    [junit] java.lang.StackOverflowError
    [junit] 	at java.awt.Container.minimumSize(Container.java:1849)
    [junit] 	at java.awt.Container.getMinimumSize(Container.java:1833)
    [junit] 	at javax.swing.JComponent.getMinimumSize(JComponent.java:1740)
    [junit] 	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.minimumLayoutSize(BasicComboBoxUI.java:1896)
    [junit] 	at javax.swing.plaf.basic.BasicComboBoxUI$ComboBoxLayoutManager.minimumLayoutSize(BasicComboBoxUI.java:670)
    [junit] 	at java.awt.Container.minimumSize(Container.java:1849)
    [junit] 	at java.awt.Container.getMinimumSize(Container.java:1833)
    [junit] 	at javax.swing.JComponent.getMinimumSize(JComponent.java:1740)
    [junit] 	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.minimumLayoutSize(BasicComboBoxUI.java:1896)
    [junit] 	at javax.swing.plaf.basic.BasicComboBoxUI$ComboBoxLayoutManager.minimumLayoutSize(BasicComboBoxUI.java:670)
    [junit] 	at java.awt.Container.minimumSize(Container.java:1849)
    [junit] 	at java.awt.Container.getMinimumSize(Container.java:1833)
    [junit] 	at javax.swing.JComponent.getMinimumSize(JComponent.java:1740)
    [junit] 	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.minimumLayoutSize(BasicComboBoxUI.java:1896)
    [junit] 	at

bobjacobsen, 2017-11-27:
That kind of thing (infinite recursion in the Swing PLAF code has appeared before due to modifications of Swing or AWT data structures from off the Swing thread.

rhwood, 2020-07-20:
This appears to have been fixed.

closed 2020-07-20 by rhwood