JMRI/JMRI #4478: Layout editor zoom misbehaving in 4.9.5

cooka7, 2017-11-27:
Any reduction in panel zoom moves icons, sensors, labels etc but does not show track that was not shown on original screen.
Reverted to 4.8 and panel zooms up & down ok.
Love the new LE features, especially selectable fonts, can have then specific to the original operating company.
Alan Cook

geowar1, 2017-11-27:
I'm unable to duplicate... I suspect because all my layouts have their panel sizes set correctly (larger than everything on the layout)... out in the "real world" I've found this to not be the case. A quick test is to "Zoom to fit" which re-calculates the panel size based on the layout's actual contained objects (plus a margin). If after doing that you're unable to reproduce then that's probably the issue (the saved panel size is smaller than the actually layout). If you save your layout after doing the "Zoom to fit" then the correct panel size will be saved and going forward everything should then work as expected.
Or, I could be wrong... if so the let me know so I may investigate; if not then close this issue. ;-)

cooka7, 2017-11-28:
Hi George
Thanks for looking in to this.
The panel is sized to show on the layout computer which has a 15" screen. This is fine.
However not all the keyboard buttons on this (Vista) machine work, so scripting and dev work is done on a 12" screen fully working XP machine.
in 4.8  on XP zoom to fit works fine, had to turn off XML validation to do that, it zooms the panel and all objects correctly, and draws the track that was off the screen.
Same panel zooms everything correctly including track on the vista machine, so must be an XP only thing.
Billy Gates Strikes Again.
None of this is a showstopper, just thought might be worth reporting.
Tried saving with the zoom to fit option, reopening and it came up full size, no zoom all track shown
Thanks for all the work you have done, LE is transformed by it and a joy to use.
Alan

cooka7, 2017-11-28:
How do i get screen shots into here?

bobjacobsen, 2017-11-28:
> How do i get screen shots into here?

You have to do it via the web interface; screen shots in email replies are removed.

Go to https://github.com/JMRI/JMRI/issues/4478 (for this specific issue) and drag your screen-shot to the comment box, then click the green (?) comment box.

Bob

geowar1, 2017-11-28:
On Nov 28, 2017, at 2:27 AM, cooka7 <notifications@github.com> wrote:
> Thanks for looking in to this.
> The panel is sized to show on the layout computer which has a 15" screen. This is fine.
> However not all the keyboard buttons on this (Vista) machine work, so scripting [...]

Ouch! What was the error (before you turned off xml validation)?
> it zooms the panel and all objects correctly, and draws the track that was off the screen.
> Same panel zooms everything correctly including track on the vista machine, so must be an XP only thing. [...]
You should use the Options menu's "Save Location and Size" command also; this saves the current location and size (Duh!) per-machine/user (where as the value saved in the layout file is per-layout).
> Thanks for all the work you have done, LE is transformed by it and a joy to use.
> 
[/blush] Glad to hear it. If you (or anyone else reading this) have suggestions for other "enhancements" for LE please let me(/us) know (best: file an incident: <https://github.com/JMRI/JMRI/issues> (as you did for this zoom issue); it's my working queue ;-).
-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

cooka7, 2017-11-28:
Not having any luck with getting the screenshots on here, will email direct.
The validation failures in 4.8 I expected as panel now uses custom colours and custom fonts.
Tried an old panel version from august and that misbehaves in 4.9.5 and 4.9.6
Cheers
Alan

cooka7, 2017-11-29:
[image]
[image]
[image]
Unsupported browser problems now solved.
Doncha just love progress.

closed 2017-12-09 by cooka7