JMRI/JMRI #4487: No DCC++ output control for turnouts

DRChess, 2017-11-29:
Using the Z command to control a pin output was to be fixed in JMRI test version 4.7.2 but yet the problem still exist and has currently been checked off. I have tested this on Linux and Windows using different production releases and test release versions of JMRI. When controlling a Z declared turnout pin the control for (Thrown) and (Close) should toggle in the turnout table. Instead it blinks and reverts to the previous state as reported long ago. The DCC++ traffic monitor shows only off commands and never any on command. No wonder it’s not working JMRI doesn’t modify the output command to on. The monitor shows formatted "Z ID 0" being sent over and over and never is a "Z ID 1" being used. The fix was developed on a MAC. Or maybe I should have just used a MAC or is it something else? And yes BSOUTPUT is used so "a" or "T" is not the issuing command structure. It is the "Z" command sent and never with an on (1) state.

pabender, 2017-11-30:
What specific version are you using?

Also, are you looking at controlling IO pins or DCC turnouts?

I will need to look at this in more detail, but the DCC++ sketch uses T for throwing DCC turnouts and Z for controlling IO pins.

Paul

DRChess, 2017-11-30:
I'll assume specific version is talking about the DCC++ which is1.2 or current version on an Arduino mega 2560 using the recommended Pololu motor shield. If version is JMRI then get ready 4.7.2   4.7.6  4.7.8  4.8  4.9.5 and 4.7.2 in the git was the fix but doesn't work using Linux and Windows but not tested on MAC.

Yes, I want outputs to control a TTL H bridge interface. The JMRI DCC++ traffic monitor both raw and text clearly shows only off or 0. What I had setup was ID 10 on pin 30 using the Z command from Arduino IDE monitor. Things work as expected and yes it was saved with the capital "E" command. So when Arduino is power off it was saved to EEPROM and returned still working as expected. The command Send is Z 10 0 recv: Y 10 0. Never and I mean never Z 10 1. Arduino IDE can turn on and off the output but JMRI can only turns it off and again never on.

Again, per the JMRI DCC++ traffic monitor which shows only off commands sent doesn't work because the output sending never shows an on (1). Very easy to check.

I'm starting to think this maybe a JDK vs. JRE vs. Oracle. Is Java that broke???? I have seen other issues stating something similar.

No, the DCC++ sketch uses "a" to sent track accessory commands "T" is stored track accessory commands and "Z" IO output commands. JMRI uses Direct, BSTURNOUT, and BSOUTPUT as identifiers in feedback pull down. Strange place to control output mode in feedback pull down.

pabender, 2017-11-30:
Sent from my iPad

> On Nov 30, 2017, at 8:33 AM, DRChess <notifications@github.com> wrote:
> I'll assume specific version is talking about the DCC++ which is1.2 or current version on an Arduino mega 2560 using the recommended Pololu motor shield. If version is JMRI then get ready 4.7.2 4.7.6 4.7.8 4.8 4.9.5 and 4.7.2 in the git was the fix but doesn't work using Linux and Windows but not tested on MAC.

Yes, I was asking about the JMRI version.

There is nothing on the source code for DCC++ that is system specific.

> I'm starting to think this maybe a JDK vs. JRE vs. Oracle. Is Java that broke???? I have seen other issues stating something similar.

No, that wouldn't be an issue.

> No, the DCC++ sketch uses "a" to sent track accessory commands "T" is stored track accessory commands and "Z" IO output commands. JMRI uses Direct, BSTURNOUT, and BSOUTPUT as identifiers in feedback pull down. Strange place to control output mode in feedback pull down.

Yes, I was just looking at the way @msunderwd did this.  It's not a good design from an object oriented perspective.  It's trying to describe 3 things that behave very differently with one object.  The stored turnout commands are really just aliases for the accessory commands, so it does make some sense to combine those into one object type.

As far as your problem with the z outputs, It looks like the error is that the feedback from the command station is never used to update the current state of the output.   Because the output state isn't updated, the next time you go to operate the device, the same command is sent.

I have the ability to test the Z output commands ( that is I have an Arduino with the DCC++ sketch loaded ), so I will try to work on this sometime over the next few days.

In a longer ( or maybe shorter ) term redesign, I think it makes sense to break apart the DCC accessory decoders from the Arduino output pins.   In addition to turnout outputs, the JMRI code also supports light objects for DCC++.  These are currently sending accessory decoder commands, but I think it makes sense to turn those into Ardunio output pin commands instead.

pabender, 2017-12-01:
So it turns out there was basically a 1 line issue with this that caused the value sent to the Arduino to be inverted from what the feedback code expected.

The issue basically is that thrown and closed were flipped when sending for the output pins vs the turnouts.

I also took the time while I was fixing that to enable the turnout manager to automatically populate the output pin turnouts into the table, like it would do with the defined accessory decoder turnouts.

DRChess, 2017-12-01:
It sounds like you are making major progress. Thanks for you effort.

pabender, 2017-12-01:
If you could, please try one of the daily builds from:

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

The package links on the page above include the changes I made last night.

pabender, 2017-12-01:
By the way, looking at this last night, I really do think moving forward we need to split the accessory decoder outputs from the arduino output pin outputs as I mentioned above.

The problem is that There are two distinct address spaces stored by the base station, one for the accessory decoders ( accessed with the T command ) and one for the arduino outputs ( accessed with the Z command ) but with the way things are currently done, we can't access both an arduino pin labeled as  "2" and an accessory decoder labeled as output "2".  The base station allows this, so JMRI should handle that.

As I mentioned earlier, I think the simplest way to do that moving forward is to turn the arduino pin outputs into JMRI light outputs.  ( most of the work I did last night could pretty easily be moved to the DCC++ objects that support that interface.)

bobjacobsen, 2017-12-01:
> The problem is that There are two distinct address spaces stored by the base station, one for the accessory decoders ( accessed with the T command ) and one for the arduino outputs ( accessed with [...]

I don't know much about the internal structure of the this code, so don't know how much work it would be, but another alternative is to code the difference in the names:

- DCCPPT21 - the usual access to accessory decoders
- DCCPPTP21 - the extra 'P' at the start means that this is a pin, P21

We have several other systems that make internal distinctions on the name that could serve as a model. 

The advantage is that all the Turnout functionality (ability to be included in a signal, invert, etc) stays with the pins this way.

Another approach would be to implement the pins as Lights, and then later on somebody could come back and add the additional coding to Turnouts.  C/MRI and I think other systems show how to handle disambiguation in that case.

pabender, 2017-12-01:
On 12/01/2017 03:10 PM, Bob Jacobsen wrote:
> I don't know much about the internal structure of the this code, so
> don't know how much work it would be, but another alternative is to code
> the difference in the names:
>
>   * DCCPPT21 - the usual access to accessory decoders
>   * DCCPPTP21 - the extra 'P' at the start means that this is a pin, P21

That would work too.

Another option that might work would be to set the arduino pins into a 
different address range.

The base station stores the names for the pins and stored turnouts in 
the Arduino's eeprom, and the tables are used as a mapping to actual pin 
numbers.  In other words, the label for pin 21 doesn't have to be "21", 
nor does the label for Turnout (stored accessory decoder address) 21 
have to be "21".

Another option is to have JMRI manage the assignment of labels.  Mark 
wrote a tool to assign the labels, but it is a manual process.

The current structure uses the feedback mode to disambiguate between the 
three possible modes:

BSOUTPUT (which under the hood is labeled as "EXACT" mode) handles the 
arduino pins, but only if they are mapped on the base station (the pins 
can also be used as sensors).

BSTURNOUT (which under the hood is labeled as "MONITORING" mode) handles 
turnouts that are mapped on the Base station to accessory decoder 
addresses.  The assigned value does not have to match the accessory 
decoder address.

DIRECT mode handles raw accessory decoder addresses.

But that only tells us how to send the data to the layout, not how we 
should disambiguate between the 3 possible interpretations of 21 as an 
address.

I haven't dug into it enough to find out, but the labels on the base 
station may actually be a string.  I suppose we could start managing 
those ourselves.  (we have a tool for that already, but it isn't automatic)

> The advantage is that all the Turnout functionality (ability to be
> included in a signal, invert, etc) stays with the pins this way.

True.  I think the lights can also invert (alot of this code was 
borrowed from the Lenz implementation, and it has a lot of commented out 
sections that need to be cleaned up).

> Another approach would be to implement the pins as Lights, and then
> later on somebody could come back and add the additional coding to
> Turnouts. C/MRI and I think other systems show how to handle
> disambiguation in that case.

In reality, we should be able to use any of the three address types for 
any output.  The current code implements raw accessory decoder addresses 
as lights.

Paul

DRChess, 2017-12-05:
Paul,
I tried the windows version 4.9.7 Jenkins. It works like it should. I like opening the DCC++ monitor window to list the sensors and all turnouts. It helps to verify the tables when setting up the tables. 

Thanks for the fix.

DRChess, 2017-12-06:
Paul,
The Linux  version 4.9.8 jenkins on the Raspberry Pi 3 is working. Both Windows and Linux seems to be working from the turnout tool. I will start setting up layout panels to continue the validation.

tpicraux, 2017-12-14:
Since this thread is about the turnout functions of dcc++ I have been following this thread and have installed JMRI and tested ( using Panel Pro) 4.97+R72446d4 on Windows 7 as well as on a Raspberry Pi 3, both tests use the same Arduino mega 2560 running dcc++ 1.2.1+. My specific interest at the moment is using the Turnout commands with a single id (BSTURNOUT mode), although I will likely use outputs, accessories, etc in the future.
Results:
3 turnouts defined and stored in dcc++ eeprom (ID 1,2,3)
For Windows 7, on startup, the turnout table shows as it should ...I have tried all permutations of open closed, so this is perfect ...
For Ras Pi, on startup results are inconsistent, most of the time correct, but sometimes one or more show as unknown.  Restarting panel pro usually gets the right values. I can not come up with a case that I can duplicate ...maybe a serial com timing issue? I can live with this...
However a problem still exists with the configure base station command ...not sure if this should be a new issue, but the problem is:
 when the configure dcc++ base station is run after startup (maybe to add a new turnout, ...) all entries in the turnout table get set to thrown, even though the dcc++ system reports correct states in the traffic monitor ...
in the test prior to running the Configure DCC++ Base Station command all turnouts were closed, and showed as closed in the turnout table. After running the configure base station command the traffic monitor indicates that all 3 turnouts are closed, but the JMRI traffic monitor interpretation is that the first is thrown, the next 2 are closed, ....yet all entries in the turnout table are set to thrown. attached is a screenshot

[JMRI Turnout Issue.docx](https://github.com/JMRI/JMRI/files/1560343/JMRI.Turnout.Issue.docx)

DRChess, 2017-12-22:
I was looking at the monitor print out and the DCC++ base station reported H1 set to 0, H2 set to 0, H3 Set to 0. This means all are closed. Some reason the first turnout was not interpreted correctly. Make sure there is no invert because it is quite tricky to have one or two inverts on. Looking at the test you may want to understand which DCC address is being sent in your case it was 5,9,13 I had fun figuring it out. Formula is (address*4)+subaddress+1 this comes from the raw packet information of DCC per the NMRA. eg address = 511 subaddress  = 3 which is 2044+3+1 for 2048 last turnout possible.

Dave

tpicraux, 2017-12-23:
Thanks for taking a look at this.  I will double check on the invert issue, but I thought that I had checked that already.  But, the major concern is that dcc++ is reporting that all 3 turnouts are all  closed (0), but continuing to look at the traffic monitor, for some reason JMRI interpretation is that turnout 1 is thrown, the other 2 are closed, and then the actual turnout table gets changed to all thrown. (The turnout table showed them all as closed prior to my issuing the configure base station command from JMRI).  It seems JMRI is misinterpreting the feedback it gets from the command. In fact, although I did not show the case of some closed, some thrown, i had run that test but always the turnouts get set or changed to thrown after the command is run.

On the dcc++ side I do not use the address and subaddress fields, that is why I arbitrarily set them to the index value and 0 respectively. Instead I have modifications to drive the turnouts using TB6612 driver ics, not dcc commands, using the bsturnout mode in JMRI.   However the JMRI results are the same with or without my mods in dcc++.  All else actually works very nicely, except sometimes on JMRI startup the turnouts state in JMRI is not set correctly, and running the configure base station command always changes them to thrown state.

DRChess, 2018-01-02:
There is a lot of miss-understanding of the term feedback. In the case of DCC++ the feedback field is used to control output type to the DCC++ interface. Sensors would need to be installed and verify by the JMRI interface for it to work and give positive train control. DCC++ has the ability to store last commands if using the BSTURNOUT and BSOUTPUT commands. A problem with loading them from the DCC++ interface is that JMRI on startup probably does not know which panel tables are loaded and how to updated any master record. I get the ? marked icons and the DCC++ knows what is set. As far as JMRI is concerned is that DIRECT commands is never recorded. This suggest a need for some type of button on the panel layouts that would initializes all layout power on which should happen first so the stationary decoders would stabilize before getting any commands and then set all turnouts.  

The creation of automation of the layout through JMRI is quite a feat. When coming to building panels and creating routing control is something JMRI is great for also programming decoders. I'm probably missing some startup script that would set things to actually what I want but for now I have time to play.

By the way, the first read of the first turnout in the list is missed as closed but gets a '0' and displays thrown. After that is seems to be working. This doesn't seem to be a big problem.

pabender, 2018-01-02:
> On Jan 2, 2018, at 10:22 AM, DRChess <notifications@github.com> wrote:
> There is a lot of miss-understanding of the term feedback. In the case of DCC++ the feedback field is used to control output type to the DCC++ interface.
> 
That is something that will soon be changing.  This kind of inconsistency is something that drives us nuts from a support perspective.

The likelihood is that we will be providing 3 separate address ranges for the three supported address types ( that is raw DCC accessory decoder addresses, stored DDC accessory decoder addresses, and stored IO pins )
> Sensors would need to be installed and verify by the JMRI interface for it to work and give positive train control. DCC++ has the ability to store last commands if using the BSTURNOUT and BSOUTPUT [...]
We have no problem asking for information from the command station and mapping that to addresses.  The biggest issue we have right now is that if there is a DCC turnout, a stored DCC turnout, and an IO pin all configured with the same address.

Paul

DRChess, 2018-01-02:
Sorry not the response I was looking for.

DRChess, 2018-01-02:
Paul,

I was bringing up some panels that has declared turnout information and was wondering why I must exercise each for what position the turnout is in (closed-thrown). Seems that information is stored in the DCC++ interface and just needs updated.

DRChess, 2018-01-03:
tpicraux 
The BSTURNOUT sends DCC packet information to the track. To change a pin from high to low you must use BSOUTPUT. JMRI will be making the much clearer.

This is **not** the forum for the next answer but the TB6621 needs two to three outputs from the Arduino or an additional circuit to make it do bi-directional voltage change and DCC++ provides only a single  output.  It is much cheaper and simpler to use a LM324 quad op-amp. To build use 15 volts to power the IC, take the inverting (-) input from amp 1 and tie it to amp 2's non-inverting (+) input now use this as the input from the Arduino. The non-connected amp inputs must be connected to a voltage around 2 volts. I use a 10K and 2K resistor to build a voltage divider powered by the supply voltage. The output from amp 1 and 2 drives the stall motor directly.  Amps 3 and 4 in the IC are for a second stall motor. 
**How it works:** Because the op-amps are being used as comparators to about 2 volts when the voltage is low (less than 0.8)  from the Arduino, 1 amp's output  is high about 11-12 volts after voltage drops and the other is low about 1-2 volt. When the Arduino's output it is high (usually 4.5) the amps polarity reverses. This works on my layout, uses 3 parts and a small supply cap around 10 microfarads and runs two stall motors, all costs about $0.60. Keep it sensible and simple

Sorry guy's about the off topic but this seems to need mentioned.

Dave

tpicraux, 2018-01-03:
Sorry I mentioned. But My approach works perfectly (i modified dcc++) and I will keep it. Feel free to contact me at tpicraux@yahoo.com if interested to discuss further.

closed 2018-02-19 by DRChess