JMRI/JMRI #4488: "compare" in symbolic programmer fails for LocoNet Ops-mode programming

devel-bobm, 2017-11-29:
The symbolic programmer's "confirm" function under Ops Mode programming causes an infinite loop without generating any LocoNet traffic.

To reproduce: In a configuration with a real LocoNet command station, open the Roster.  Open an entry for a RR-CirKits MotorMan.  (MotorMan need not be present on LocoNet!).  On the "CVs" tab, actuate the "compare" button for any of the readable CVs.  This will cause an infinite loop, which the JRE will decide to break.  A window likely will pop-up showing the two routines which are calling each other repretitively.

This is seen with JMRI/JMRI master code updated from head on 29Nov2017, but the source code history implies this problem has been around for a long time.

It appears to be caused by a mis-implementation of, or "placeholder code" in, the `confirmCV()` method in `LnOpsModeProgrammer.java`.

bobjacobsen, 2017-11-29:
Happy to fix this, but what triggered it?  AFAIK, the "compare" function was removed from DecoderPro a long time ago.

The back story:  Back when paged mode was the thing, it took a long time to read a CV value.  "compare" was proposed as a way to get a check on a (thought to be) known value much more quickly. Instead of reading 0,1,2,...N, just try a read of the current N and return.  

But no manufacturer wanted to put it into their command station code.  (There was an EasyDCC EPROM with it, but I don't think that was ever shipped in new command stations, and CVP didn't want to tempt people to ask for updates)  

So now it's not really different from read.  Hence, in the interest of simplification, there's been some slow work at removing it.  The basic infrastructure is deprecated, and I thought that the GUI-level uses had been removed.

Where is it still hanging around?

Thanks

devel-bobm, 2017-11-30:
In the symbolic programmer, on the "CVs" tab.  There's a "Compare Full Sheet" button that appears only for this "tab".

And in the table of CVs, Readable CVs have individual "Read" buttons.  Writable CVs have their own "Write" buttons.  And Readable CVs also have their own "Compare" buttons.

This seems to be true across a small variety of decoders that I've recently tried: LokPilot V4, RR-CirKits MotorMan, the Digitrax_DS64, Digitrax_BDL168, Digitrax_PM4, and Digitrax_SE8c.

dheap, 2017-11-30:
@devel-bobm As far as I know, the "Compare" buttons all use `readCV()` rather than the deprecated ~`confirmCV()`~. I don't have time to check at present.
Certainly few, if any, command stations expose the NMRA Direct Mode "verify Byte" instruction to a a JMRI interface.

devel-bobm, 2017-11-30:
It appears that `LnOpsModeProgrammer` is the only "Ops Mode" implementation of `compareCV(String CV, ...)` that does not simply report "not supported".

dheap, 2017-11-30:
Most systems don't support Read or Confrm in Ops mode. But I can see where this may be applicable in this case. But I'm still surprised that the "Confirm" buttons actually try the deprecated ~`confirmCV()`~ methods anyway.

My computers are all shut down due to electrical storms (we're in storm season now), so can't follow this up.

devel-bobm, 2017-12-01:
Pull Request #4486 includes an attempt to fix this problem.

closed 2021-04-14 by stale[bot]