JMRI/JMRI #4491: Issues with "compare" in symbolic programmer Service Mode

dheap, 2017-12-01:
Assisting a club member with DecoderPro training today and found an issue with "Compare" of indexed CVs.
`2017-12-01 16:07:36,670 symbolicprog.CvValue                  WARN  - Exception during CV read: java.lang.NumberFormatException: For input string: "16.2.458" [AWT-EventQueue-0]
`
His system was Digitrax PR3 and V4.9.6, but is replicated on my system with HEAD of master, NCE Serial and NCE Simulator.

I'm almost certain it was fine in the past but don't have time to run a `git bisect` tonight.

@devel-bobm @bobjacobsen This could be connected with #4488.

devel-bobm, 2017-12-01:
For the LocoNet case, `SlotManager` handles only CV Numbers of `int`.  It does not support `String`s.  The `String` to `int` process is likely handled by the facades.

`MultiIndexProgrammerFacade` implements `readCV()` and `writeCV()` but does not implement a `confirmCV()` method.  It seems to me that it will need to handle the parsing of the `String` CV number before passing the operation on to the connection's programmer which implements only the methods which accept an `int` CV number.

dheap, 2017-12-01:
`MultiIndexProgrammerFacade` extends `AbstractProgrammerFacade` `AbstractProgrammerFacade` implements `Programmer`.
~`writeCV(int CV, ...)`~, ~`readCV(int CV, ...)`~ and ~`confirmCV(int CV, ...)`~ are all `@Deprecated` in `Programmer`.

I would read this to say that new and modified code should use/implement the non-deprecated `writeCV(String CV, ...)`, `readCV(String CV, ...)` and `confirmCV(String CV, ...)` methods. 

`MultiIndexProgrammerFacade` implements `writeCV(String CV, ...)` and `readCV(String CV, ...)`. It should also override `confirmCV(String CV, ...)`. It does not implement or directly invoke the `@Deprecated` methods. The `String` parsing is handled by either the underlying physical programmer or `AbstractProgrammer`, which the physical programmer `extends`.

dheap, 2017-12-01:
Looking again at my initial comment, I suspect the "Compare" code is invoking ~`readCV(int CV, ...)`~ and hence bypassing the `MultiIndexProgrammerFacade` code.

Can't check tonight.

closed 2020-02-10 by dheap