JMRI/JMRI #4508: LayoutEditor: Add "decorations" to TrackSegments and PositionalPoints (Anchors, End Bumpers, Edge Connectors)

geowar1, 2017-12-04:
Decorations include: tunnel entrances/exits, bridge entrances/exits, etc. for track segments; end bumpers, rail gap markers, single/double in/out arrows for positional points, etc.)
Note: In the XML file the "decorations" are just a comma separated string of decoration strings. Unknown decorations are ignored. To flip the direction (in<->out, entrance<->exit, etc.) precede the decoration string with an exclamation point ("!").

geowar1, 2017-12-07:
Just want to share some of the work I've done so far...

[image]
[image]
[image]

As always: Questions, suggestions, feedback, etc. welcome.

geowar1, 2017-12-13:
Latest PR #4554

geowar1, 2017-12-14:
[image]

pnbgit, 2017-12-14:
All looks good, and nicely backwards compatible with the "none" options.

Can I suggest defaults of "none" for the arrows on edge connectors, and for end bumpers the default should be the same width as the adjoining track and length adjoining track width x 6 perhaps?  For hidden track segments, defaults for everything should be none.  That would produce something that looks about right in most cases without the user having to fiddle with it unless they want to.

geowar1, 2017-12-14:
On Dec 14, 2017, at 11:02 AM, Pete Brownlow <notifications@github.com> wrote:
> Can I suggest defaults of "none" for the arrows on edge connectors, and for end bumpers the default should be the same width as the adjoining track and length adjoining track width x 6 perhaps? For [...]

Overall the defaults are "none". If you don't opt-in you don't get any of these decorations.

I can tweak the bumpers to match the width… I don't quite understand what you mean by "length adjoining track width x 6 perhaps".

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

pnbgit, 2017-12-15:
The length of the cross stub on the end bumpers needs to depend on the track width.  For example, if the track width is 6,  the a cross stub that is 6 long won't show at all, to maintain the same relative length it should then be 18 long.

geowar1, 2017-12-15:
The length & width is not hard wired; it’s user specified. You can make it any length/width you want. I can base its default/ initial length/ width on some other value: you think it should be double the track width?

Right now I’m drawing the end bumper as just a line across the rails. I think I’d like to change that to be more like the prototypes: kinda like a saw horse leaning toward one end that has a bumper plate. Top down would look like two stacked “V”’s with their points connected by a line that then slightly extends to a short cross line (the bumper plate).

KenC57, 2017-12-15:
Since the code could see the track width coming to the end bumper, doing
something like 120% of width should be an easy change. I just guessed the
ratio, that may need checking against what defaults do. Then again maybe
being a fixed number added to track width is better. Then you wouldn't have
issues with tracks close to each other. At least not as much of an issue.
Doing the ratio could be bad with lots of parallel tracks.

-ken c

geowar1, 2018-01-05:
See #4613…

icklesteve, 2018-06-04:
Hi @geowar1 

The decorations features are brilliant, I especially like the bridge rendering :-)

I do think that this warrants a separate page within the LE documentation featuring some of the graphics now available ( Using a few different track widths / background colours etc. ? ).

The images above in this post really showcase the features well, would be good to have all of the settings used next to the image to help users get the look they're after?

( eg. negative length number to get buffer far end of track trick ) 
For the effect I was after, this kind of works :  width : 3 length -6 , though would have preferred them to be a little taller, they were getting wider? Minor niggle though, good fun playing with them :-)

[image]

Am still a bit confused re. “length”, I first presumed it was extra length, as in length of track!  It’s not really length as in screen height either ( as opposed to width )?
Enough playing around and I got it looking ok :-)

Also found a few mini bugs relating to negative numbers raised on #5422 

I’m sure there will be some very funky looking LE panels in a few months time :-)

Many thanks for your time invested in this,
Steve

pnbgit, 2018-06-11:
See also #5449 #5450 #5451 #5452