JMRI/JMRI #4512: Internal setting confusing to many ordinary users

billchown, 2017-12-05:
While the "Internal" setting is useful to some, it confuses many.
Can we please identify it differently than a typical track or layout connection. Instead of showing Internal as Green in the footnote text on DP, could it be Yellow/Orange.
As a support resource for DCC, JMRI and specifically SPROG users, I often get questions about no connection, seeing 123 as a response, etc. and recently all of those are due to the Defaults settings in Preferences being flipped to Internal by some means. With no easy visibility of its condition, many users of JMRI just see this as another complicating confusion, but I think it could be remedied.

bobjacobsen, 2017-12-15:
I've started work on this along the following lines:

When a system connection fails to initialize, it ends up providing null references for some of the managers it would usually create. When the defaults configuration is loaded, line 224 (or so) of jmri.managers.ManagerDefaultSelector detects that and handles it by just logging and leaving the default unset, which later translates into “use the last system”, which in turn is generally Internal.

So the plan is to:

- Move toward "Defaults only change at startup" in both implementation and documentation

- When a system fails at startup, we leave the default set at the blown system. The instance of the program will be badly damaged, and probably behave badly if there are e.g. panel files and/or scripts that need e.g. a command station for SignalHeads, etc.  But that wasn’t going to do the right thing (talk to the layout) anyway.

- When a system fails at startup, the defaults pane is not marked as dirty (e.g. not forcing save of preferences) but still does the user to change the default(s) if this is a permanent change (e.g. not a temporary error)

How to show this to the user on the defaults pane?

Start with the existing behavior

- Each connection that has a manager that was pre-configured default shows as a selected and active radio button

- Each connection that has a manager that was _not_ pre-configured default shows as a not-selected, active radio button

- Each connection does not have a manager and wasn’t pre-configured shows as a not-selected, not-active radio button - you can't choose this one because it's not there (this could arguably be no button, but that's now what we have now)

Those already exist, so now add

- If the connection does not have a manager and it _was_ pre-configured, e.g. it was supposed to work but the connection was blown, that’s a selected, not-active radio button. You can select away from it via _another_ selection in the category, e.g. ask for another connection to provide that default.  But you don’t have to; it’ll still show what’s in your preferences files. You can’t select back to it while it’s inactive, so if somebody manually switches away with a non-initializing connection, they have to wait to have it working before they can select it. (Otherwise, the initial automatic selection of defaults for e.g. LocoNet+CMRI won’t work)

- To clarify the saved/dirty state, add a save/restart icon on each column to show when a default has been changed, but not (yet) written

KenC57, 2017-12-15:
For dealing with the 'blown system' case, do we (or can) have a pop-up to
prevent running the startup actions (panels, scripts, etc...) or at least
ask the user if they would like to skip that?? I know most of the time I'm
starting with a 'blown system', I don't want to load any of those things.
-ken c

rhwood, 2017-12-15:
> For dealing with the 'blown system' case, do we (or can) have a pop-up to
prevent running the startup actions (panels, scripts, etc...) or at least
ask the user if they would like to skip that?? I know most of the time I'm
starting with a 'blown system', I don't want to load any of those things.
-ken c

We already do show such a message, and prevent later startup. The problem is that the defaults get changed as a result, and this is non-obvious.

billchown, 2017-12-15:
An additional indication that I hope can be accommodated is to change the colored messages on the DP 3 screen."Internal" means "Not doing anything" in the case of SPROG, and I'm suspecting some other system connections, yet since the 4.7/4.8 release sequence we show the Internal selected non-available mode as Green, where prior to that it showed as Red.A neophyte user cannot comprehend the "Internal", and so seeing two sets of Green messages assumes that they can use both. In the case of the SPROG, this is not true, and has become the most common support call issue that I have had to deal with over the past year. 
Many users that I talk to have no knowledge of any of the Preferences screens beyond Connection. Please can Internal selections that manifest as a control of the availability pf "Service" or "Operations" mode be clearly indicated on the visible screen, as we had in the past. I suggested Yellow/Orange, but I would be happy with anything other than green!
ThanksBill
 Bill Chownorrrbrit2004@yahoo.comhttp://sprog.us

bobjacobsen, 2017-12-16:
> On Dec 15, 2017, at 9:29 AM, Bill Chown <notifications@github.com> wrote:
> 
> An additional indication that I hope can be accommodated is to change the colored messages on the DP 3 screen."Internal" means "Not doing anything" in the case of SPROG, and I'm suspecting some other system connections, yet since the 4.7/4.8 release sequence we show the Internal selected non-available mode as Green, where prior to that it showed as Red.A neophyte user cannot comprehend the "Internal", and so seeing two sets of Green messages assumes that they can use both. In the case of the SPROG, this is not true, and has become the most common support call issue that I have had to deal with over the past year. 
> Many users that I talk to have no knowledge of any of the Preferences screens beyond Connection. Please can Internal selections that manifest as a control of the availability pf "Service" or "Operations" mode be clearly indicated on the visible screen, as we had in the past. I suggested Yellow/Orange, but I would be happy with anything other than green!

With the proposed solution, this should never accidentally happen.  Instead of getting internal when something goes wrong, there just won’t be a programmer available.

I’ll try to do the color coding anyway, just in case “never” turns out to be optimistic. 

But the goal of the current work is to get selection of “Internal” back under the user’s conscious control.

Not on my current list is to revamp the warnings when a connection fails to start. If people think those should be updated (they’re easy to test!), please open a new GitHub issue with the preferred new behavior.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

bobjacobsen, 2018-05-21:
This has been on hold for a while due to not really understanding how it should best work in the context of some of the above.

There's a [conversation on jmriusers](https://groups.io/g/jmriusers/topic/19672278) about it just now.

bobjacobsen, 2019-01-13:
Believed to be improved in JMRI 4.14

closed 2019-01-13 by bobjacobsen

billchown, 2019-01-14:
vastly improved, and thank you!
Bill

Bill Chownorrrbrit2004@yahoo.comhttp://bchown.com/