JMRI/JMRI #4514: Unable to delete some Sensors, Turnouts et al.

Pugwash1, 2017-12-05:
When a user attempts to delete one of these items nothing appears to happen. 
Unfortunately that is not the case. BlockBossLogic takes a silent dive for the team and no one notices. The delete process iterates over all the listeners via "AbstractManager.fireVetoableChange" and hits the BlocKBossLogic" item that isnt. The one created by the method annotated with    "Create a default object, without contents. Used when registering a dummy with the configuration system.", it contains nulls, a lot of nulls, so the call fails on a N.P.E. that no one listens to.

bobjacobsen, 2017-12-05:
Happy to work on this if somebody posts a recipe for recreating.  Just tried to delete a Sensor in use by SSL without problem.  Maybe a panel file that can be loaded and then shows the failure when a specific item is deleted?

Pugwash1, 2017-12-06:
@bobjacobsen Here you go. See if you can delete LT2, or any other turnout. The BlockBossLogic line number that it dies on is 1267, message.append....

[IsabelleAndMicheals_SSL.zip](https://github.com/JMRI/JMRI/files/1533701/IsabelleAndMicheals_SSL.zip)

bobjacobsen, 2017-12-06:
Thanks. See it now.

KenC57, 2017-12-06:
Bob,

Why would I not get a NPE displayed on my system?? I had tried the file and
saw 'nothing' happen. I also set a break point this morning when I saw your
fix. While is see the null, it doesn't throw a NPE. So, how did you find it
without the NPE??
-ken c

bobjacobsen, 2017-12-06:
> Why would I not get a NPE displayed on my system?? I had tried the file and
> saw 'nothing' happen. I also set a break point this morning when I saw your
> fix. While is see the null, it doesn't throw a NPE. So, how did you find it
> without the NPE??

The NPE is happening off the Swing thread.  (Not arguing that’s a good or bad thing, just that it is)  That propagates up and kills that thread, but in the process it’s not encountering JMRI code that logs it, so it doesn’t get logged. Steve’s hint was enough to track that down.

closed 2017-12-06 by Pugwash1

pnbgit, 2017-12-09:
As this issue is closed, does that mean this one is now fixed?  I can't see any associated pull request (should I?)   I'm just helping somebody out who has this problem, and I'm still seeing it with build #4 of 4.9.7

bobjacobsen, 2017-12-09:
One NPE problem was fixed in #4516.  But there certainly could be other problems, particularly if you're still seeing this. If you've got a regrettable case, perhaps open another Issue?

One thing I noticed is that the UncaughtExceptionHandler no longer seems to be being initialized.  That seems to be (but I haven't confirmed) why this NPE wasn't visible.  That should probably also be another Issue.

pnbgit, 2017-12-09:
Hi Bob, Freudian slip that, regrettable  :-)

But I do have a repeatable case.  I will create a new issue.

Pugwash1, 2017-12-09:
The fix was not in the build of 4.9.7 as of this morning. It is in master though, and works.
Steve G

On December 9, 2017 3:27:36 PM EST, Pete Brownlow <notifications@github.com> wrote:
>As this issue is closed, does that mean this one is now fixed?  I can't
>see any associated pull request (should I?)   I'm just helping somebody
>out who has this problem, and I'm still seeing it with build #4 of
>4.9.7

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

bobjacobsen, 2017-12-09:
> The fix was not in the build of 4.9.7 as of this morning. It is in master though, and works.

Oops. Steve is right.  When the fix was recreated for inclusion in 4.9.7, it wasn't done right.  So it's on master branch, but not in release-4.9.7

KenC57, 2017-12-13:
So if a user wanted to test their panel against this fix, what do they download??
-ken c