JMRI/JMRI #4519: Layout Editor changed behaviour: Difficult for users to set transparent background for labels and icons, and change to transparent does not save

pnbgit, 2017-12-06:
The default background for a label is clear (transparent).
If the user changes this to something else, it used to be quite easy to select a transparent background again:

[image]

However, since the change to using the colour swatches interface, it is now not obvious how to select a transparent background:

[image]

You now have to go into the colour set dialog, which shows the swatches and still does not offer a transparent option:

[image]

So you have to select another tab, such as HSV, and drag the transparency slider all the way to the right:

[image]

I don't think the average user is going to work this out.

To add to the difficulty, if you change the background to transparent in this way, and then save the panel, when you reload the panel it has reverted to the non-transparent background, so the transparency attribute has not been saved.

Could I suggest an additional button or checkbox in the initial dialog to set a transparent background to avoid the need to go into the multi-tabbed colour dialogs to achieve this?

Perhaps something like this:

[image]

Thanks,

Pete

geowar1, 2017-12-06:
The problem here is that those swatch colors are saved as screenColorType which is defined as a hash "'#') followed by exactly six (6) hex digits (IOW: "#RRGGBB").
That excludes the Alpha (transparent) channel information. Not how I would have done it but I didn't do it… (and "code wins" or in this case loses).

On Dec 6, 2017, at 11:58 AM, Pete Brownlow <notifications@github.com> wrote:
> The default background for a label is clear (transparent).
> If the user changes this to something else, it used to be quite easy to select a transparent background again:
> ...
> However, since the [...]

> You now have to go into the colour set dialog, which shows the swatches and still does not offer a transparent option:...
> ...
> So you have to select another tab, such as HSV, and drag the [...]

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

pabender, 2017-12-06:
@pnbgit 

I'm more concerned about saving and restoring the alpha (transparency) than anything else you raise here.

With the old selection mechanism (which we can't go back to, they were not got for internationalization among other things) there was a specific value labeled "clear".  That was created by setting the color to be nothing.

With the new selection mechanisms, which are Java standard color choosers, we actually have a lot more control over the color selection than we did before, but "no color" isn't an option (it really should never have been an option).

Adding alpha values to what we store and retrieve in the files is something I'm working on, but that's not entirely the issues here, because the colors are stored differently than in other parts of JMRI....

@geowar1 

Colors for the text labels are actually stored as:
'''
    <positionablelabel x="0" y="0" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" text="hello world" size="12" style="1" red="51" green="51" blue="51" hasBackground="yes" redBack="153" greenBack="153" blueBack="0" justification="centre" class="jmri.jmrit.display.configurexml.PositionableLabelXml">
      <tooltip>Text Label</tooltip>
    </positionablelabel>
'''

For the text, it's that "hasBackground" attribute that was previously used to set the background on and off (this differs from some other code...)

pabender, 2017-12-07:
For the GUI part of this, we will make some accommodation when we address #4392.  

#4523 should take care of the color storage.  I'm going to go ahead and close this once that is merged, but we'll reference this issue from #4392.

closed 2021-04-14 by stale[bot]