JMRI/JMRI #4529: Table View All tab missing some items, sort suspected

KenC57, 2017-12-07:
First it seems the sort method now tries to split up a field so the number follow like number sort instead of straight character sorting. In this file the example is IS141, IS141R, IS141N and IS141A values. The would come somewhere after IS79* instead of the old way of before IS4A1.

The attached file shows that many entries are not showing in the 'All' tab but show in the 'Internal' or the other way around. I tested using 4.9.8 and nothing in the console about this.
[CWLayoutSensors.zip](https://github.com/JMRI/JMRI/files/1540081/CWLayoutSensors.zip)
The original user is leo pesce and the user group thread is "JMRI 4.9.6 problem in the sensor table".
-ken c

bobjacobsen, 2017-12-07:
I took a quick look at the code.  The key items seem to be `jmri.util.SystemNameComparator` and `jmri.util.AlphanumComparator`, which implement `java.util.Comparator`.  

What _might_ be happening is that their `compareTo(..)` methods don't quite obey the contract.  They're supposed to return only -1, 0, +1 but it looks like they can return larger negative or positive numbers.

geowar1, 2017-12-07:
For the sorting SystemNameComparator.java is what you're looking for… (Not that I understand why you'd want IS141* to sort before IS4A1.)
For the rest… not a clue.

KenC57, 2017-12-07:
To me it looked like it was not doing a lexical (char by char) sort anymore but trying to keep alpha and numeric fields grouping and sorting together. To the non-programmer, seeing the sensors as IS28, IS29, IS30, IS31 makes more sense than IS28, IS29, IS3, IS30, IS31, IS32.

I'm used to the old way, lexical, but I guess somebody 'improved' the sort. From Bob's comment I think it may be flawed.
-ken c

bobjacobsen, 2017-12-07:
Looking at it more, `SystemNameComparator` has a _bunch_ of dead code (after `if (true)`).  Ignoring that, it seems to be stripping the 1st two (specifically two) characters off the name and then doing a `AlphanumComparator` comparison on the rest.  So, if it's the comparison that's the problem, `AlphanumComparator` is the place to look.

But I don't see how that can be the problem, because what you're seeing is that the rows are not showing at all. Why do you think it's a sort issue?

geowar1, 2017-12-08:
> @bobjacobsen Why do you think it's a sort issue?

I don't think that the disappearing items is a sort issue… but the 1st paragraph of the 1st message said: "First it seems the sort method now tries to split up a field so the number follow like number sort instead of straight character sorting. In this file the example is IS141, IS141R, IS141N and IS141A values. The would come somewhere after IS79* instead of the old way of before IS4A1.."; that sounds like a sorting issue (and IMHO the correct behavior).

KenC57, 2017-12-08:
My reason for guessing the sort was I saw some irregularities in the order. So if while sorting it was over-writing entries in the index was my guess.

bobjacobsen, 2017-12-08:
Getting back to the original problem statement, what are examples of things that are not visible in the "All" tab?

I've created a profile with LocoNet and OpenLCB Simulator connections, and the file loads fine. I don't see any missing items, but perhaps I'm not looking at the right one.

KenC57, 2017-12-08:
The example I've seen is the IS141* sensors. When I'd sort the systemNames
to see what's there only some of them show. On my system only one showed in
the All, but all of them showed in the Internal. But the original reporter
said he saw the reverse, they all showed in the All, but only IS141 showed
in the Internal.
-ken c

KenC57, 2017-12-20:
Here is a posting from a user that sounds like the same issue, but better explaination of what seems to happen.
-ken c
Missing Sensor problem occurs after 4.7.7 and is still present in 4.10.0

The Sensor Table view when All sensors are displayed has a new and defective sort/display algorithm. The causes erroneous sensors displays for all pick-list and table displays which have multiple connections. Since many features including a change of connection reference use the All Sensor view this problem essentially disables the feature.

The All Connections table view the Internal system names are evaluated as numeric rather than alpha-numeric if the leading characters after the "IS" is numeric. Subsequent alpha characters are discarded and if it resolves to a duplicate with a previous number the sensor is discarded.

The problem can be seen with a system name sort comparing an IS Sensor table view and the same table with All table view.

Example: JMRI 4.10.0

Add Sensor to IS connection view.

123 User Name 123
123A User Name 123A

The second entry will not appear in the All connection view

Additional problem: Pressing the Create Sensor button removes the Add Sensor Box requiring reselecting the Add Sensor box for each sensor if a list of sensors is being added. This works correctly and retains the Add Sensor box with 4.7.7

David Parks
Los Altos, CA

bobjacobsen, 2017-12-20:
I'm working on this.

bobjacobsen, 2017-12-21:
git bisect finds the cause in PR #3907
```
commit a78aee4035e76a2154da3888bfae1331c0ec5f8e
Author: George Warner <geowar1@mac.com>
Date:   Thu Aug 10 10:52:06 2017 -0700

    Alphanumeric sorting system and user name columns in tables
    
    Sorry, couldn’t take it anymore…
```

That commit seems to have rokenb `java/src/jmri/util/SystemNameComparator.java` at a minimum.

geowar1, 2017-12-21:
Looking at Alphanumeric sort there are several places where it sets results thusly: `int result = length1 - length2;`. There are definitely conditions that would return values other than -1, 0 and +1. Perhaps the result should be pinned between +/- 1: `return MathUtil.pin(result, -1, +1);`

geowar1, 2017-12-21:
According to the Comparator documentation at <https://docs.oracle.com/javase/8/docs/api/java/util/Comparator.html>: “Compares its two arguments for order. Returns a negative integer, zero, or a positive integer as the first argument is less than, equal to, or greater than the second.”. This says that the AlphaNumeric compare returning values other than -1, 0 and +1 is acceptable.

bobjacobsen, 2017-12-21:
The problem is that the comparator doesn't obey the "equals(..) <-> compare(..)==0" contract.  Comparing e.g. IS100 and IS100A returns 0, but they're not equal.

I'm going to do some cleanup on the entire set of jmri.util Comparator classes and particularly their tests.

PR #4607 (2017-12-23): Really basic fix to Comparator for system names
Resolves issue with some Sensors not appearing in the Sensor table, broken since in 4.9.2, see #4529 for further background.

Does not address system-specific comparison, esp. proper handling of complicated hexadecimal system names. System prefixes of more than 2 characters (e.g. that weird long DCC++ prefix) might not sort in system-prefix-order.

Closes #4529

This is a reposting of the change in (partially, but only partially, redundant) #4605 with minimal changes to get it through faster
closed 2017-12-23 by bobjacobsen