JMRI/JMRI #4531: Custom signal system location problem

dsand47, 2017-12-07:
If a custom signal system has the same name as a supplied signal system **AND** it is located in **_preference:resources/signals_**, the installed system is selected.  If the custom signal system is located in **_preference:xml/signals_**, then it is selected.

Custom signal systems with unique names still work when located in **_preference:resources/signals_**.

Dave Sand

rhwood, 2017-12-08:
Looking at the code, it appears that the aspect.xml file of custom signal systems with the same name as a supplied system should never be selected; i.e. that loading a custom signal system with the same name as a supplied system from __preference:xml/signals__ is considered a bug per comments in the code.

Are you wanting to be able to override the aspect.xml file contents? (It is documented that everything _but_ the aspect.xml is overridable.)

dsand47, 2017-12-08:
I am helping a group that copied an existing signal system and modified it.  I don't know the full extent of the modifications, but a significant modification is adding entries to the imagetypes list, which resides in aspects.xml, along with changes to the appearance xml files and icons.  Their current process is to copy the entire custom signal system to the JMRI install location when they change JMRI versions.

I helped convert everything to use a common user files location in Dropbox, but then the imagetypes problem came up.  Now they are faced with the decision to revert to the bulk copy method or remember to add 3 lines to aspects.xml each time there is an upgrade.  A more drastic solution is to rename the custom signal system so that it is unique.

The custom signal system was originally at preference:resources.  When that wasn't working, I moved it to preference:xml and it appeared to work for a bit and then stopped.  Adding the 3 lines to the production aspects.xml fixes it.  Now I just have to remember not to commit the change.

Dave Sand

pnbgit, 2017-12-08:
As a thought for the future, if we added a user interface for doing custom icons for signalling systems, then it could automatically put them in the right place in preference:,  at the moment the user just has to decide where to put them and then manually edit the signalling system XML files.

PS: Dave - all of a sudden I am getting an error when trying to email you - "<your email address> asked to not receive your emails."  Not sure if the problem is your end or mine.

closed 2018-02-23 by dsand47