JMRI/JMRI #4541: JMRI digitrax throttle will not set a locos slot to in use if the slots are all empty in the slotmonitor

Walter71342JMRI, 2017-12-10:
To create this I "Release" all the slots in the Slot monitor and restart JMRI. Repeating that until all the slots are not displayed.
Next open a jmri throttle and select the address of the loco on the main.  At that point the JMRI throttle can not control the loco/decoder.
Then doing a "Release" on the JMRI throttle and then a "Set" again allows the loco to be control the loco

I think JMRI does not understand what the slot being "Free" means"

Here is a loconet trace of what happens. 
18:53:55.562: [BF 01 6A 2B]  Request slot for loco address 234.
18:53:55.586: [E7 0E 01 03 6A 00 00 07 00 01 00 00 00 78]  Report of slot 1 information:
	Loco 234 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
18:53:55.591: [EF 0E 01 03 6A 00 00 07 00 01 00 71 02 03]  Write slot 1 information:
	Loco 234 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
18:53:55.596: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
18:53:55.602: [BA 01 01 45]  Set status of slot 1 to IN_USE.
18:53:55.602: [B4 3A 00 71]  LONG_ACK: The Move Slots command was rejected.
18:53:59.083: [A1 01 02 5D]  Set loco in slot 1 direction Forward F0=Off F1=Off F2=On F3=Off F4=Off.
18:53:59.294: [A1 01 00 5F]  Set loco in slot 1 direction Forward F0=Off F1=Off F2=Off F3=Off F4=Off.
18:54:02.541: [A1 01 10 4F]  Set loco in slot 1 direction Forward F0=On F1=Off F2=Off F3=Off F4=Off.
18:54:05.017: [A0 01 64 3A]  Set speed of loco in slot 1 to 100.
18:54:06.923: [A0 01 00 5E]  Set speed of loco in slot 1 to 0.
18:54:36.966: ABOVE DOES NOT CONTORL LOCO
18:54:50.976: ,
18:54:51.601: ,
18:54:56.925: [A0 01 00 5E]  Set speed of loco in slot 1 to 0.

ON THE JMRI THROTTLE I CLICKED 'RELEASE', 

18:55:38.969: [B5 01 13 58]  Write slot 1 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
18:55:38.974: [B5 01 23 68]  Write slot 1 with status value 35 (0x23) - Loco is Not Consisted, Idle and operating in 128 speed step mode.
18:56:33.875: ON THE JMRI THROTTLE DO A RELEASE OR DISPATCH ABOVE

NEXT I 'SET' THE THROTTLE ADDRESS ON THE JMRI THORTTLE

18:57:28.917: [BF 01 6A 2B]  Request slot for loco address 234.
18:57:28.937: [E7 0E 01 23 6A 00 10 07 00 01 00 71 02 3B]  Report of slot 1 information:
	Loco 234 is Not Consisted, Idle, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=On, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
18:57:28.947: [EF 0E 01 23 6A 00 10 07 00 01 00 71 02 33]  Write slot 1 information:
	Loco 234 is Not Consisted, Idle, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=On, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
18:57:28.952: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
18:57:28.952: [BA 01 01 45]  Set status of slot 1 to IN_USE.
18:57:28.972: [E7 0E 01 33 6A 00 10 07 00 01 00 71 02 2B]  Report of slot 1 information:
	Loco 234 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=On, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).

NOW I CAN CONTROL THE ;LOCO WITH THE JMRI THROTTLE

18:57:59.016: [A1 01 12 4D]  Set loco in slot 1 direction Forward F0=On F1=Off F2=On F3=Off F4=Off.
18:57:59.566: [A1 01 10 4F]  Set loco in slot 1 direction Forward F0=On F1=Off F2=Off F3=Off F4=Off.
18:58:04.619: [A0 01 21 7F]  Set speed of loco in slot 1 to 33.
18:58:06.956: [A0 01 00 5E]  Set speed of loco in slot 1 to 0.

THE PROBLEM I SEE IS THAT AT 18:53:55.591: JNRI WRITEs THAT THE SLOT IS FREE AT 18:57:28.947   JMRI WRITE THE SLOT AS IDLe  SHOULDN'T WE KNOW THAT FREE MEANS IT CAN BE SET TO “IN-USE'?

Walt

Pugwash1, 2017-12-10:
See #4293 . It's all related. To get extended slots to work we need to write the throttle ID. This has to happen only after the slot in use has been acknowledged. 4293 will move us forward. The fundamental issue is that digitax never expected that anyone would ask for a throttle faster than a tortoise with a limp. There is a fun game you can play with a dcs51 and a dt401 entering loco numbers slowly.

Walter71342JMRI, 2017-12-10:
Steve:
I would be more than happy to test on my DCS100. Is the change in the
current build? if yes I will load the bleeding edge and test it.
Walt

devel-bobm, 2017-12-13:
This is one aspect of a problem I previously identified which relates to how a JMRI "throttle" and perhaps its LocoNet "throttle" can retain knowledge of a LocoNet slot even though the LocoNet slot is released.  I do not currently find an open "issue" where I noted the behavior.  And I do not understand how the current code-base allows something to retain use of a slot, as I believe (without looking at code) that the slot manager attempts to dispose LocoNet "throttle"s when the slot is released.  I have not had time to look into this failure-mode.

Pugwash1, 2017-12-17:
The status of a Digitrax slot in the command station has no effect on anything much. Weird I know.  Take a DCS51, factory reset it (sw 39->c), send a set slot 3 to speed 13 (A0 03 0D 51) and watch a DC loco (address 0) go. Write a slot with a dcc loco address,and any status, and set the slot speed, it moves, doesnt have to be "in-use", the default "free" works, as does "common". Any slot with a speed > 0 gets sent to the rails. JMRI updates its slot based on what it sees on the Loconet, so if it sees a speed message it sets the slot speed, even JMRI doesnt check the status. Most Digitrax command stations do very little validation, they rely on the throttle doing that.

devel-bobm, 2017-12-17:
@pugwash1,
You wrote: " Take a DCS51, factory reset it (sw 39->c), send a set slot 3 to speed 13 (A0 03 0D 51) and watch a DC loco (address 0) go."

What mechanism did you use to apply the factory reset?

Did you watch the DCS51 display while performing the "factory reset?   I believe that the DCS51 display will chane and briefly show the firmware revision number while doing the reset.  Did you observe this behavior while performing the "factory reset"?

Did you use JMRI's Configure Command Station tool to perform the Factory Reset?  If so, it may be necessary to remove and restore DCS51 power in order for the operation to "take".  I believe that the same statement applies to other Digitrax command stations, as well as the other three similar OpSws which clear various command station internal storage.  I believe that it might be sufficient to switch the "run/ops/sleep" mode switch to "sleep" and then back to "run" in order for the operation to "take", for those command stations with such a switch.

Pugwash1, 2017-12-17:
I closed OpSw 39 as usual, "8888" and then "-cs-" appeared. Removed power, powered back up. Digitrax command stations don't have to be used to allocate slots, or even do slot management, thats why they don't check the status.  You can do stuff like get a loco on a DT402, then from a pc do a slot write to change the loco address and the dt402 display doesn't change, but it still controls the slot and the new loco moves. You can try this using the "Send Loconet Packet" screen. 
Using this knowledge, I tried changing the aquistion process to , read all slots once on startup, then to aquire  syncronized(scan slots for the loco or get empty, set in status in-use, address, step mode), write slot, give slot to throttle (where I removed the null slot move). I can load 20 throttles toot suite. I even tried 2 pc's, one using the first 10slots and the other the next 10. No conflicts. DT402 aquisition still worked. Now how does that fit in with extended slots?

devel-bobm, 2017-12-17:
@pugwash1,

I am highly confused by what you are trying to accomplish or prove....

There is some misleading info and some inconsistent data in your report.

You wrote: "Digitrax command stations don't have to be used to allocate slots, or even do slot management, thats why they don't check the status. "

Who is "they"?  What "status" is it that "they" don't check?  In what scenario is it reasonable not to use a LocoNet-based command station to allocate slots, if you are going to use slots?

If there is no LocoNet command station, there are no slots.  If there is no command station, one cannot allocate slots.  That is a specific funciton of the LocoNet-based command station.  A throttle can query slots until the cows come home, but if there is no command station providing the slots, then the throttle isn't going to get a reply with slot data.  As an example, you cannot "dial up" a loco on a DT402 if there is nothing connected to LocoNet which fills the "command station" role.

Secondly, There is something odd going in if cleared out the DCS51 command station's slots and if you sent a "set slot n speed to 37" and the DCC track signal actually changed for a loco.  It implies either that something else "re-allocated" the slot, without your knowledge, or that the slot wasn't really cleared.  I can envision ways that "OpSw-based slot clearing" would leave the old data in the slot, and leave the command statoin in a situation where it could still update the DCC track signal based on LocoNet messages, even though the slot was not "In-Use".  That would be unexpected behavior, though.

But my most important question is: what does this have to do with JMRI?

Pugwash1, 2017-12-17:
Q:I am highly confused by what you are trying to accomplish or prove
A: I am trying to answer your initial question. "This is one aspect of a problem I previously identified which relates to how a JMRI "throttle" and perhaps its LocoNet "throttle" can retain knowledge of a LocoNet slot even though the LocoNet slot is released."
Q: You wrote: "Digitrax command stations don't have to be used to allocate slots, or even do slot management, thats why they don't check the status. " Who is "they"? What "status" is it that "they" don't check? In what scenario is it reasonable not to use a LocoNet-based command station to allocate slots, if you are going to use slots?
A: "they" is   the Digitrax Command Station. The scenario that multiple sources wish to concurrently access new slots, either within the same device or independent devices.

Q: If there is no LocoNet command station, there are no slots. If there is no command station, one cannot allocate slots. That is a specific funciton of the LocoNet-based command station. A throttle can query slots until the cows come home, but if there is no command station providing the slots, then the throttle isn't going to get a reply with slot data. As an example, you cannot "dial up" a loco on a DT402 if there is nothing connected to LocoNet which fills the "command station" role.
A: The Digitrax command station is still connected. But doesnt do all the slot management."That is a specific funciton of the LocoNet-based command station." - No. The specific function is to  "maintain the refresh stack for DCC packet generation". An additional function is to allocate slots.

Q: Secondly, There is something odd going in if cleared out the DCS51 command station's slots and if you sent a "set slot n speed to 37" and the DCC track signal actually changed for a loco. It implies either that something else "re-allocated" the slot, without your knowledge, or that the slot wasn't really cleared. I can envision ways that "OpSw-based slot clearing" would leave the old data in the slot, and leave the command statoin in a situation where it could still update the DCC track signal based on LocoNet messages, even though the slot was not "In-Use". That would be unexpected behavior, though.
A: After factory resetting, every slot has address 0 in it (at least on a DCS51). If you send a Loconet Message "Slot n Speed x" all the command station does is update its slot info and broadcast the new speed to the track on every refresh. It does not validate the slot status.

Q: But my most important question is: what does this have to do with JMRI?
A: Maybe nothing at all. But maybe a lot. Currently the JMRI process for acquiring a slot is code heavy and doesnt work. No more than 1 source can ask for a throttle at the same time. If there is a conflict the outcome can be confusing, no throttle, 2 or more clients on the same throttle, slot MAX after 2 requests.

I just thought you might like to know what I found.

bobjacobsen, 2017-12-18:
> A: After factory resetting, every slot has address 0 in it (at least on a DCS51). If you send a Loconet Message "Slot n Speed x" all the command station does is update its slot info and broadcast the new speed to the track on every refresh. It does not validate the slot status.

I don’t think I understand this.  My understanding is that you’re saying:

> A: After factory resetting, every slot has address 0 in the DCC address field when you read it via the LocoNet (at least on a DCS51). 

(Just changed to make it clearer where the address 0 is)

> If you send a Loconet Message "Slot n Speed x" the command station does is update its slot info and broadcast the new speed to the track on every refresh

So is this saying that the Command Station has somehow remembered the DCC address that used to be in that slot, so it can make up speed packets to that address with speed X?

That doesn’t seem likely to me, so I think I’m misunderstanding what you’re saying here.

Walter71342JMRI, 2017-12-18:
Dealing with address 0, this Offsets the DCC signal on the track to move DC
motors. Why would you send a speed cmd to with out acquiring a throttle
address? Yes, doing a DCS reset with OpSw 36 or higher sets all the Slot
address to 0, But none are Idle, Common, or in Use, they are all Free.
Walt

Pugwash1, 2017-12-18:
Try it.
Take a DCS51, factory reset it (sw 39->c), send a set slot 3 to speed 13 (A0 03 0D 51) and watch a DC loco (address 0) go.
Use the locket debug send to send the raw packets. Obviously you have to use a real command station. Also read the digitax documentation for Loconet it's all there.

Pugwash1, 2017-12-18:
Youre correct about common, sort of. Its actually command station dependent and the order of events. Setting the speed after setting the status a DCS50 sends speed packets to the track for Common and In-Use, but not for idle and free, a DCS51 sends speed packets to the track for In-Use, Free and Idle. If on the DCS51 the status is set to common and the speed to non zero by writing the complete slot, then the speed is placed on the track. Also if I change the slot to common, even if already common, after doing a speed to > 0 the speed starts to be placed on the track. If you send a speed change > 0 to a common slot it does not go to the track, unless you follow it with a direction/function message. Truly weird, maybe something to do with dispatching.. A slot stolen & released will still process messages from the original throttle and put them on the track, depending on the command station type and throttle type. As I said, I dont think the command station does much validity checking.

bobjacobsen, 2017-12-18:
@Pugwash1 Thanks, I'd misunderstood. I thought you were saying that the DCC address had been retained; I hadn't remembered about the zero DC address.

In a perfect world, communications would always be verified against state, validated, etc. But a lot of model railroad hardware is a far cry from that.  I think it's not unreasonable for a LocoNet command station to act as if the attached devices are doing what they're supposed to do:  When "Slot n Speed x" is received, it can assume that the sender has put slot n in the right state before sending.

The alternative is both a much more complicated protocol (what's the command station supposed to  say in this case?  LocoNet is neither a sequenced nor addressed protocol) and much more complicated/voluminous code to implement it.

To get back to the original issue: I'm hoping to get some time and hardware access over the holiday break to figure out what a DT400 and DCS51 actually do. My experience with earlier JMRI LocoNet code is that the spec is a nice reference, but what's really needed is to emulate the Digitrax implementations.

pabender, 2018-01-08:
Closed via #4726

closed 2018-01-08 by pabender