JMRI/JMRI #4545: Regression: 4.9.7 - Changing text label properties font size or bold changes background color

pnbgit, 2017-12-10:
This behaviour was originally reported in #4347. Please refer back to that issue for more details of the problem.

The problem recently came back with build #4 of 4.9.7 - the problem was not present in build #3.

To duplicate the problem:  With a new blank layout editor panel, make the background non white, then add a text label.  The Label background is transparent, as is the correct default.

Now right click the text label and select properties, change something like font size or bold, then click apply and/or OK.   Label background becomes white instead of transparent as it should be.

Given that, to fix it, the user now has to work out how to go into "Set background", switch to HSV or HSL tab and drag transparency slider to 100%,  it would be good if we could fix this one again. 

Since the switch to the swatches color interface, there is no longer any simple way to set the background to clear, you can only set transparency = 100% or alpha = 0% with a number of clicks and drags.

I think the problem is caused by PR #4523, which was fixing the problem that setting the transparency or alpha to a transparent background was not being saved. 

I am wondering if the problem is in this code change, when the background color is null or alpha zero, it now writes a color instead of null as it did before?  

[image]

pabender, 2017-12-11:
Ok, so the problem here is actually that the preview background on the properties popup isn't correctly initialized.  When you press either Apply or OK in the popup, the preview's properties are translated to the actual object.

I'm working on a fix, but I've run out of time for tonight, so it will have to wait for the morning.

pnbgit, 2017-12-11:
Thanks Paul - you've worked out the cause of the problem better than I could.

 Whilst you are in that bit of code, is it possible to add a button in the popup to set the background to clear (transparent) as we had before the change to the swatches interface?  That would make life much easier for a user who wants to change back to a transparent background after changing it to something else. 

At the moment, the user has to go into the set background window, navigate to one of the other tabs, and then drag the slider for transparent to 100% or alpha to 0%.  As well as being a bit of an unwieldy process compared to before, I think that most users will not work out how to do that.

silverailscolo, 2017-12-11:
You might solve this by adding a line like:
```
_textComponent.isOpaque(color.getAlpha != 0);
```
For the UI, not sure if overriding ``JColorChooser``  could achieve this but integrating a Fill:None swatch (square with red diagonal) akin to many drawing apps might be a nicer way than to add yet another checkbox somewhere on the pane:
[image] [image]

pabender, 2017-12-11:
@silverailscolo 

The UI change you suggest is not actually an option, since we don't control the swatches available.  That's a standard widget.  We've got some other things in the works though that we will be able to control (see #4392)

I actually tried calling setOpaque to the object, and that didn't work either.

I've come to the conclusion that the color handling in PositionablePropertiesUtil just needs to be re-written.

There are color choosers that are created, but never shown to the user.

There is an internal class that stores colors and other properties.  The interactions between this internal class and the GUI are confusing (this is partly because the class handles two different types of positionables, ones that have 1 set of properties to display (such as labels) and ones that have 4 set of properties to display (such as sensors).

Also, at least some of the classes that define the positionable objects  define hasBackground(), but the PositionableProperties only ever looks at isOpaque, which is 1) defined in the interface and 2) inherited from the parent object type(s).  

isOpaque() may never look at the actual color choice we use, hasBackground() does.

Oh, and we need many more tests...

silverailscolo, 2017-12-11:
I get it; mine was just a suggestion from the side line. And yes, more tests...

geowar1, 2017-12-11:
Can we change the color hex values to support the alpha channel? The xml schema is hard-wired to 6 hex digits… and most of the color utilities only parse or output 6 hex digits. There are a ton of places in the LayoutEditor where the alpha channel would be appropriate. (Yeah, another suggestion from the side lines… ;-)

bobjacobsen, 2017-12-11:
> Can we change the color hex values to support the alpha channel? 

That would be fine for the 4.11.* line.  

We're not changing anything new like that in 4.9.*; the goal is stability now.

rhwood, 2017-12-12:
We should stick with the 6 digit color RGB color set, since that is valid in HTML, CSS, and SVG, and understood by a number of tools and the swing toolkit, but the eight character RGBA value set is nonstandard. I recommend the use of a separate opacity value from 0.0 to 1.0, as used in the aforementioned standards, and known to a number of tools and toolkits instead. This specifically means that the stored values can be passed by the JMRI web server without modification to web panel clients.

pabender, 2017-12-13:
@pnbgit,

I just meted in a fix for this ( #4548 ) that didn't require as much surgery as I thought ( though I did remove the unused color changers ).

At some point in the next few hours, installers will be built and placed here:

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

If you could grab the appropriate installer and try that later today or tomorrow, I would appreciate it.
( just make sure the top build is #2092 or later )

pnbgit, 2017-12-14:
Tested in 4.9.8 Build #3 and working fine, so the issue of changing properties affecting the background color is now fixed.

I'll create a separate issue for creating a user friendly way of selecting a transparent background

pabender, 2017-12-16:
closing as this now is being reported as fixed.  Please reopen if the problem returns.

closed 2017-12-16 by pabender