JMRI/JMRI #4550: Surface Pro 4 and JMRI

Squizzyt, 2017-12-13:
The Surface Pro 4 has only one USB port (COM3) which is apparantly shared.  When initializing DecoderPro, I get the following message which I have also attached:
_'Error Initializing DecoderPro
Unable to create connection "LocoNet" (L)
Serial Port COM3 is in use
Unable to run startup actions due to earlier features
....'_

The session.log shows the following:
'2017-12-13 12:50:43,744 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-12-13 12:50:43,759 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Al\JMRI\log\messages.log [main]
2017-12-13 12:50:43,759 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Al\JMRI\log\session.log [main]
2017-12-13 12:50:43,759 apps.AppsBase                         INFO  - DecoderPro version 4.9.7+R72446d4 starts under Java 1.8.0_151 on Windows 10 amd64 v10.0 at Wed Dec 13 12:50:43 AEST 2017 [main]
2017-12-13 12:50:44,006 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f0c58c9 [main]
2017-12-13 12:50:44,229 node.NodeIdentity                     INFO  - Using jmri-BC8385C75891-3f0c58c9 as the JMRI Node identity [main]
2017-12-13 12:50:47,971 locobuffer.LocoBufferAdapter          ERROR - COM3 port is in use: errno() == 0 [main]
2017-12-13 12:50:47,971 jmrix.ConnectionConfigManager         ERROR - Unable to create jmri.jmrix.loconet.pr3.configurexml.ConnectionConfigXml for [Element: <connection/>], load returned false [main]
2017-12-13 12:50:47,971 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create several of your connections. [main]
2017-12-13 12:50:48,040 loconet.LnPowerManager                ERROR - Power Manager Created, yet there is no Traffic Controller [main]
2017-12-13 12:50:48,257 plementation.JmriConfigurationManager ERROR - Exception initializing apps.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Al\JMRI\My_JMRI_Railroad\ [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Al\JMRI\My_JMRI_Railroad\ [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Al\JMRI\ [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Al\ [main]
2017-12-13 12:50:56,882 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2017-12-13 12:50:57,483 gui3.TabbedPreferencesAction          INFO  - We are already waiting for the preferences to be displayed [main]
2017-12-13 12:51:07,902 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]'

I have tried adding a 4 port USB HUB (which plugs into Port COM3) but still get the same 'in use' error on Ports COM4, 5, 6 and 7.

I am using JMRI 4.8 and have Java 8 Update 151

Can anybody please suggest a fix.
[image]

devel-bobm, 2017-12-13:
I infer from the text above a basic misunderstanding of how Windows assigns COM ports.  Windows does not assign one COM port per USB port.

Instead, it assigns one COM port to each USB device which provides a "serial communications device" function.  Windows will also make certain assumptions about some USB hardware, and this means that Windows can assign a different COM port number to some USB devices if the device is plugged into a different USB port or via a (different) USB Hub.  Other USB devices are better-implemented and do not force Windows to this behavior.

This sort of user-level problem is typically one of user misunderstanding of which actual COM port number Windows has assigned.  And the typical "fix" is to have the user open the Windows "Control Panel" "Device Manager" feature to find the device entry and its associated COM port.  First, unplug the USB device.  Then open "Device Manager" open.  Then plug in the USB device.  A new entry (or possibly multiple entries) should appear.  The one that is important will be in the "Ports (COM & LPT)" entry, or an entry that has a similar name.   Examination of the new entry shows the COM port.

On every Windows-based PC that I've seen recently, there has ALWAYS been at least one COM port assigned to INTERNAL functions of the PC.  I would guess that COM3 is assigned to this internal functionality, and not to the "layout interface".  In the Windows-based laptop (from which I am posting this reply) with no USB devices attached, Device Manager shows "COM3" is assigned to "Intel(R) Active Management Technology - SOL".

Are you using a LocoBuffer-USB or some sort of USB-to-Serial device plus a LocoBuffer?  I suspect the later, and I suspect that your serial device is not properly installed in Windows.  

Note that many USB-to-Serial devices are not supported by modern Windows O/Ses.  I have several USB-to-Serial devices which haven't been usable since Windows 7.  Also, some USB-to-Serial devices can be electrically disabled by some "more-recent" Windows driver software if the driver identifies the integrated circuit as a "not-authorized clone".

You really need to focus on getting the Windows USB functionality to "enumerate" the USB device, and figure out what COM port Windows actually assigns to it.  I described that earlier in this reply.  Then use that COM port assignment within JMRI.

Squizzyt, 2017-12-14:
Thanks for your prompt reply Bob.
You are right, I do not fully understand how Windows assigns ports, but I did check 'Device Manager' where there is no 'Ports (COM & LPT)' shown until I plug the PR3 in and then COM3 pops up.
At the point where the JMRI wizard should start I get the Error message instead (I posted the wrong message, the second line should read 'Serial Port COM3 is in use').
My computer is brand new and I have no problems using the single USB port with hubs, DVD player, iPhone, camera or thumb drives.
I am not sure if I am using LocoBuffer-USB or USB-to-Serial device plus a local buffer.  My DCC system uses a Digitrax Zephyr DCS51 with a PR3 connecting the computer via USB.
I did plan to replace my old slow Fujitsu 880 with the Surface Pro 4.  The Fujitsu has five USB ports with its stand and works with JMRI as advertised but it is extremely slow.
Thank you so much for your explanation and assistance.

devel-bobm, 2017-12-14:
(I had guessed LocoBuffer based on a message in the log; it turns out that JMRI uses its LocoBuffer code to communicate with the PR3; that led me to an incorrect assumption.)

The PR3 is most definitely one of the "poorly-behaving" devices where the COM port assignment changes frequently.  

If you plug the PR3 into a USB hub, you will get a different COM port number.  If you move it to a different port on the USB hub, you will get yet again a different COM port number. This is typical Windows behavior with the PR3, and it makes it critical that you check Device Manager to ensure that you have the correct COM port number each time you change USB connectivity.  And because other USB device types can implement the same USB "serial communication" protocol, you probably need to check Device Manager for the PR3 COM port number before starting JMRI if you have changed any USB connectivity.

I have seen a PR3 COM port assignment go from COM8 all the way up to COM43 over the course of a few months.  All because the PR3 doesn't implement one critical aspect of the USB standard, and because other devices which implement USB "serial communication" protocols also do not implement that critical aspect of the USB standard.

So, I strongly suggest that you re-check Device Manager to ensure the PR3 is _still_ at COM3.

Beyond that, I would guess that the old machine was running a 32-bit Windows O/S, and that the new machine is running a 64-bit Windows O/S.  For 64-bit Windows O/Ses, it is _necessary_ to update the PR3 firmware with the firmware update file from the Digitrax web site.  If the older PR3 firmware is in-use with a 64-bit Windows operating system, the typical failure is one which stops the JAVA runtime environment.  I have not seen "old PR3 firmware" cause JMRI to fail to start, but it _could_ have such a result.

These are my best guesses.

Squizzyt, 2017-12-23:
Due to family commitments leading up to Christmas, I have not had time to explore all possibilities, so I have cleaned up the Fujitsu and put it back on-line.
I am determined to solve this problem so I will get back to you in the New Year when I try again.
Thanks for your interest and help.
A Merry Christmas to you and your family and I wish you a happy and healthy New Year.

rhwood, 2020-05-13:
Has this issue been resolved?

danielb987, 2020-08-19:
I'm closing this issue since there has been no comment on the problem for over two years.

closed 2020-08-19 by danielb987