JMRI/JMRI #4558: Layout editor toolbar combo box color coding does not work with Windows display preferences

pnbgit, 2017-12-14:
A really useful feature (from 4.7.3 I believe) is that the the Layout editor toolbar combo boxes are colour coded - quote from help:

> If you enter the name of an existing item then the background color of the combo box will turn green. If the item doesn't match the name of an existing item then the background will ether turn yellow if a new item with that name will be created or red if that name can not be used.

However, that feature does not work if you have the display preferences set to "Windows" or "Windows Classic". In that case the combo box backgrounds remain stubbornly white. 

These display preferences  are the preferred settings for Windows users, because the open and save file dialogs are much more user friendly using the OS native display settings.

bobjacobsen, 2017-12-14:
Comparing java/src//jmri/util/swing/JmriBeanComboBox.java (where the problem is reported) to java/src/jmri/jmrit/symbolicprog/EnumVariableValue.java  (which seems to work fine in the DecoderPro panes), I think what's missing is a call to `setOpaque(true)` after the call to set the background color.

geowar1, 2017-12-15:
See PR #4560

closed 2021-04-14 by stale[bot]