JMRI/JMRI #4562: Misbehavior when closing preferences window

bobjacobsen, 2017-12-15:
From @geowar1:

And if you open the prefs and try to close them it asks you if you want to save your changes... even if you didn’t change anything or used the “save” button. And then it asks if you want to restart. Regardless of what you answer it will then ask if you want to save the (NON) changes when you quit. The quickest road to sanity is to quit after you open prefs; that way you'll only get one dialog asking if you want to save your (NON) changes instead of three.

(copied from #4378 about a different issue)

rhwood, 2017-12-15:
I can only recreate this by:
- using profiles that start with errors that trigger changes to the defaults
- creating a Profile using an older version of JMRI (4.2 or earlier) and then using that Profile in a recent version of JMRI (head of current master, or 4.4 and later) -- this is because how startup preferences are stored and managed changed during the 4.3 development cycle

If @geowar1 saved his preferences just once, this behavior would stop.

SteveRawlinson, 2017-12-15:
I see behavior exactly as described by @geowar1 every time I open preferences. I have saved (real) changes to my preferences regularly. I've saved them again just now, restarted JMRI and seen the same thing: open preferences, close preferences with nothing touch, get prompted to save them. Get prompted again on closing JMRI.

rhwood, 2017-12-15:
@SteveRawlinson please zip and upload your entire profile to this issue, so I can see what’s going on in it.

SteveRawlinson, 2017-12-16:
Here you go.
[My_JMRI_Railroad (2).zip](https://github.com/JMRI/JMRI/files/1564835/My_JMRI_Railroad.2.zip)

bobjacobsen, 2017-12-16:
I think I see the behavior @SteveRawlinson is seeing, but it's very hard to tell for sure.  I have to change the serial port name and script location manually (via edit), and I'm not at all sure I've done that right given the large amount of replication in how the data is stored across properties and XML in multiple directories (@rhwood, might you get time sometime to document those relationships?)

What I'm seeing is that the serial connection defines the LocoNet command station as DCS50, but when it's time to check the "dirty" status of the preferences, jmrix.AbstractPortController ends up marking the connection pane "dirty" at about line 250 because the "Command station type" is set to "DCS50 (Zephyr)" instead of the  "DCS100 (Chief)" it expects.

But this seems weird, because it's comparing the current value (DCS50) to "options[0]", the first default.  It's as if it's comparing whether somebody has modified the value off the initial default (e.g. as if first time config) instead of comparing whether it's been modified off the previously configured value.

@SteveRawlinson could you test this, please?  Temporarily set the command station in the preferences to DCS100, save, restart, then see if the window is behaving properly (OK to set it back at the end).  Thanks

SteveRawlinson, 2017-12-17:
@bobjacobsen as you suspected, changing the command station type to DCS100 makes the preferences panel behave itself.

bobjacobsen, 2017-12-17:
> changing the command station type to DCS100 makes the preferences panel behave itself.

Thank you!

Am away from hardware for a bit, but will check how this works in a couple other hardware systems & develop a fix. If it’s just localized to Digitrax command station, I’ll edit the Issue description. 

Anybody have a non-Digitrax example?

rhwood, 2017-12-18:
I get the feeling that this particular occurrence of this is localized to that connection, and that any other connections that have similar problems also have specific fixes.

geowar1, 2017-12-18:
@rhwood >If @geowar1 saved his preferences just once, this behavior would stop.
ROFL!!! Uh, "No". I've saved hundreds of times… Like I said in my OP… the "Save" button doesn't change the behavior: I'm still asked to save/relaunch every time I open prefs.

geowar1, 2017-12-18:
I've tracked down the issue: I set a break point in the isDirty() method of AbstractPortController. Run, open prefs, close prefs... hit breakpoint, check if "currentValue.equals(options[0])" is true; continue until it's false and... currentValue ("V7.x.x") != options[0] ("V6.x.x"). If I then set the "USB Version" to "V6.x.x" and save… then everything works as expected.

geowar1, 2017-12-18:
Nope: Still dirty. System "PowerPro" != "Power Cab". (After (non-)"Save")

PR #4685 (2018-01-03): Fix Preferences "wants to save when nothing changed" behavior
Resolves preferences "wants to save when nothing changed" behavior due to poor isDirty checking in the connection code. Does this with a heuristic, which is that the first configured non-null value for options is the value from the previous configuration.  This works with the existing preferences code in any case.

Closes #4562 at least in that specific case.  I think this is the general solution, but it's certainly possible that there's other causes of similar behavior in the communications code. Please open a new issue if there's  [... cut]
closed 2018-01-03 by bobjacobsen