JMRI/JMRI #4576: Digitrax DCS100, Slot monitor, Config DCS with JMRI

Walter71342JMRI, 2017-12-18:
Using JMRI 4.11.1ish+jenkins+20171214T1517Z+Rb816599 , with the slot monitor open do a config of DCS100 setting OpSw 36 to C throws this on the System Console:

2017-12-17 23:26:15,992 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-12-17 23:26:15,992 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Walt\JMRI\log\messages.log [main]
2017-12-17 23:26:15,992 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Walt\JMRI\log\session.log [main]
2017-12-17 23:26:16,008 apps.Apps                             INFO  - PanelPro version 4.11.1ish+jenkins+20171214T1517Z+Rb816599 starts under Java 1.8.0_151 on Windows 10 x86 v10.0 at Sun Dec 17 23:26:16 CST 2017 [main]
2017-12-17 23:26:18,884 apps.Apps                             INFO  - Starting with profile HOn3.3f3c2e75 [main]
2017-12-17 23:26:19,040 node.NodeIdentity                     INFO  - Using jmri-64006A824CD2-3f3c2e75 as the JMRI Node identity [main]
2017-12-17 23:26:19,399 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: true CTS: true  CD: false [main]
2017-12-17 23:26:20,540 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_151 [main]
2017-12-17 23:26:20,540 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Walt\Dropbox\jmri\ [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Walt\JMRI\HOn3\ [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Walt\JMRI\ [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Walt\ [main]
2017-12-17 23:26:20,556 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Walt\Dropbox\jmri\py scripts\ [main]
2017-12-17 23:26:20,946 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2017-12-17 23:26:34,757 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Invalid index
	at javax.swing.DefaultRowSorter.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.repaintSortedRows(Unknown Source)
	at javax.swing.JTable.sortedTableChanged(Unknown Source)
	at javax.swing.JTable.tableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableRowsUpdated(Unknown Source)
	at jmri.jmrix.loconet.slotmon.SlotMonDataModel$Notify.run(SlotMonDataModel.java:642)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
2017-12-17 23:26:45,563 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Invalid index
	at javax.swing.DefaultRowSorter.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.repaintSortedRows(Unknown Source)
	at javax.swing.JTable.sortedTableChanged(Unknown Source)
	at javax.swing.JTable.tableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableRowsUpdated(Unknown Source)
	at jmri.jmrix.loconet.slotmon.SlotMonDataModel$Notify.run(SlotMonDataModel.java:642)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
2017-12-17 23:27:17,885 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Invalid index
	at javax.swing.DefaultRowSorter.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.repaintSortedRows(Unknown Source)
	at javax.swing.JTable.sortedTableChanged(Unknown Source)
	at javax.swing.JTable.tableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableRowsUpdated(Unknown Source)
	at jmri.jmrix.loconet.slotmon.SlotMonDataModel$Notify.run(SlotMonDataModel.java:642)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

Doing the Unexpected?
Walt

rhwood, 2017-12-18:
This is happening only when the slot that changes is hidden because "show unused slots" is not checked.

Walter71342JMRI, 2017-12-18:
With show All Slots checked it throws this on the System console:

2017-12-18 09:37:26,776 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-12-18 09:37:26,783 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Walt\JMRI\log\messages.log [main]
2017-12-18 09:37:26,783 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Walt\JMRI\log\session.log [main]
2017-12-18 09:37:26,787 apps.Apps                             INFO  - PanelPro version 4.11.1ish+jenkins+20171214T1517Z+Rb816599 starts under Java 1.8.0_151 on Windows 10 x86 v10.0 at Mon Dec 18 09:37:26 CST 2017 [main]
2017-12-18 09:37:35,333 apps.Apps                             INFO  - Starting with profile HOn3.3f3c2e75 [main]
2017-12-18 09:37:35,486 node.NodeIdentity                     INFO  - Using jmri-64006A824CD2-3f3c2e75 as the JMRI Node identity [main]
2017-12-18 09:37:35,851 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: true CTS: true  CD: false [main]
2017-12-18 09:37:37,044 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_151 [main]
2017-12-18 09:37:37,048 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [main]
2017-12-18 09:37:37,065 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-12-18 09:37:37,066 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Walt\Dropbox\jmri\ [main]
2017-12-18 09:37:37,066 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Walt\JMRI\HOn3\ [main]
2017-12-18 09:37:37,066 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Walt\JMRI\ [main]
2017-12-18 09:37:37,066 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Walt\ [main]
2017-12-18 09:37:37,066 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Walt\Dropbox\jmri\py scripts\ [main]
2017-12-18 09:37:37,485 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2017-12-18 09:37:53,921 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Invalid index
	at javax.swing.DefaultRowSorter.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.repaintSortedRows(Unknown Source)
	at javax.swing.JTable.sortedTableChanged(Unknown Source)
	at javax.swing.JTable.tableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableRowsUpdated(Unknown Source)
	at jmri.jmrix.loconet.slotmon.SlotMonDataModel$Notify.run(SlotMonDataModel.java:642)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
2017-12-18 09:38:04,264 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IndexOutOfBoundsException: Invalid index
	at javax.swing.DefaultRowSorter.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.convertRowIndexToView(Unknown Source)
	at javax.swing.JTable.repaintSortedRows(Unknown Source)
	at javax.swing.JTable.sortedTableChanged(Unknown Source)
	at javax.swing.JTable.tableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableChanged(Unknown Source)
	at javax.swing.table.AbstractTableModel.fireTableRowsUpdated(Unknown Source)
	at jmri.jmrix.loconet.slotmon.SlotMonDataModel$Notify.run(SlotMonDataModel.java:642)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
[... cut ...]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

Walt

rhwood, 2017-12-18:
This is a real issue, I just mentioned the apparent cause so that if someone looks at it, they have some hint about why this is happening.

rhwood, 2017-12-21:
Fixed by #4601

closed 2017-12-21 by rhwood

Walter71342JMRI, 2017-12-21:
Thanks Randall!
Did not realized I had closed it. Not sure how I did that.