JMRI/JMRI #4593: Delete Signal Head fails 4.11.1ish

KenC57, 2017-12-20:
With current head we at least get a fault in the log. Yesterday, we got nothing in the log.

2017-12-20 07:52:54,567 beantable.BeanTableDataModel          ERROR - Exception while deleting bean [AWT-EventQueue-0]
java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.util.concurrent.FutureTask.report(Unknown Source)
	at java.util.concurrent.FutureTask.get(Unknown Source)
	at javax.swing.SwingWorker.get(Unknown Source)
	at jmri.jmrit.beantable.BeanTableDataModel$DeleteBeanWorker.done(BeanTableDataModel.java:952)
	at javax.swing.SwingWorker$5.run(Unknown Source)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(Unknown Source)
	at sun.swing.AccumulativeRunnable.run(Unknown Source)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(Unknown Source)
	at javax.swing.Timer.fireActionPerformed(Unknown Source)
	at javax.swing.Timer$DoPostEvent.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NullPointerException
	at jmri.jmrit.blockboss.BlockBossLogic.vetoableChange(BlockBossLogic.java:1272)
	at jmri.managers.AbstractManager.fireVetoableChange(AbstractManager.java:362)
	at jmri.managers.AbstractManager.deleteBean(AbstractManager.java:165)
	at jmri.jmrit.beantable.BeanTableDataModel$DeleteBeanWorker.doInBackground(BeanTableDataModel.java:843)
	at jmri.jmrit.beantable.BeanTableDataModel$DeleteBeanWorker.doInBackground(BeanTableDataModel.java:831)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

KenC57, 2017-12-21:
A SSL without a signal head, I don't think you can even access it, so I'd see good reason to either not load it or consider it an error in the panel. I suspect it is leftover from other edits in the past.

I wonder how SML acts if either of the masts doesn't exist?

bobjacobsen, 2017-12-27:
Is this fixed now?  #4626 drops (with logging) SSL that don't refer to a signal head, and has more armoring against NPEs if that happens during execution.

KenC57, 2017-12-27:
Do we have a panel that had the problem (I likely do, but don't recall which
user that was)? I'll try to back trace my emails and spot who and which
panel file.
-ken c

KenC57, 2017-12-27:
I think I found the file, testing now.
-ken c

KenC57, 2017-12-27:
I did find a file that had a case of a SSL that had a signal name, but no
sensors or turnouts. It leaves these in the console. But this isn't this
problem, correct?

2017-12-27 16:35:23,909 automat.Siglet                        ERROR - Siglet
start invoked for "Dairy EastBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]
2017-12-27 16:35:23,910 automat.Siglet                        ERROR - Siglet
start invoked for "Quarry EastBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]
2017-12-27 16:35:23,910 automat.Siglet                        ERROR - Siglet
start invoked for "Dairy WestBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]
2017-12-27 16:35:23,910 automat.Siglet                        ERROR - Siglet
start invoked for "Dairy West LBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]
2017-12-27 16:35:23,910 automat.Siglet                        ERROR - Siglet
start invoked for "Quarry WestBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]
2017-12-27 16:35:23,911 automat.Siglet                        ERROR - Siglet
start invoked for "CY EastBlockBossLogic", but no inputs provided
[AWT-EventQueue-0]

-ken c

bobjacobsen, 2017-12-27:
That’s a different form of blown signal definition. But it would have previously caused the same problem with not being able to delete Turnouts (any of them).  Could you try deleting a few and see if they go away? And thanks for working on this. 

Bob

KenC57, 2017-12-28:
Problem is when I load the panel file, I lose function of the main window
and can't open the tables. Went back and opened the tables first, then
loaded file. I still didn't have a working main window, so I couldn't get to
the 'simple signal logic' tool via that but did find it via the panel. After
fiddling around longer I did find the main window started working again many
minutes later. If I try doing an update/apply of the SSL, I get the same
message about no inputs provided. Is there a way to delete the SSL??

Anyhow, moving on to the turnout delete, I can't figure which one. It seems
this panel has all the signals now in Logix instead of the SSL. But found
NT114 was in the Logix. Trying to delete it gave: turnout in use by Logix...

So I'm guessing the panel file that what is needed for this issue is not
this file. Now to continue trying to find one. You remember any clue in the
xml to search for?

-ken c

bobjacobsen, 2017-12-28:
That's a different problem, different file.

I'm closing this. Please either re-open it or create a new issue if there's another example of not being able to delete something.

closed 2017-12-28 by bobjacobsen