JMRI/JMRI #4599: Table's will mot open

crowninvesting, 2017-12-20:
Using 4.8 I could not get any tables to open. Upgraded to 4.10 and same problem.

KenC57, 2017-12-20:
After you try to open the tables, please check the 'Windows' tab on the menu bar. If you see a 'Untitled Table' or something close to that, click on it. That should show a very small window with very little in it. If you wait a very long time it will populate, and I mean like minutes or an hour. If this happens, you are having the audio library bug.

To work around this, create a file in your $HOME/JMRI directory named jmri.conf and put this line in the file:

default_options="-J-Djogamp.gluegen.UseTempJarCache=false"

crowninvesting, 2017-12-21:
Hi; thanks for the quick reply. I don’t think this is my problem. I tried it but it is still doing the same thing. You are correct; when I do try to open a table and look at the open windows I do see a window called Turnouts but it is blank; nothing at all (same for all other windows). I let set for about two hours and nothing. I tried your config file, but it is doing the same thing. Everything seemed fine until this last build (even though I did not upgrade to 4.10 until after the problem came up). Also, when I try to open a table the blank window that does show cannot be closed and I can’t close JMRI at all I have to Ctrl-Alt-Del to end the Java task that is running to close JMRI.

Dan Murphy
CEO, Broker
CrownHouses Realty
dmurphy@crownhouses.com
www.crownhouses.com
Mobile: (949) 353-9100

crowninvesting, 2017-12-21:
That did it, thanks very much.

KenC57, 2017-12-23:
Was wrongly closed. Issue still exists. Only a workaround has been provided on user by user basis. Keeping this open makes it easy to find.

KenC57, 2017-12-26:
This needs to stay open, so people can find the workaround, until we are shipping a version that fixes this. A workaround is not a fix.

bobjacobsen, 2017-12-26:
@kenc57 I thought 4.11.1 had the fix for this in its LaunchJMRI.exe. Is that not working?

klk32003, 2017-12-26:
JogAmp-JOAL, the JMRI audio implementation, provides two ways for library loading:
1) Automated Native Library Loading
2) Traditional Native Library Loading

JMRI is designed to use 2), but the default is 1). The default way uses a cache and that cache is always created at JOAL initialisation time.
For JMRI that means, the cache is created, but it is never used by JOAL.
When Windows 10 came out with enhenced security policies, this cache creation made some problems in rare cases. The JogAmp-JOAL developers discussed it intensively. But, as often on sporadic problems, a solution was not found. So the developers recommended (for such sporadic cases) to go back to 2). 
Well, JMRI is still using 2). The cache is not used and not needed. With jmri.conf and a Java option we demonstrated the effectiveness. Now, after more than 10 cases, we moved the Java option to the JMRI Windows launcher.
Yes, it is not a fix. But I think it is an obvious way to bypass this issue.

rhwood, 2017-12-27:
4.11.1 has been confirmed to fix this.

closed 2017-12-27 by rhwood

KenC57, 2017-12-27:
I just confirmed that 4.11.1 does indeed fix this. But I didn't see mention of that in the release notes. Not sure which section of the note should have this? Would this be installation, launcher, tables, or what?? I feel we've had enough of this issue from users that I'll post about it. But it does become one candidate for a 4.10.1 release.

KenC57, 2017-12-27:
I had just proved that for myself. But I raise the question of where it
should be listed in the release note?? Any ideas??
-ken c

bobjacobsen, 2017-12-27:
> I had just proved that for myself. But I raise the question of where it
> should be listed in the release note?? Any ideas??
> -ken c

The “Miscellaneous” section seems appropriate.

esfeld, 2018-08-31:
I downloaded ver 4.12 and still can not get a TO table!  Pressing Window\ Turnouts I get a blank empty TO table.  Has anyone found a real fix for this?  Obviously it has not been fixed!

KenC57, 2018-08-31:
After trying to open a table, check the 'Windows' tab, does it show a table
listed? If you click on that item, does it bring up an extra small blank
window?

bobjacobsen, 2018-08-31:
@esfeld Are you loading a panel file when the program starts up?  Problems in that file can cause a similar problem.

Please look in the Console (under the Help menu) to see if there are any WARN or ERROR messages.

KenC57, 2018-08-31:
One more comment, give the system an hour or two, does the table build
eventually? This is a symptom of a particular issue. If you are having this
issue with 4.12, that hints that the install may be incomplete or there were
some leftover parts from prior installs, usually that doesn't happen.

If this is the case, do a complete uninstall of the program. Make sure the
C:\Program Files (x86)\JMRI is gone. Then do a fresh install, make sure you
select the 'install for all users' on the install.

commit 99cef39a (2017-12-24): Merge pull request #4612 from klk32003/Windows-audio-issue

Add jvm option to the Windows launcher to fix the audio issue
Fixes #4599