JMRI/JMRI #4608: Uhlenbrock Lissy and JMRI

michael71, 2017-12-22:
is anyone actually using (Uhlenbrock) Lissy-detectors with JMRI? 

I tried, but in the LocoNet Monitor the loco-address is displayed as "speed" and the loco-speed is displayed as address. (lissy config: no automatics, just report address and speed)

thanks,
Michael

devel-bobm, 2017-12-25:
@michael71,

So far as I can tell, the JMRI code-base, going back at least 18 months, implements code to report "LISSY IR" messages, in a way that is consistent with the only documentation I have seen for the "Lissy IR" LocoNet messages.  

Do you have any references you can point us to a detailed description of the actual LocoNet messages which the Lissy device can generate?

Can you provide some LocoNet Monitor logs which include RAW HEX data for a couple of LocoNet LISSY messages, for a couple of different addresses?  Please include a description of what you would expect to see for each of the LISSY messages.

michael71, 2017-12-29:
@devel-bobm - hello Bob,

A made a few experiments with different settings of the lissy transmitter and receivers - and compared the HEX-messages with the output text from the Uhlenbrock "LocoNet Monitor" tool.

here is the result (raw data below):

There are 3 different types of messages:
1) loco address (and train category) reports without directional information (when "lissy" is receiving with a single IR receiver)
2) loco address (and train category) reports with directional information ("lissy" configured for double IR receiver)
3) loco speed information (in km/h, if calibrated correctly for scale and distance between the two sensors)

The Lissy-messages m[] are 8 bytes long, when the bytes are given in an array m[], the meaning of the bytes is (my interpretation, I did not find offcial information):

m[0] == 0xE4  always
m[1] == 0x08  always
(m[2] & 0x03) + 0x01 = train category (1..4)

m[3] & 0x60 
   = 0x60 => loco addr message, direction S2->S1  (IR Sensors)
   = 0x40 => loco addr message, direction S1->S2
   = 0x20 => speed message
   = 0x00 => loco addr message, no direction

(m[3] & 0x1F ) *128 + (m[4] & 0x7F) = lissy address (1 .. 4095)

(m[5] * 128) + (m[6] & 0x7F) = loco address OR loco speed

------------------------------------------- raw data ------------------------------

Lissy 1, Lok 43 (=2B), Cat=4
E4 08 00 20 01 00 1F 2D    => loco speed 0x1F
E4 08 03 40 01 00 2B 7A    => loco addr and dir and cat

Lissy2, Lok 43 (=2B), Cat=4
E4 08 00 20 02 00 21 10     => loco speed 0x21 = 33km/h
E4 08 03 60 02 00 2B 59    => loco addr and dir

higher speed, direction reversed

Lissy2, Lok 43 (=2B), Cat=4
E4 08 00 20 02 00 30 01  => loco speed
E4 08 03 40 02 00 2B 79  => loco addr and dir

Lissy1, Lok 43 (=2B), Cat=4
E4 08 00 20 01 00 2C 1E  => loco speed
E4 08 03 60 01 00 2B 5A => loco addr and dir

Lissy1, Lok53, cat 3
E4 08 00 20 01 00 2B 19
E4 08 02 40 01 00 35 65

Lissy2, Lok53, cat 3
E4 08 00 20 02 00 2C 1D
E4 08 02 60 02 00 35 46 

reversed direction

Lissy2, Lok53, cat 3
E4 08 00 20 02 00 27 16
E4 08 02 40 02 00 35 66

Lissy1, Lok53, cat 3
E4 08 00 20 01 00 2A 18
E4 08 02 60 01 00 35 45 

Lissy2, Lok40, cat ?
E4 08 00 20 02 00 3D 0C
E4 08 00 40 02 00 28 79

Lissy1, Lok40, cat ?
E4 08 00 20 01 00 33 01
E4 08 00 60 01 00 28 5A 

reversed dir
Lissy2, Lok40, cat ?
E4 08 00 20 01 00 4D 7F

E4 08 00 40 01 00 28 7A

Lissy1, Lok40, cat ?
E4 08 00 20 02 00 53 62
E4 08 00 60 02 00 28 59 

LISSY/MARCo Empf. 2: Adr.= 53 Kat.= 3 , von S1 nach S2
LISSY/MARCo Empf. 2: Geschw.= 26 Km/h
E4 08 00 20 02 00 1A 2B  => v
E4 08 02 40 02 00 35 66   => LISSY/MARCo Empf. 2: Adr.= 53 Kat.= 3 , von S1 nach S2

Einzelempfänger – ohne geschwindigkeit
E4 08 02 0D 33 00 35 1A 
LISSY/MARCo Empf. 1715: Adr.= 53 Kat.= 3 

E4 08 02 1F 7B 00 35 40 => LISSY/MARCo Empf. 4091: Adr.= 53 Kat.= 3 
Lissy Address = (msg[3] &0x1F ) *128 + (msg[4] & 0x7F) =
                0x1F * 128 + 0x7B 
          (31 *128)  + 123 = 4091
Lissy address range 1...4095 (LCNV[0] / LNCV[1])

Lissy #4091 (double-channel, speed and addr)
LISSY/MARCo Empf. 4091: Adr.= 53 Kat.= 3 , von S1 nach S2
E4 08 02 5F 7B 00 35 00 
                40 => adr, dir-S1->S2

LISSY/MARCo Empf. 4091: Geschw.= 37 Km/h
E4 08 00 3F 7B 00 25 72

Lok 53 = rückwärts F0 
LISSY/MARCo Empf. 4091: Adr.= 53 Kat.= 3 , von S2 nach S1
E4 08 02 7F 7B 00 35 20

LISSY/MARCo Empf. 4091: Geschw.= 39 Km/h
E4 08 00 3F 7B 00 27 70

LISSY/MARCo Receiver 4091: Addr.= 53 Cat.= 3 , from S2 to S1
E4 08 02 7F 7B 00 35 20 
               60 = 40+20 => adr, dir-S2->S1
LISSY/MARCo Receiver 4091: Speed = 58 Km/h
E4 08 00 3F 7B 00 3A 6D
                1F-7B => lissy-adr 4091
                20 => speed

LISSY/MARCo Receiver 1: Addr.= 53 Cat.= 3 , from S1 to S2
E4 08 02 40 01 00 35 65 
                40 => adr, dir-S1->S2

LISSY/MARCo Receiver 1: Speed = 52 Km/h
E4 08 00 20 01 00 34 06

Loco 0 = forwardsF0 
LISSY/MARCo Receiver 1: Addr.= 53 Cat.= 3 , from S2 to S1
E4 08 02 60 01 00 35 45
               60 = 40+20 => adr, dir-S2->S1
LISSY/MARCo Receiver 1: Speed = 55 Km/h
E4 08 00 20 01 00 37 05
                20 => speed

Train-Category changed from 3 to 1
LISSY/MARCo Receiver 4091: Addr.= 53 Cat.= 1 , from S2 to S1
E4 08 00 7F 7B 00 35 22 
               60 = 40+20 => adr, dir-S2->S1

LISSY/MARCo Receiver 4091: Speed = 21 Km/h
E4 08 00 3F 7B 00 15 42
               20 => speed

Train-Category changed to 4
LISSY/MARCo Receiver 4091: Addr.= 53 Cat.= 4 , from S2 to S1
E4 08 03 7F 7B 00 35 21 

LISSY/MARCo Receiver 4091: Speed = 15 Km/h
E4 08 00 3F 7B 00 0F 58

LONG address 

LISSY/MARCo Receiver 4091: (long)Addr.= 2465 Cat.= 4 , from S2 to S1
E4 08 03 7F 7B 13 21 26 
                          adr = 0x13*128+0x21 = 2465
           03 => cat 4
                1F / 7B => lissy#=4091
                 06 => dir-S2->S1

LISSY/MARCo Receiver 4091: Speed = 25 Km/h
E4 08 00 3F 7B 00 19 4E
R

devel-bobm, 2017-12-30:
@michael71,

You have provided some good information here.  I will see what I can do with it, and apply the results to LocoNet Monitor in some future release.

closed 2021-04-14 by stale[bot]